mne-tools/mne-python #8360: ValueError: metadata must have the same number of rows as events

SophieHerbst, 2020-10-13:
I am  trying to create epochs with metadata from raw, but run into an error with a piece of code that worked fine about 2 months ago (some previous version, likely 0.20.dev).
I updated to the latest master. 

I want to use a subset of events of interest, while my raw data contains more events.
Creating epochs without metadata works fine.

(I tried to comment out the size check in _check_metadata, but it breaks later than.)

>   File "/home/sh254795/anaconda3/lib/python3.7/site-packages/mne/utils/mixin.py", line 358, in _check_metadata
>     % (len(metadata), len(self.events)))
> 
> ValueError: metadata must have the same number of rows (38) as events (86)

Have there been any changes to this code that could explain the error?

Example: 

```
import os
import numpy as np
import mne
import pandas as pd

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False)
raw.crop(tmax=60).load_data()

# find all events
events = mne.find_events(raw, stim_channel='STI 014')

# define events of interest 
event_id = dict(A=4, B=3)

tmin = -0.1
tmax = 0.5

#%% create metadata for the events of interest

# count events of interest 
n_events = sum(events[:,2] == 3) + sum(events[:,2] == 4)

# create arbitrary metadata
METADATA = np.arange(0,n_events)
 
META_DF = pd.DataFrame(data=METADATA, columns = ['trial'])
   
# Epoch the data without metadata - FINE
print('creating epochs...')
epochs = mne.Epochs(raw, events, event_id, tmin, tmax)
      

# Epoch the data with metadata - BREAKS
print('creating epochs...')
epochs = mne.Epochs(raw, events, event_id, tmin, tmax,
                    metadata=META_DF)
```

cbrnr, 2020-10-13:
#7733 broke it - @larsoner?

larsoner, 2020-10-13:
I think we need to decide if we want to require metadata to have the same length as the original events array, or the events array that gets created by trimming to just what's in `event_id`. I think it makes the most sense to have it be the same length as events, especially when considering for example `event_repeated` behavior (and that way things don't depend on `event_id` -- you can change it without needing to change `metadata` to match). Under this definition I think the behavior is correct, and the fix would be to first restrict your `events` array that you pass to `epochs`.

larsoner, 2020-10-13:
Indeed our docs say:

> A pandas.DataFrame specifying metadata about each epoch. If given, len(metadata) must equal len(events).

larsoner, 2020-10-13:
(In other words, it was a bug that this worked in previous versions, and the error here is correct)

cbrnr, 2020-10-13:
👍  - should we add a test?

larsoner, 2020-10-13:
Looks like it's already covered:

https://github.com/mne-tools/mne-python/blob/master/mne/tests/test_epochs.py#L2695-L2707

SophieHerbst, 2020-10-13:
Hm, I understand your point, but it seems logical to me that one would want to use a selection of events to epoch, not all. 
I couldn't even predict how many events my raw has, because for instance button presses are in there and those depend on the subject. 
So if the metadata has to match the length of selected events, this would require to create metadata for events for which it is not needed. Especially tricky if metadata are loaded from a separate file which contains behavioral responses. 
I am afraid that this would increase the difficulty of working with metadata.

larsoner, 2020-10-13:
> So if the metadata has to match the length of selected events, this would require to create metadata for events for which it is not needed. Especially tricky if metadata are loaded from a separate file which contains behavioral responses.
I am afraid that this would increase the difficulty of working with metadata.

I would suggest you trim your `events` to the ones you plan on using. This could just be a quick mask line like:
```
events = events[np.in1d(events[:, 2], list(event_id.values())]
```

SophieHerbst, 2020-10-13:
Ok, great! 
Would the recommendation now be to add metadata to epochs after creating them, [as described here](https://mne.tools/stable/auto_tutorials/epochs/plot_30_epochs_metadata.html#sphx-glr-auto-tutorials-epochs-plot-30-epochs-metadata-py)? 
Otherwise, I think it would be good to have a tutorial which explains how to go from raw to epochs with metadata.

larsoner, 2020-10-13:
I think you can do it either way. In your case adding it afterward would make it so you don't need to restrict/mask your `events`

SophieHerbst, 2020-10-13:
Adding metadata later seems dangerous though when a rejection threshold is used in creating epochs.

larsoner, 2020-10-13:
Yes but you can do rejection afterward instead if you want:
```
epochs = Epochs(raw, events, event_id)
epochs.metadata = metadata
epochs.drop_bad(reject, flat)
```

SophieHerbst, 2020-10-13:
Yes, thanks! My point is more that metadata is a super useful feature, which we use more and more, but it is quite difficult to implement for people less proficient with python. 
Hence, one nice complete tutorial where the above points are made explicit would be great.

larsoner, 2020-10-13:
Maybe a good one for @drammock at some point?

hoechenberger, 2020-10-13:
> Yes but you can do rejection afterward instead if you want:
> 
> ```
> epochs = Epochs(raw, events, event_id)
> epochs.metadata = metadata
> epochs.drop_bad(reject, flat)
> ```

I think that's super easy to get wrong tbh… my muscle memory is already hard-coded to `Epochs(..., reject=...)`

just my 2 ct

agramfort, 2020-10-14:
yes events and metadata must have the same length. If it’s not the case limit the events in the first place before passing them to epochs