mne-tools/mne-python #8363: BUG: save_movie doesn't block inputs

larsoner, 2020-10-13:
When creating a bunch of movies in a script, windows pop up. If they get focus (e.g., on Linux where focus stealing isn't always handled properly) then doing something like hitting "f" will cause the "fly" (I assume?) callback to occur and the brain goes away, ruining the movie. It would be nice if we could block / skip / ignore both mouse and keyboard inputs to the window during the `save_movie` loop.

cc @GuillaumeFavelier

GuillaumeFavelier, 2020-10-13:
Would it work if you set `pyvista.OFF_SCREEN = True` to avoid having the windows opening in the first place?

larsoner, 2020-10-13:
I doubt this would work with `time_viewer=True` / traces

GuillaumeFavelier, 2020-10-14:
Alternatively, I was thinking about a dedicated offline thread with disabled interaction. In theory, this could build the movie in the background without blocking interactions on the main thread.

larsoner, 2020-10-14:
That sounds a lot more complicated code-wise (threads always are). I'm also not sure it will work because the rendering thread will for example need the camera in one fixed position and the user could be moving it, so somehow there have to be two states. I'm not sure it would really even end up being usable since the rendering thread will not release the GIL very often and could be CPU intensive.

To "disable interaction" I was really just thinking to override the event handlers/listeners of PyVista+VTK, and once rendering is complete (via try/finally) restore them. Ideally this would mean making some private PyVista attributes public, or just living with modifying some private things to be no-ops, and modifying them. At the VTK level I think we can just iterate over all renderer layers, [disable their interaction](https://github.com/pyvista/pyvista/blob/master/pyvista/plotting/plotting.py#L3879), then at the end, reenable their interaction -- so the question is what do we need to do to disable all PyVista shortcuts...

GuillaumeFavelier, 2020-10-14:
> so the question is what do we need to do to disable all PyVista shortcuts...

AFAIK those are bound to vtk KeyPressEvent

closed 2020-10-20 by larsoner