mne-tools/mne-python #8369: Issues on events of EEG data

BarryLiu-97, 2020-10-15:
My EEG Data's marker directly got from the machine is wrong, which we knew already, while we have some marker channels, aka the stimulus electrodes, the calculate the markers. Here are my  two questions:
1. How do I not import the default events got from the machine in this data, for example, in **eeglab**, we can set _importevent_ as off so we won't import the events. Then we add the events manually. Or, how do I delete all the information about events and then add events manually?
2. How do I get epoched data with markers(or events) into a 3_d tensor.

BarryLiu-97, 2020-10-15:
About the events calculated manually, I got the event latency and the value of the event.

agramfort, 2020-10-15:
This appears to be a question about usage, not a bug report or a new feature request. Please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) for usage questions.

closed 2020-10-15 by agramfort