mne-tools/mne-python #8376: Mapping Annotations to different time scales

adam2392, 2020-10-15:
#### Describe the new feature or enhancement
Is there a way to map annotations from the raw dataset onto the AverageTFR object? Since the tfr is occurring over windows, then the time scale of the annotations from the original time scale are inherently different.

E.g. if I have an event at sample 10000, with sfreq=1000, then it occurs at 10 seconds into the raw dataset. If i have an averageTFR that was computed with a moving window of say 1 second step size, then the new sample point of the event would be 10.

#### Describe your proposed implementation
Ideally this would consist of two things that I could see being useful:

1. A public function `map_annotations_to_sfreq` that allows mapping `Annotations` to different time-scales based on a passed say... `sfreq` value. So based on the above example, if I wanted to map `Annotations` that are based on `sfreq=1000`Hz, to some tfr object with `sfreq=1` Hz, then I would somehow pass that information to `map_annotations_to_sfreq` that returns me a new `Annotations` object that has `onset` and `duration` in the new time-scale.

2. Use the public function above to make `AverageTFR`, or `EpochsTFR`, etc. to keep their `Annotations` when the time-scale changes. 

I strongly would prefer having 1 because besides mne-python objects, I create other derivatives of the raw data that inherently have a "sliding" window component to them, and hence change the time-scale in the same way as a sliding FFT would.

#### Additional comments
@larsoner's comment on gitter for reference:

```
@adam2392 in principle we should be able to do that at least for EpochsTFR, can you open an issue feature request? For AverageTFR I'm not sure because we don't preserve annotations in Evoked even (just Raw and Epochs) so we'd probably want to do that first before doing it for AverageTFR
```

larsoner, 2020-10-19:
> A public function map_annotations_to_sfreq that allows mapping Annotations to different time-scales

I actually don't think we need this. Annotations [store data in seconds](https://mne.tools/stable/generated/mne.Annotations.html) not samples -- so they have no notion of (or need for) a `sfreq` (even if `sfreq` and sample numbers are used in the `Raw` instance to create the times in seconds). So I think all we really need to do is ...

> Use the public function above to make AverageTFR, or EpochsTFR, etc. to keep their Annotations when the time-scale changes.

... make EpochsTFR (and the functions that create them) respect the annotations properly.

mmagnuski, 2020-11-07:
That would be useful! I would first start with `.annotations` for Epochs as currently this attribute is only available for raw.
@larsoner although annotations contain time in seconds it might be more user friendly to turn this information into "epoch-centered" times during epoching, so: epoch index, epoch time, duration. Then for TFR we would only need to know the window length for each freq (n_cycles for example, which is not retained, as far as I know, in EpochTFR) and the time or percentage overlap of window with annotation required to make the window annotated.
I'd be happy to help with with this in december, now I don't have much time.

larsoner, 2020-11-09:
> I would first start with .annotations for Epochs as currently this attribute is only available for raw.
@larsoner although annotations contain time in seconds it might be more user friendly to turn this information into "epoch-centered" times during epoching, so: epoch index, epoch time, duration.

Makes sense. We might even be able to do this in a backward-compatible way by adding a `epoch_idx` attribute to the `Annotations` class that is `None` for raw, or a `ndarray` of `int` with shape `(n_epochs,)` when used with an Epochs instance.

As a concrete example, if you have events at sample numbers equivalent to raw t=0 and t=1 and *one* Raw annotation from 0.25-1.25 sec, if you epoch with `tmin=0., tmax=1.`you end up with *two* annotations in `epochs`, one for epoch index 0 with onset 0.25 and duration 0.75, and one for epoch index 1 with onset 0 and duration 0.25?

mmagnuski, 2020-11-09:
> Makes sense. We might even be able to do this in a backward-compatible way by adding a `epoch_idx` attribute to the `Annotations` class that is `None` for raw, or a `ndarray` of `int` with shape `(n_epochs,)` when used with an Epochs instance.

Oh, yes, good idea.

> As a concrete example, if you have events at sample numbers equivalent to raw t=0 and t=1 and *one* Raw annotation from 0.25-1.25 sec, if you epoch with `tmin=0., tmax=1.`you end up with *two* [...]

Yes, I think they should be split like this, that would keep the representation simple.

larsoner, 2021-02-24:
Since we have channel-specific annotation support now, I've re-titled this as the missing component, namely what @mmagnuski mentioned:

> I would first start with .annotations for Epochs as currently this attribute is only available for raw.

We need to preserve annotations in the Epochs (and EpochsTFR, etc.) objects next.

adam2392, 2021-10-15:
`Epochs` are constructed from `raw` so this should be straightforward to then add to the Annotations with `epoch_idx`. Then internally, formulate the full Annotations object that also consists of the Epoched events (what's currently stored as events in Epochs). I think this probably could be a `set_annotations` for `Epochs`, where it should operate more or less the same as it does with `Raw` except it checks for `epoch_idx`.

`EpochsTFR` is a bit more troublesome because it is formed from the TFR functions. Inside any TFR functions that operate on `Epochs` data structure, then it can just extract the annotations and do `set_annotations` on `EpochsTFR`.

I suppose there might be some complexities with `set_annotations` functionality for `Epochs`...

Is there anything internally more difficult? What is `FIFFB_MNE_ANNOTATIONS` used for?

larsoner, 2021-10-24:
> with EpochsArray the events onsets are just place older and the meas_date is absent.

`meas_date` being missing doesn't matter, we have the same problem sometimes for raw. Regarding placeholder -- sure, but I think for the "continuous time" representation to work for EpochsArray under the hood, we just need to make sure that the event onsets are spaced out by more than `(tmax - tmin) * sfreq`.

> So I don't see how annotations would make sense unless you mark them with the GUI and it becomes more meaningful. So maybe....

I think whatever *user-facing API* you are envisioning for adding annotations to individual epochs (one of these being the GUI example) can/will be the same regardless of whether our "storage backend" for the annotations keeps track of them in the annotations in terms of continuous time, or in terms of epoch-by-epoch annotations separate from the original times. The user won't "feel" our use of a continuous time backend for EpochsArray, but they definitely can for Raw->Epochs->Raw `annotations` round-tripping for example.

> What also worries me is the use change of sfreq eg if you resample the epochs you lose the provenance of sfreq don't you think?

We should probably adjust the `events[:, 0]` sample times when resampling and decimating, or alternatively (maybe better) store a `epochs._raw_sfreq` to keep track of the original sample rate. It actually seems like a bug that we don't do either of these. If we do change this behavior either way, then this issue becomes moot because we still can get from `events[:, 0]` (maybe we also need a `epochs._raw_first_samp`?) to the original/true human times (what raw Annotations tries to be in reference to, in some way) of the epoch t=0 points.

If you're not convinced by these arguments @agramfort I say we hold off for another couple weeks and hash it out at the next developer meeting. I think iterating there will be faster, and we don't need to do this for 0.24 so waiting a couple weeks isn't a bad idea anyway.

PR #10019 (2022-01-12): [MRG, ENH] Enable IO for saving/reading Epochs with Annotations
#### Reference issue
Closes: #8376 

Follow up to #9969 but requires that one to be merged first.

#### What does this implement/fix?
Enables IO for Epochs with Annotations:
- adds 1 FIFF constant for Raw_Sfreq. The First_time and Annotations constants already exist
- Adds saving logic to `def _save_part` in `epochs.py`
- [ ] Add logic for `def read_epochs`

#### Additional information
In order to add the IO portion, we need to add the Annotation functions and logic to `EpochsFIFF` object. Would the best way to do this is change a [... cut]
closed 2022-01-12 by larsoner