mne-tools/mne-python #8378: Setting standard montage with match_case=False failes

adam2392, 2020-10-15:
#### Describe the bug
When I have a dataset loaded in where all the channel names are in upper-cased, I want to set a standard montage, but I just realized that the casing matters by default. So I tried turning off `match_case`, but then I realized it doesn't work on the standard montages because it does not reach that part of the code in `set_montage`. 

Then the corresponding channels that don't match by case are dropped.

#### Steps to reproduce

```
montage =  = mne.channels.make_standard_montage('standard_1020')

# map channel names to upper-case

# try to set the montage
    raw.set_montage(montage, on_missing='warn',
                    match_case=False)
```

#### Expected results
*Replace this text with a description of what you expected to happen.*

#### Actual results

standard 1020 montage
```
<DigMontage | 0 extras (headshape), 0 HPIs, 3 fiducials, 343 channels> channel names: ['Fp1', 'Fpz', 'Fp2', 'AF9', 'AF7', 'AF5', 'AF3', 'AF1', 'AFz', 'AF2', 'AF4', 'AF6', 'AF8', 'AF10', 'F9', 'F7', 'F5', 'F3', 'F1', 'Fz', 'F2', 'F4', 'F6', 'F8', 'F10', 'FT9', 'FT7', 'FC5', 'FC3', 'FC1', 'FCz', 'FC2', 'FC4', 'FC6', 'FT8', 'FT10', 'T9', 'T7', 'C5', 'C3', 'C1', 'Cz', 'C2', 'C4', 'C6', 'T8', 'T10', 'TP9', 'TP7', 'CP5', 'CP3', 'CP1', 'CPz', 'CP2', 'CP4', 'CP6', 'TP8', 'TP10', 'P9', 'P7', 'P5', 'P3', 'P1', 'Pz', 'P2', 'P4', 'P6', 'P8', 'P10', 'PO9', 'PO7', 'PO5', 'PO3', 'PO1', 'POz', 'PO2', 'PO4', 'PO6', 'PO8', 'PO10', 'O1', 'Oz', 'O2', 'I1', 'Iz', 'I2', 'AFp9h', 'AFp7h', 'AFp5h', 'AFp3h', 'AFp1h', 'AFp2h', 'AFp4h', 'AFp6h', 'AFp8h', 'AFp10h', 'AFF9h', 'AFF7h', 'AFF5h', 'AFF3h', 'AFF1h', 'AFF2h', 'AFF4h', 'AFF6h', 'AFF8h', 'AFF10h', 'FFT9h', 'FFT7h', 'FFC5h', 'FFC3h', 'FFC1h', 'FFC2h', 'FFC4h', 'FFC6h', 'FFT8h', 'FFT10h', 'FTT9h', 'FTT7h', 'FCC5h', 'FCC3h', 'FCC1h', 'FCC2h', 'FCC4h', 'FCC6h', 'FTT8h', 'FTT10h', 'TTP9h', 'TTP7h', 'CCP5h', 'CCP3h', 'CCP1h', 'CCP2h', 'CCP4h', 'CCP6h', 'TTP8h', 'TTP10h', 'TPP9h', 'TPP7h', 'CPP5h', 'CPP3h', 'CPP1h', 'CPP2h', 'CPP4h', 'CPP6h', 'TPP8h', 'TPP10h', 'PPO9h', 'PPO7h', 'PPO5h', 'PPO3h', 'PPO1h', 'PPO2h', 'PPO4h', 'PPO6h', 'PPO8h', 'PPO10h', 'POO9h', 'POO7h', 'POO5h', 'POO3h', 'POO1h', 'POO2h', 'POO4h', 'POO6h', 'POO8h', 'POO10h', 'OI1h', 'OI2h', 'Fp1h', 'Fp2h', 'AF9h', 'AF7h', 'AF5h', 'AF3h', 'AF1h', 'AF2h', 'AF4h', 'AF6h', 'AF8h', 'AF10h', 'F9h', 'F7h', 'F5h', 'F3h', 'F1h', 'F2h', 'F4h', 'F6h', 'F8h', 'F10h', 'FT9h', 'FT7h', 'FC5h', 'FC3h', 'FC1h', 'FC2h', 'FC4h', 'FC6h', 'FT8h', 'FT10h', 'T9h', 'T7h', 'C5h', 'C3h', 'C1h', 'C2h', 'C4h', 'C6h', 'T8h', 'T10h', 'TP9h', 'TP7h', 'CP5h', 'CP3h', 'CP1h', 'CP2h', 'CP4h', 'CP6h', 'TP8h', 'TP10h', 'P9h', 'P7h', 'P5h', 'P3h', 'P1h', 'P2h', 'P4h', 'P6h', 'P8h', 'P10h', 'PO9h', 'PO7h', 'PO5h', 'PO3h', 'PO1h', 'PO2h', 'PO4h', 'PO6h', 'PO8h', 'PO10h', 'O1h', 'O2h', 'I1h', 'I2h', 'AFp9', 'AFp7', 'AFp5', 'AFp3', 'AFp1', 'AFpz', 'AFp2', 'AFp4', 'AFp6', 'AFp8', 'AFp10', 'AFF9', 'AFF7', 'AFF5', 'AFF3', 'AFF1', 'AFFz', 'AFF2', 'AFF4', 'AFF6', 'AFF8', 'AFF10', 'FFT9', 'FFT7', 'FFC5', 'FFC3', 'FFC1', 'FFCz', 'FFC2', 'FFC4', 'FFC6', 'FFT8', 'FFT10', 'FTT9', 'FTT7', 'FCC5', 'FCC3', 'FCC1', 'FCCz', 'FCC2', 'FCC4', 'FCC6', 'FTT8', 'FTT10', 'TTP9', 'TTP7', 'CCP5', 'CCP3', 'CCP1', 'CCPz', 'CCP2', 'CCP4', 'CCP6', 'TTP8', 'TTP10', 'TPP9', 'TPP7', 'CPP5', 'CPP3', 'CPP1', 'CPPz', 'CPP2', 'CPP4', 'CPP6', 'TPP8', 'TPP10', 'PPO9', 'PPO7', 'PPO5', 'PPO3', 'PPO1', 'PPOz', 'PPO2', 'PPO4', 'PPO6', 'PPO8', 'PPO10', 'POO9', 'POO7', 'POO5', 'POO3', 'POO1', 'POOz', 'POO2', 'POO4', 'POO6', 'POO8', 'POO10', 'OI1', 'OIz', 'OI2', 'T3', 'T5', 'T4', 'T6', 'M1', 'M2', 'A1', 'A2']

Dropping 20 channels from original dataset of 42 to set <DigMontage | 0 extras (headshape), 0 HPIs, 3 fiducials, 343 channels> montage...
['FP1', 'FP2', 'FZ', 'CZ', 'PZ', 'E', 'PG1', 'PG2', 'T1', 'T2', 'ECG1', 'ECG2', 'X9', 'X10', 'DC01', 'DC02', 'DC03', 'DC04', '$A1', '$A2']
```

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*

cbrnr, 2020-10-16:
I cannot reproduce. The following example works for me:

```python
from numpy.random import default_rng
import mne

fs = 256
data = default_rng(seed=1).standard_normal((3, fs * 10))
info = mne.create_info(["C3", "CZ", "C4"], fs, "eeg")
raw = mne.io.RawArray(data, info)

montage = mne.channels.make_standard_montage("standard_1020")
raw.set_montage(montage, match_case=False)
```

adam2392, 2020-10-18:
Hmm okay I tried reinstalling mne and it seems to work now. I might have had a corrupted file somewhere... Closing now.

closed 2020-10-18 by adam2392