mne-tools/mne-python #8385: Error during read of Nihon file

tharikjose, 2020-10-19:
#### Describe the bug
I am facing an error during the reading phase of a Nihon EEG file. The problem is during the reading of the .21E file. The EEG is not corrupted since it can be opened in the Nihon program.  

#### Steps to reproduce
import mne

raw = mne.io.read_raw_nihon('Nihon/Nihon_Salvo_em_Tres_Montagens/DA7521RO.EEG')

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
Expected to get the raw data of the EEG file.

#### Actual results
Loading DA7521RO.EEG
Found 21E file, reading channel names.
Traceback (most recent call last):
  File "C:/Users/Sorrino/Documents/Faculdade/EESC/01 - Mestrado/Reading EEG/Reading  Nihon.py", line 6, in <module>
    raw = mne.io.read_raw_nihon('Nihon/Nihon_Salvo_em_Tres_Montagens/DA7521RO.EEG')
  File "C:\Users\Sorrino\anaconda3\envs\mne_dev\lib\site-packages\mne\io\nihon\nihon.py", line 46, in read_raw_nihon
    return RawNihon(fname, preload, verbose)
  File "<decorator-gen-218>", line 24, in __init__
  File "C:\Users\Sorrino\anaconda3\envs\mne_dev\lib\site-packages\mne\io\nihon\nihon.py", line 311, in __init__
    header = _read_nihon_header(fname)
  File "C:\Users\Sorrino\anaconda3\envs\mne_dev\lib\site-packages\mne\io\nihon\nihon.py", line 126, in _read_nihon_header
    _chan_labels = _read_21e_file(fname)
  File "C:\Users\Sorrino\anaconda3\envs\mne_dev\lib\site-packages\mne\io\nihon\nihon.py", line 119, in _read_21e_file
    _chan_labels[idx] = name.strip()
IndexError: list assignment index out of range

Process finished with exit code 1

#### Additional information
Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\Sorrino\anaconda3\envs\mne\python.exe
CPU:           Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 8 cores
Memory:        7.9 GB

mne:           0.21.0
numpy:         1.19.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.0
matplotlib:    3.3.1 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.1.1
cupy:          Not found
pandas:        1.1.3
dipy:          1.2.0
mayavi:        4.7.2
pyvista:       0.26.1 {pyvistaqt=0.2.0, OpenGL 4.5.0 NVIDIA 451.48 via GeForce GTX 960M/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.1
[DA7521RO.zip](https://github.com/mne-tools/mne-python/files/5402744/DA7521RO.zip)

agramfort, 2020-10-19:
@fraimondo can you have a look?

fraimondo, 2020-10-20:
Hello, I checked the attached files.

The problem is not only with the 21E file which has 1096 channels defined (which I fixed easily).

This is a newer EEG system: EEG-1200A V01.00

Can you please provide an EDF file of the same recording exported with the Nihon Kohden software? I wan't to do some tests before pushing the fixes.

tharikjose, 2020-10-20:
I have the EDF file of another reading in which the same problem occurs. 
[DA7521RN.zip](https://github.com/mne-tools/mne-python/files/5410076/DA7521RN.zip)

fraimondo, 2020-10-21:
Just to be sure. In the ZIP file you sent there is a `DA7521NN.edf` and a `DA7521RN.EEG` file. Are you sure it's the right EDF? (I noticed the NN vs RN at the end).

If this is the case, then I don't have good news:

```IPython
In [13]: raw_edf                                                                                                                        
Out[13]: <RawEDF | DA7521NN.edf, 48 x 3200 (16.0 s), ~1.3 MB, data loaded>

In [14]: raw_nk                                                                                                                         
Out[14]: <RawNihon | DA7521RN.EEG, 1 x 200 (1.0 s), ~11 kB, data loaded>
```

Even with allowing the reader to process `EEG-1200A V01.00` files, the data is not correctly loaded. Without documentation or code that properly reads this kind of files, it will be a bit difficult to make it work.

It is an issue that the error message is not correct. I will create a PR for this.

closed 2025-10-23 by drammock