mne-tools/mne-python #8387: A question about "-trans.fif"

zhchsh19, 2020-10-19:
Sorry to bother all of you , but I have a big question.

[image]

but I can't find the file in 'bem' file. and I also get this error:
[image]

This is my code to get 'bem'

export FREESURFER_HOME=/home/zhanglingxi/freesurfer
source $FREESURFER_HOME/SetUpFreeSurfer.sh
 export SUBJECTS_DIR=/home/zhanglingxi/freesurfer2/subjects
my_subject=subdw
my_NifTI=/media/zhanglingxi/zlx-1/SMRI/NIF/s20180417_T09_LongZYL_ZHN_DaiWei-0002-00001-000144-01.nii.gz
recon-all -i $my_NIfTI -s $my_subject -all
subjects_dir=/home/zhanglingxi/freesurfer2/subjects
 mne watershed_bem -s subdw

And I used to get a ' fsaverage' in '/freesurfer2/subjects/sub' ,then I don't have the question above, but now I can't get it 
[image]

[image]

[image]

zhchsh19, 2020-10-19:
[s20180417_T09_LongZYL_ZHN_DaiWei-0002-00001-000144-01.nii.gz](https://github.com/mne-tools/mne-python/files/5402840/s20180417_T09_LongZYL_ZHN_DaiWei-0002-00001-000144-01.nii.gz)

This is my data

larsoner, 2020-10-19:
> And I used to get a ' fsaverage' in '/freesurfer2/subjects/sub' ,then I don't have the question above, but now I can't get it

You should not put `fsaverage` in `/home/zhanglingxi/freesurfer2/subjects/sub/fsaverage` -- if your `subjects_dir` is `/home/zhanglingxi/freesurfer2/subjects`, you should put it in `/home/zhanglingxi/freesurfer2/subjects/fsaverage`. `fsaverage` is a subject, it should be directly in your SUBJECTS_DIR, not in a subdirectory.

> and I also get this error:

The error message recommends running `mne make_scalp_surfaces` for your subject. Did you do this? If you did, you should have `SUBJECTS_DIR/SUBJECT/surf/lh.seghead` and `SUBJECTS_DIR/SUBJECT/bem/SUBJECT-head.fif`.

zhchsh19, 2020-10-19:
yes, I have done it, and I also have SUBJECTS_DIR/SUBJECT/surf/lh.seghead and SUBJECTS_DIR/SUBJECT/bem/SUBJECT-head.fif ,but there is still the same question

larsoner, 2020-10-19:
The coreg error image says it's trying to use `/home/zhanglingxi/freesurfer2/subjects/subdw` as the SUBJECTS_DIR, which is incorrect -- it should be `/home/zhanglingxi/freesurfer2/subjects/`. Can you try `mne coreg -d /home/zhanglingxi/freesurfer2/subjects`, or set `export SUBJECTS_DIR=/home/zhanglingxi/freesurfer2/subjects` and then run `mne coreg` and see if it works? Do you get the same error message as before but then with `/home/zhanglingxi/freesurfer2/subjects` as the path?

zhchsh19, 2020-10-19:
It works, I didn't set the path before, thanks very much

closed 2020-10-19 by larsoner