mne-tools/mne-python #8388: Plotting MNI transformed ECoG

timonmerk, 2020-10-19:
For decoding analysis across patients it becomes relevant to mainly work with MNI transformed coordinates. Using this tutorial https://mne.tools/stable/auto_tutorials/misc/plot_ecog.html, instead of the native FreeSurfer file the 'fsaverage' MNI brain can thus be used:

```
sample_path = mne.datasets.sample.data_path()
subject = 'fsaverage'
subjects_dir = sample_path + '\\subjects'

```

I installed MNE according to https://mne.tools/dev/install/mne_python.html, and using the `fsaverage` and run through the tutorial https://mne.tools/stable/auto_tutorials/misc/plot_ecog.html. 
At the line `mne.viz.set_3d_view(fig, 200, 70, focalpoint=[0, -0.005, 0.03])` I get the following error: 
"AttributeError: module 'mne.viz.backends._notebook' has no attribute '_set_3d_view'" 

My mne.sys_info() looks as follows: 
(base) C:\Users\ICN_admin\Documents\WRITE_BERLIN_BIDS>python -c "import mne; mne.sys_info()"
Platform:      Windows-10-10.0.18362-SP0
Python:        3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Users\ICN_admin\Anaconda3\python.exe
CPU:           AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD: 64 cores
Memory:        127.9 GB

mne:           0.21.0
numpy:         1.19.2 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.5.2
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.22.1
numba:         0.45.1
nibabel:       3.0.1
cupy:          Not found
pandas:        0.25.1
dipy:          Not found
mayavi:        4.7.2
pyvista:       0.26.1 {OpenGL 4.5.0 NVIDIA 432.00 via GeForce RTX 2070 SUPER/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.9.2

larsoner, 2020-10-19:
@GuillaumeFavelier can you look into the bug here?

A separate issue is how we should make it clear how to use MNI-coordinate sEEG/ECoG data. @adam2392 do you think it would make sense to take the tutorial idea I proposed in #8364 and make the dataset we use have electrode coordinates in MNI space? It might allow us to show how to use MNI-coordinate data and sEEG data in the same tutorial.

GuillaumeFavelier, 2020-10-20:
It seems that `_set_3d_view()` is not exposed in the `notebook` 3d backend so I opened https://github.com/mne-tools/mne-python/pull/8392

However, even using the standard `pyvista` backend, I got the following which I believe is a regression:

```py
...
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 21657  =      0.000 ...    21.670 secs...
Plotting 0 ecog locations
...
```

[image]

GuillaumeFavelier, 2020-10-20:
I tracked it to https://github.com/mne-tools/mne-python/pull/8190, let me know if you can reproduce @larsoner

larsoner, 2020-10-20:
#8190 just exposed a bug from #7747, see fix in #8393

adam2392, 2020-10-20:
> A separate issue is how we should make it clear how to use MNI-coordinate sEEG/ECoG data. @adam2392 do you think it would make sense to take the tutorial idea I proposed in #8364 and make the [...]

I've been a bit mIa but I will try to start the series of PRs sometime. Just a disclaimer I'm not too familiar with using freesurfer brains yet so might need help along the way. Will begin with say a tutorial on "loading in SEEG data" that more or less looks EXACTLY like the ecog example.

PR #8402 (2020-10-26): [WIP] Trying to allow plotting SEEG activity in volumetric source space
## Reference issue
Fixes: #8364 
Closes #8388

#### What does this implement/fix?
1. Adds a `plot_seeg.py` tutorial for working w/ SEEG data analogous to `plot_ecog.py`.

2. Attempts to refactor `stc_near_sensors` to show volumetric SEEG activity with either `stc.plot` or `stc.plot_3d`

#### Additional information
Opening up PR to start iterating.

Ideally want to allow SEEG activity to be visualized as such: https://mne.tools/dev/auto_tutorials/source-modeling/plot_beamformer_lcmv.html#volumetric-rendering-3d
Assumes user [... cut]
closed 2020-10-26 by larsoner