mne-tools/mne-python #8391: read_raw_curry() with preload=True  produces error

todflak, 2020-10-20:
#### Describe the bug
**read_raw_curry()** with **preload=True**  produces error (for some DAT files)

I am working with Curry version 7 data files, using **read_raw_curry()** to load data.  Out of 16 datasets that I am processing, 7 of them produce an error similar to this:

> RuntimeError: Incorrect number of samples (5302650 != 5379500), please report this error to MNE-Python developers

I also determined that the error **_only_** occurs using the method option  **preload=True**.  If I use **preload=False**, the data loads fine for all 16 datasets!

Note that the same problem was reported in [this comment](https://github.com/mne-tools/mne-python/issues/7300#issuecomment-584300596) to issue 7300 .  At that time, there was no clear cause for this error.

#### Steps to reproduce
This is a link to Curry data files (.dat, .rs3, .dap)  for reproducing the error:
https://drive.google.com/file/d/1842rPYAf_LkYF9hQxkKjsMz4MY240m3r/view?usp=sharing

test code:
```
import mne
data_path = 'C:\\read_curry example data\\PH005_baseline.dat'

raw = mne.io.read_raw_curry(data_path,preload=False,verbose=True)
print('Data loaded, with preload=False: ' + data_path +  ', ' + str(raw.info["nchan"]) + ' channels X ' + str(raw.n_times) + ' points')

raw = mne.io.read_raw_curry(data_path,preload=True,verbose=True)
print('Data loaded, with preload=True: ' + data_path +  ', ' + str(raw.info["nchan"]) + ' channels X ' + str(raw.n_times) + ' points')

```

#### Expected results
Would expect to load correctly with either "preload=False" or "preload=True", which would produce this output:

```
Leaving device<->head transform as None (no landmarks found)
Event file not found. No Annotations set.
Data loaded, with preload=False: C:\read_curry example data\PH005_baseline.dat, 70 channels X 76850 points
Leaving device<->head transform as None (no landmarks found)
Reading 0 ... 76849  =      0.000 ...   307.396 secs...
Event file not found. No Annotations set.
Data loaded, with preload=True: C:\read_curry example data\PH005_baseline.dat, 70 channels X 76850 points
```

#### Actual results
First load (with  preload=False) works fine; second load (with  preload=True) throws error:
```
Data loaded, with preload=False: C:\read_curry example data\PH005_baseline.dat, 70 channels X 76850 points
Leaving device<->head transform as None (no landmarks found)
Reading 0 ... 76849  =      0.000 ...   307.396 secs...
Traceback (most recent call last):
  File "c:\Users\todfl\.vscode\extensions\ms-python.python-2020.2.64397\pythonFiles\ptvsd_launcher.py", line 48, in <module>
    main(ptvsdArgs)
  File "c:\Users\todfl\.vscode\extensions\ms-python.python-2020.2.64397\pythonFiles\lib\python\old_ptvsd\ptvsd\__main__.py", line 432, in main    
    run()
  File "c:\Users\todfl\.vscode\extensions\ms-python.python-2020.2.64397\pythonFiles\lib\python\old_ptvsd\ptvsd\__main__.py", line 316, in run_file
    runpy.run_path(target, run_name='__main__')
  File "C:\ProgramData\Anaconda3\lib\runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\ProgramData\Anaconda3\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\ProgramData\Anaconda3\lib\runpy.py", line 87, in _run_code
  File "c:\Users\todfl\Documents\Decker\EEG Processing\EEG Processing Code\read_raw_curry error MWE.py", line 7, in <module>
    raw = mne.io.read_raw_curry(data_path,preload=True,verbose=True)
  File "<decorator-gen-218>", line 20, in read_raw_curry
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\curry\curry.py", line 406, in read_raw_curry
    return RawCurry(fname, preload, verbose)
  File "<decorator-gen-219>", line 20, in __init__
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\curry\curry.py", line 439, in __init__
    super(RawCurry, self).__init__(
  File "<decorator-gen-160>", line 20, in __init__
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 247, in __init__
    self._preload_data(preload)
  File "<decorator-gen-164>", line 20, in _preload_data
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 528, in _preload_data
    self._data = self._read_segment(
  File "<decorator-gen-162>", line 20, in _read_segment
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 412, in _read_segment
    _ReadSegmentFileProtector(self)._read_segment_file(
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1816, in _read_segment_file
    return self.__raw.__class__._read_segment_file(
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\curry\curry.py", line 469, in _read_segment_file
    _read_segments_file(
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\utils.py", line 221, in _read_segments_file
    raise RuntimeError('Incorrect number of samples (%s != %s), '
  RuntimeError: Incorrect number of samples (5302650 != 5379500), please report this error to MNE-Python developers
```

#### Additional information
Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.3 (default, Jul  2 2020, 17:30:36) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\ProgramData\Anaconda3\python.exe
CPU:           AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD: 16 cores
Memory:        16.0 GB

mne:           0.21.0
numpy:         1.18.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.0
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.1
numba:         0.50.1
nibabel:       Not found
cupy:          Not found
pandas:        1.0.5
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

todflak, 2020-10-20:
Just a followup -- no surprise that the following code also produces the same error:
```
import mne
data_path = 'C:\\read_curry example data\\PH005_baseline.dat'

raw = mne.io.read_raw_curry(data_path,preload=False,verbose=True)
print('Data loaded, with preload=False: ' + data_path +  ', ' + str(raw.info["nchan"]) + ' channels X ' + str(raw.n_times) + ' points')
raw.load_data() 
```

The **raw.load_data()**  line throws the same error as described above:
`RuntimeError: Incorrect number of samples (5302650 != 5379500), please report this error to MNE-Python developers
`

agramfort, 2020-10-20:
can you look into it yourself? otherwise we'll see who can help

>

todflak, 2020-10-20:
Ah, you mean look into the code to find the problem _myself_... I hadn't considered doing that! ;-)
Sure, I will give it a shot!  I'm not an expert in Python dev, but I will have a go at it.

agramfort, 2020-10-20:
yes ! let us know if you need help

>

todflak, 2020-10-21:
Well, I've tracked down the problem, and fixed the code locally.  I will be making a pull request soon.

The difference between my datasets that loaded properly and those that failed relates to whether they denoted a reference channel or not.   For my example files, in all cases, there were 70 sensors in the acquisition:  66 EEG electrodes, 1 reference electrode, 3 "other" channels (VEO, EKG, Trigger).
Also, in ALL cases, in the DAP file (DAP = data parameter file), in the "DATA_PARAMETERS" section, the value of  NumChannels =  69.   (And I can tell from the file size that this is correct... it is always 69 channels saved in the data files.)

However, when I use read_raw_curry() with preload=False (to just load the information), in some cases it reports raw.info["nchan"]=69, and in other cases it reports raw.info["nchan"]=70.

In either case, there are actually only 69 channels saved in the file.  It seems to me that Curry DOES NOT save the Reference electrode data into the file.  But in the second case, the "Ref" channel was listed as one of the channel labels, and the MNE code was counting the "Ref" channel as one of the saved data channels and expecting it to be represented in the data file. In that case, the file seemed to be too short, thus causing the error denoted in my initial comment.

For the datasets that load properly, in the *.rs3 file (sensor geometry file, or the labels file), it lists 66 normal EEG channel labels, plus the 3 "other" channels.
For the datasets that fail to load properly, in the *.rs3 file, it lists 67 EEG channel labels (including the "Ref"), plus the 3 "other" channels.

As to why some of the datasets listed the "Ref" channel and some did not, I'm not sure about that ... I am not very familiar with data acquisition in the Curry system, but I assume there is some option that a user must select to define the reference channel and maybe sometimes they failed to do so, resulting in some datasets that include that channel in the list and some that don't.

For the datasets that load properly, in the corresponding DAP files, the "DEVICE_PARAMETERS" section looks like this:
DEVICE_PARAMETERS START
   DataUnit             = uV
   NumChanThisGroup     =  66
   NumSensorsThisGroup  =  66
   CommonOffset         =  0
   NumRefChannels       =  0
   CommonScale          =  1
DEVICE_PARAMETERS END

while for other DAP files (the datasets that fail to load), the "DEVICE_PARAMETERS" section looks like this:
DEVICE_PARAMETERS START
   DataUnit             = uV
   NumChanThisGroup     =  67
   NumSensorsThisGroup  =  67
   CommonOffset         =  0
   NumRefChannels       =  1
   CommonScale          =  1
DEVICE_PARAMETERS END

I also notice in the DAP file a section called  "CHAN_IN_FILE", which seems to list the index in the data file for each of the channels. (That section seems to be present only in the datasets in which the "Ref" channel is listed in the "labels" rs3 file, but not saved into the datafile). It is basically just the numbers 1...66 in order, and for the 67th position, it shows the value as "0", which I suppose indicates that this channel (the "Ref" channel) is not actually present in the datafile.

To resolve the problem, I revised code in "curry.py".  It now examines the "CHAN_IN_FILE" sections in the DAP file (if those sections are present), and then for each channel it checks whether the channel index is greater than 0; if it is 0, it simply bypasses that channel in the channel list.

Just for future reference, I am attaching a ZIP with four files that show examples of the two cases:

- **no_ref.rs3** and **no_ref.dap**: these show the situation of no "Ref" channel; these types of datasets were loading fine previously.
- **includes_ref.rs3** and **includes_ref.dap**: these show the situation of having a"Ref" channel listed; these types of datasets were previously failing to load.

[example DAP and RS3 files.zip](https://github.com/mne-tools/mne-python/files/5412452/example.DAP.and.RS3.files.zip)

agramfort, 2020-10-21:
great you find your way. Please open a PR and we'll discuss there.

We'll see then how to add tests etc.

PR #8400 (2020-10-23): MRG FIX read raw curry
#### Reference issue
Fixes #8391 
Fixes #8398

#### What does this implement/fix?
Fixes problem loading some Curry data files in which a Reference electrode is denoted in the list of channel labels, but that channel is not actually present in the data file.  Revised code examines the info file (DAP file for Curry 7 format), looking in CHAN_IN_FILE section for channels that are not actually saved into the corresponding raw data file.

While I was at it, I added code to populate info['meas_date'] property with the acquisition start time, which is extracted from the info fi [... cut]
commit f4077c05 (2020-10-23): MRG FIX read raw curry (#8400)

* Resolve problem loading Curry7-format data files that include an unsaved Reference channel.
Also ensures that the channel order that may be explicitly declared in the DAP file is respected when loading the data.
fixes #8391

* Reading acquisition start time from Curry DAP file, populating the info['meas_date'] property.
resolves #8398

* Fixing style issues

* More style fixes

* More style fixes

* Final style commit, I hope!
I finally learned how to use flake8 linter in my IDE!

* Deal with files that do not have acquisition start time information available. [... cut]
closed 2020-10-23 by larsoner