mne-tools/mne-python #8401: BUG: _Brain has no attribute set_time

drammock, 2020-10-21:
#### Describe the bug
3D viz with pyvista does not have a `set_time` method (which for me at least is fairly important for *non*-interactive plotting, e.g., generating a bunch of static figures for each subject at t=0.5 or whatever).

#### Steps to reproduce
```python
import os
import mne

mne.viz.set_3d_backend('pyvista')
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=10)
raw.set_eeg_reference(projection=True)
events = mne.find_events(raw, stim_channel='STI 014')
event_dict = {'auditory/left': 1, 'auditory/right': 2}
epochs = mne.Epochs(raw, events, event_id=event_dict, preload=True)
evoked = epochs.average()
inverse_operator_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                     'sample_audvis-meg-oct-6-meg-inv.fif')
inv_operator = mne.minimum_norm.read_inverse_operator(inverse_operator_file)
stc = mne.minimum_norm.apply_inverse(evoked, inv_operator)
brain = stc.plot(subject='sample')
brain.set_time(0.5)
```

#### Expected results
the figure would update to show activity at time = 0.5 seconds

#### Actual results
`AttributeError: 'Brain' object has no attribute 'set_time'`

#### Additional information
ping @GuillaumeFavelier.  Maybe you already have this on your TODO list? (if so feel free to close this issue)... but I didn't find anything when searching open issues for `set_time` so I thought I'd mention in case you had missed it.

GuillaumeFavelier, 2020-10-21:
Did you try `set_time_point()`?

drammock, 2020-10-22:
nope. I was re-running an old script that had worked on 0.20.x (when mayavi was default) and failed on 0.21 (when pyvista is default)...  Testing now on master, I'm seeing that if the backend is set to mayavi, then only `set_time` works, and if set to pyvista, only `set_time_point` works.  The API should be backend-agnostic.

larsoner, 2020-10-22:
Agreed we should have `set_time`, I've had the same issue. @drammock can you open a PR to add it to `Brain`, then modify some test to use `set_time` instead of `set_time_point`?

drammock, 2020-10-22:
>  @drammock can you open a PR to add it to `Brain`, then modify some test to use `set_time` instead of `set_time_point`?

sure, I'll tackle that later today or tomorrow morning.

PR #8415 (2020-10-24): ENH: add Brain.set_time(time_in_seconds)
closes #8401
closed 2020-10-24 by agramfort