mne-tools/mne-python #8407: 'Time' text is not in the view when saving a topomap plot animation - (using animate_topomap)

2010019970909, 2020-10-22:
#### Describe the bug
I was trying to save a topomap as described in the [tutorial](https://mne.tools/stable/auto_examples/visualization/plot_evoked_topomap.html#animating-the-topomap). Apparently the title (where the time is displayed) is set via the `ax.text` method ([here](https://github.com/mne-tools/mne-python/blob/4eb2997d932453a4aea4278bef1cd878aa47f085/mne/viz/topomap.py#L2185)). I tend to believe that  using `tight_layout` somewhere makes it out of the view (only when saving the animation).

##### Other remarks/propositions
I would find it nice to be able to remove the ticks from the color bar and also to remove the title of the colormap (maybe even remove the color map).

For the time, I'll find it interesting to be able to move it, set the font, the position, remove it, etc.

Those are just proposition, I'm aware about the among of work needed to achieve the implementation of thoses features and I don't have myself the time to implement them. Thanks for your work guys.

#### Steps to reproduce
Try `show = True` and `show = False`. And change `fontsize` when `show` is true.

```python
# -*- coding: utf-8 -*-
import numpy as np
import matplotlib.pyplot as plt

from mne.datasets import sample
from mne import read_evokeds

# If True show plot, otherwise, save plot
show = False
recording_filepath = "saving file"

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'

# load evoked corresponding to a specific condition
# from the fif file and subtract baseline
condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))

# sphinx_gallery_thumbnail_number = 9
times = np.arange(0.05, 0.151, 0.01)
fps = 2

fig, anim = evoked.animate_topomap(
            times=times, ch_type=None,
            frame_rate=fps, time_unit='s', 
            blit=False, show=show,
            butterfly=False)

# Add another title
ax = fig.add_subplot(111)
ax.get_yticks()
ax.xaxis.set_ticks([])
ax.yaxis.set_ticks([])
ax.set_frame_on(False)
ax.set_title(
    '%s (%s)\n%s, %s' % (
        "A filename",  # Title
        0,             # Event index
        "Information", # Some information
        r"{info}"),    # More information
        fontsize=12)   # <-- used trick to show the time (fontsize=35)

# Adapt the figure size
fig.set_size_inches(16, 9, forward=True)

if not show:
    anim.save('%s_%s.mp4' % (recording_filepath, str(0)), dpi=300)
    plt.close(fig)
input("Close ")
plt.close(fig)

```

#### Expected results
Seeing all the the elements seen when showing the plot.

#### Actual results
The time is not in the view when saving the animation.

#### Additional information
```bash
Windows fatal exception: code 0xc0000139

Current thread 0x000047ec (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1101 in create_module
  File "<frozen importlib._bootstrap>", line 556 in module_from_spec
  File "<frozen importlib._bootstrap>", line 657 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1101 in create_module
  File "<frozen importlib._bootstrap>", line 556 in module_from_spec
  File "<frozen importlib._bootstrap>", line 657 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "C:\Users\vince\AppData\Local\Programs\Python\Python38\lib\site-packages\mne\utils\check.py", line 626 in _check_pyqt5_version
  File "C:\Users\vince\AppData\Local\Programs\Python\Python38\lib\site-packages\mne\utils\config.py", line 567 in sys_info
  File "<stdin>", line 1 in <module>
Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:57:54) [MSC v.1924 64 bit (AMD64)]
Executable:    C:\Users\vince\AppData\Local\Programs\Python\Python38\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        15.9 GB

mne:           0.21.0
numpy:         1.19.2 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.5.2
matplotlib:    3.1.3 {backend=TkAgg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.1.1
cupy:          Not found
pandas:        0.24.2
dipy:          Not found
mayavi:        4.7.2
pyvista:       0.26.1 {OpenGL 4.5.0 - Build 27.20.100.8476 via Intel(R) UHD Graphics 620}
vtk:           9.0.1
PyQt5:         unknown

```

larsoner, 2020-10-23:
PR(s) welcome to add and improve this function. Hopefully someone hits a use case that motivates fixing the issues soon

larsoner, 2023-10-01:
Closing for https://github.com/mne-tools/mne-python/issues/9018 which discusses what I think is the same issue. It's also possible both would be fixed by #12024

closed 2023-10-01 by larsoner