mne-tools/mne-python #8423: plot_sensors(kind='select') does not fill the list with selected channel names

mmagnuski, 2020-10-25:
For some reason interactive channel selection in `plot_sensors(kind='select')` does not update the returned list of selected channels.

### Reproducible example
Confirmed this on latest master:
```python
import mne
import os
from mne.datasets import multimodal

fname_raw = os.path.join(multimodal.data_path(), 'multimodal_raw.fif')
raw = mne.io.read_raw_fif(fname_raw)
```
now in interactive mode perform:
```python
fig, sel = raw.plot_sensors(kind='select')
```
select a few channels and see what happens to `sel` list - its empty.

drammock, 2020-10-26:
What you describe is true in 0.20.8 and 0.21 as well as current master.  This works as expected:

```python
fig, sel = raw.plot_sensors(kind='select', block=True)
```

In light of that, what do you propose we should change?  Some ideas:

1. automatically set `block=True` when `kind == 'selection'`
2. better documentation that `sel` will always be an empty list when `block=False`
3. something else?

mmagnuski, 2020-10-26:
Maybe this is specific to the notebook too - I never had to do `block=True` with `kind='select'` previously (in the notebook). I can take a look later today to check which version it worked in, but I think for 0.21 is works ok in the notebook.

drammock, 2020-11-02:
@mmagnuski anything to be done here?

From my above numbered options, after some thought I'm actually opposed to option 1, given that `plot_sensors` is now called from `plot_raw()` when channel names are right-clicked, and how I used `kind='select'` to get the correct sensor highlighted.  Admittedly that was a bit of a hack (setting it up as a selectable sensor plot, programmatically selecting the sensor to be highlighted, then disconnecting the interactivity before the figure is drawn), but unless/until `plot_sensors` is refactored, option 1 above would probably break some things.

mmagnuski, 2020-11-02:
Ok, so currently the old behavior, where the list would update without blocking is not possible?
Then I'd consider throwing a warning when `kind='select'` and `block=False`.

drammock, 2020-11-02:
> Ok, so currently the old behavior, where the list would update without blocking is not possible?

Well, you can still access it via `fig.lasso.selection`.  But actually, after digging around in the code, I think I know what changed. Previously I think the original `list` object created during init of the lasso selector was preserved throughout. That is no longer happening; in the `on_select` and `select_many` methods it's now getting rewritten instead of just appended, popped, etc.  I'll see if I can fix it.

commit 7a87d0c6 (2020-11-05): closes #8423 (#8469)
closed 2020-11-05 by larsoner