mne-tools/mne-python #8424: both show=False and show=True in plotting functions leads to two figures in a notebook

mmagnuski, 2020-10-25:
I'm pretty sure some time ago plotting in a notebook with inline matplotlib turned on gave two figures only with `show=False`.
For example: `raw.plot()` would give you two identical figures, but `raw.plot(show=False)` would give just one.
Currently both `show=True` and `show=False` lead to two figures in the notebook. Its not a problem given that it can be suppressed with `;` but still may be worth fixing.

larsoner, 2020-10-26:
Probably for @drammock in the refactor?

drammock, 2020-10-26:
I can confirm that in jupyter lab, ending a cell with `raw.plot()` yields 2 figs, `raw.plot(show=False)` also (incorrectly) yields 2 figs, and `fig = raw.plot()` or `raw.plot();` each yield 1 fig.  In my testing today, all this is also true of versions 0.20.8 and 0.21.0

In a regular ipython console, `raw.plot(show=False)` seems to be working correctly on current master (figure is not shown; subsequent `plt.show()` will show it).  So, whatever is causing the undesired notebook behavior when `show=False` is not due to the recent refactor.  That doesn't mean it's not a bug, but I'm not really sure how to fix it since it seems to be backend-specific and I have little experience tinkering with / working around the limitations of the notebook backend.  So I will treat this one as low priority unless someone else volunteers, or at least offers me some pointers on how to approach this / what a solution might look like.

larsoner, 2020-10-26:
One idea is that Jupyter notebooks treat non-assignment statements in a special way. In other words, this:
```
fig = raw.plot(...)
fig
```
is functionally the same as
```
raw.plot(...)
```
i.e., Jupyter will look at the non-assignment in the last line. This is important because it then tries to do `_repr_html_` on the resulting object. So it's possible that with the refactor the output of `raw.plot` being different (assuming it's not just a plain, standard, `matplotlib.pyplot` engine-managed figure) could be causing the notebook backend to behave incorrectly somehow.

drammock, 2020-10-26:
> So it's possible that with the refactor the output of `raw.plot` being different (assuming it's not just a plain, standard, `matplotlib.pyplot` engine-managed figure) could be causing the notebook backend to behave incorrectly somehow.

I don't think that explanation can be right.  As I said above, the behavior @mmagnuski describes happens in 0.20.8 and 0.21.0 as well as in current master.  And the new version of `raw.plot()` figures are instances of a subclass of `matplotlib.figure.Figure` (instantiated by `matplotlib.pyplot.figure(..., FigureClass=SomeCustomMNESubclass)`), and are still managed by whatever `FigureManager` is provided by the backend.  It's not impossible that I've done something to muck up how figures interact with the notebook backend, but so far I'm not seeing evidence that this was caused by #7955.

drammock, 2020-11-09:
Here's a screenshot showing what I think is the real problem here (and showing that it is not related to recent changes to the figure classes).  The `show=False` argument simply does not work with matplotlib inline backend.

[image]

mmagnuski, 2020-11-09:
@drammock You may be right that it never worked, I might have been using `;` to suppress dual figures all the time. I'll check tomorrow with some ancient mne version :)

mmagnuski, 2020-11-09:
@drammock However, I must say I've seen multiple instances where `show=False` does take an effect in inline mode. For example when I create a topomap and then modify the figure somehow - without using the `show=False` the modifications would not be visible. For example:
```
n_chs = len(info['ch_names'])
act = np.random.randn(n_chs)
img, _ = mne.viz.plot_topomap(act, info, show=False)
ax = img.axes
chans = ax.findobj(mpl.collections.PathCollection)[0]
chans.set_sizes([22] * n_chs)
```
you can see the channel sizes changed only with `show=False`.
In general the inline backend tends to show figures that were created in the cell above. But that is expected, my problem was related to the fact that both with and without `show=False` I would get two figures in inline backend: one created by mne and one created by notebook auto-showing figures.

drammock, 2020-11-10:
> However, I must say I've seen multiple instances where `show=False` does take an effect in inline mode.

I can reproduce your topomap example, and I agree that it shows that `show=False` does have an effect in that it *delays* the figure being shown until after all code in the cell has been run.  Which is better than nothing.  But ideally, I would think that passing `show=False` should prevent the figure from appearing at all (that's what I meant in saying "it doesn't work").

mmagnuski, 2020-11-11:
Yes, in the example you show, I would expect `show=False` not to show anything. But the specific issue that I mention in the first post is just:
[image]
inline automatically plots the last output (if not captured to a variable - that is ok and expected), and mne plots the figure automatically too (even with `show=False` which is not expected), so I get two figures, even if I specify `show=False`.
It's mostly annoying when teaching because I have to addtionaly explain why two figures show up instead of focusing on what is in the figure.

mmagnuski, 2020-11-11:
@drammock but you seem to be correct that `show=False` never worked according to my expectations. I can't go back to the old versions of mne because I get errors when plotting then, but the issue clearly existed before the mne figure refactor.

drammock, 2020-11-11:
It's possible I suppose that the behavior of notebooks is what changed. Regardless, the specific issue you mention (where `raw.plot(show=False)` alone in a cell yields 2 figures) is a manifestation of the same problem I've been pointing out: using `show=False` does not prevent the figure, it merely delays showing it until the rest of the code in the cell has been run.  And then, since you neither assign nor use a semicolon, you get both the delayed figure and the `__repr_html__` version of the figure "returned" from the cell.

So, OK to close this?  Everything I've read online suggests that

1. returning the figure instance is what MPL devs recommend
2. it's an intentional design decision of the `inline` backend to both show every open figure created in a cell (in addition to showing the "return value" of the last line of the cell, which is built into ipython but can be turned off in settings)
3. using an assignment or semicolon on the last line is the officially recommended workaround.

mmagnuski, 2020-11-11:
@drammock You are right, I didn't realise the inline plots figures created within the cell, although I clearly experienced it many times. I thought the two-figure issue was mne's fault, but you are right, it is not. I'm closing then!

closed 2020-11-11 by mmagnuski