mne-tools/mne-python #8425: Default topomap interpolation a bit uglier than some time before

mmagnuski, 2020-10-25:
I noticed that current topomap interpolation with `extrapolate='head'` produces more artifacts at the edges than in the previous version. Take a look at this:
### before
[image]

### now
[image]

I think I may be partly responsible for this. I have a vague memory of adding more extrapolation points some time ago to `extrapolate='head'`, but I can't find the PR now.
I'll try to take a look at this in the upcoming week.

larsoner, 2020-12-01:
@mmagnuski we're making a push to release in a couple of weeks, do you have time to look into this in that time?

mmagnuski, 2020-12-02:
I might have some time for this next week, I'll try to look into it!

mmagnuski, 2020-12-06:
I still remember about this issue, I might find some time for it tomorrow.

mmagnuski, 2020-12-10:
I finally got to it. It turns out my intuition was correct, plot_topomap creates too many extrapolation points:
[image]
you can also see that the whole circle was not divided evenly (the red arrow), which was the cause of the funny extrapolation artifacts on the right side of the topos.
I reduced the number of points in the PR (#8645) and fixed the uneven circle division:
[image]

It is better, but still could be improved (left: master, right: PR)
[image]

I'll see how increasing the distance of extrapolation points and further reducing their number helps, but I would prefer not to overfit to my specific cap layout :), so I would appreciate if other EEG users that experienced this problem let me know.

`extrapolate='box'` works good in these cases so increasing the distance between the extrapolation points and channels positions should help.

agramfort, 2020-12-11:
PR on the way?

I would be great if the default could be the best option... just saying

>

mmagnuski, 2020-12-11:
@agramfort Yes, its in progress, I linked it in the post above (#8645).

PR #8645 (2020-12-16): MRG, FIX: prettier "head" extrapolation
Fixes #8425
closed 2020-12-16 by larsoner