mne-tools/mne-python #8432: How to apply ica on brainvision raw data

arthurlgk, 2020-10-26:
I am getting this error when trying to fit the data:

ValueError: picks (NoneNone, treated as "data") yielded no channels, consider passing picks explicitly

How to resolve this issue?

hoechenberger, 2020-10-26:
Hello @arthurlgk, we'd need a little more information how you got there.

- Which version of `mne-python` are you running?
- Could you share the code that leads to this problem?

Thanks!

arthurlgk, 2020-10-26:
I am currently using mne 0.21.0.
And the code is the following:
```
fname = "demo_3.vhdr"
bv_raw = mne.io.read_raw_brainvision(dirname+fname)
ica = ICA(n_components=63, random_state=97)
ica.fit(bv_raw)
```
Thank you.

hoechenberger, 2020-10-26:
@arthurlgk Ok, this seems a little odd to me (the behavior you're seeing, not your code, which looks ok) @cbrnr Maybe you have an idea?

@arthurlgk Could you (privately) share with me one of the datasets that produces this problem? So I can try look into this myself. My email address is richard.hoechenberger@gmail.com

arthurlgk, 2020-10-27:
@hoechenberger Sent. Did you receive it?

cbrnr, 2020-10-27:
Not sure, but maybe the channel types are not EEG/MEG? @arthurlgk can you post the output of `bv_raw.info`?

hoechenberger, 2020-10-27:
> @hoechenberger Sent. Did you receive it?

Thank you, I received the email. Will look into this within the next 2 hours.

hoechenberger, 2020-10-27:
@arthurlgk I just looked at the `.vhdr` file you included, and there's something off.

You file looks something like this:
```
...
Ch1=Channel 1,,1,
Ch2=Channel 2,,1,
Ch3=Channel 3,,1,
...
```

while it's supposed to look like this:
```
...
Ch1=FP1,,0.5,µV
Ch2=FP2,,0.5,µV
Ch3=F3,,0.5,µV
...
```
i.e., there are no units specified in your file. I'm not sure if that's even valid (but I'm not an expert on BV data). How did you generate those files?

hoechenberger, 2020-10-27:
@sappelhoff Maybe you would know which unit is the default if not explicitly specified in the `vhdr` file? µV, I suppose?

sappelhoff, 2020-10-27:
yep, seems to me like that.

pdf source: https://www.brainproducts.com/filedownload.php?path=products/more/BVCDF-1.0_DocVersion-2.5_2019-12-12.pdf

See screenshot:

[image]

hoechenberger, 2020-10-27:
Yes just found out the same. Thanks for looking. I will now try to figure out whether `pybv` or MNE is doing something wrong when reading @arthurlgk's file.

hoechenberger, 2020-10-27:
… just to realize that pybv only **writes**, apparently ;) So it must be something in MNE

hoechenberger, 2020-10-27:
Ok found it. Working on a fix.

sappelhoff, 2020-10-27:
> … just to realize that pybv only writes, apparently

yes :-) better to make the MNE reader great, than have a second and less complete/battle-tested version in pybv.

hoechenberger, 2020-10-27:
(strictly speaking, @abramhindle's file do not follow the specification. Because as I read the specs: if you don't have a unit, also the comma separating unit from the rest of props must go. but it's easy to deal with this, so… still it would be interesting to know which software @abramhindle used to create this file, because either it's buggy or the specs need to be amended)

hoechenberger, 2020-10-27:
(Am I right? Or am I reading this incorrectly? it says `...[,<unit>]`, so the comma should only be there if there's a unit, right?)

sappelhoff, 2020-10-27:
> Am I right?

I agree. it might be a typo :-/ coul be worth looking into how they handle the remaining *optional* trailing fields ... if they all go away including the comma, it may be legit.

To me it'd make much more sense to leave the comma in any case (even when unit is empty)

sappelhoff, 2020-10-27:
We have the same for "markers" -> the last (trailing) optional parameter seems to remove the final comma if it's not present:

[image]

cbrnr, 2020-10-27:
I wouldn't trust these specs too much but try to catch as many cases as possible (so in this case I'd try to handle a trailing comma even though it's technically invalid).

hoechenberger, 2020-10-27:
> We have the same for "markers" -> the last (trailing) optional parameter seems to remove the final comma if it's not present:

Ok so this seems to be a conscious decision in the specs, then.

sappelhoff, 2020-10-27:
I am still interest in how @arthurlgk obtained the file

hoechenberger, 2020-10-27:
> I am still interest in how @arthurlgk obtained the file

The header they sent me starts with

```
Brain Vision Data Exchange Header File Version 1.0
; Data created by the Vision Recorder
```

I wonder which version of BV Recorder that was.

closed 2020-10-27 by cbrnr

hoechenberger, 2020-10-27:
@arthurlgk This problem has now been fixed in the development version of MNE-Python. The fix will be included in the next bugfix release, for which we haven't determined a release date yet.

arthurlgk, 2020-10-27:
@hoechenberger Thank you very much. Is it possible for me to access this version now?

hoechenberger, 2020-10-27:
> @hoechenberger Thank you very much. Is it possible for me to access this version now?

Yes. How did you intall MNE?

Also: I'd still be interested to know how you got your hands on those files :) Do you know which version of BV recorder was used?

arthurlgk, 2020-10-27:
> > @hoechenberger Thank you very much. Is it possible for me to access this version now?
> 
> Yes. How did you install MNE?
> 
> Also: I'd still be interested to know how you got your hands on those files :) Do you know which version of BV recorder was used?

I used pip install in the terminal. Could you please tell me how to install the development version? Thank you.
The file is given from my professor. Because now I have no access to the lab computer, I will ask him about that.

hoechenberger, 2020-10-27:
> I used pip install in the terminal.

Ok, it's easy then. I suggest you install our maintenance branch (which will soon become 0.21.1) with my fix. To do that, simply run:

```
pip install -U https://github.com/hoechenberger/mne-python/archive/backport-8434.zip
```

> The file is given from my professor. Because now I have no access to the lab computer, I will ask him about that.

That would be great! Thank you

arthurlgk, 2020-10-27:
> Ok, it's easy then. I suggest you install our maintenance branch (which will soon become 0.21.1) with my fix. To do that, simply run:
>
> ```
> pip install -U [...]

It is working now. Thank you very much.
But when I am trying to run plot_component function, it cannot find electrode locations. Is this a problem with my recorder?

```
WARNING:root:Did not find any electrode locations (in the info object), will attempt to use digitization points instead. However, if digitization points do not correspond to the EEG electrodes, this will lead to bad results. Please verify that the sensor locations in the plot are accurate.
RuntimeError: No digitization points found.
```

hoechenberger, 2020-10-27:
> But when I am trying to run plot_component function, it cannot find electrode locations. Is this a problem with my recorder?

No, it's not a problem with your recorder, but you need to tell MNE which montage (i.e., which cap) was used when recording. Since it's a BrainProducts system, I assume you used some kind of EasyCap or ActiCap. You need to ask your professor which cap they used specifically. Once you have that info, we can show you how to tell MNE about it.

arthurlgk, 2020-10-27:
> No, it's not a problem with your recorder, but you need to tell MNE which montage (i.e., which cap) was used when recording. Since it's a BrainProducts system, I assume you used some kind of EasyCap [...]

 Oh it’s actiCap

hoechenberger, 2020-10-27:
You'd still need to find out the specific **type** of ActiCap.

But I can share with you the template montage I generated for what we used in our old lab.

You can load the montage and visualize it:
```python
fname_montage = base_dir / 'achtiCHamp_64_channels_and_fiducials_Theta_Phi.txt'
dig_montage = mne.channels.read_custom_montage(fname_montage)
dig_montage.plot()
```

When you assign it to your raw data:
```python
raw.set_montage(dig_montage)
```
you will get:
```
ValueError: DigMontage is only a subset of info. There are 64 channel positions not present in the DigMontage. The required channels are:
...
```
This is because in your VHDR file, the channels are named `Channel 1` etc., which doesn't match the names in the montage file (proper channel locations like `FP1`, `Oz`). So that's something you'll need to fix in your input data, otherwise there's no way to know **where** those channels used to be located on a participant's head.

[achtiCHamp_64_channels_and_fiducials_Theta_Phi.txt](https://github.com/mne-tools/mne-python/files/5447897/achtiCHamp_64_channels_and_fiducials_Theta_Phi.txt)

arthurlgk, 2020-11-03:
Hi @hoechenberger , so we were using pycorder with python 2.7 to record the data. 
For the montage plot, we have our own stp file. In 3d, it looks fine. But in the 2D plot, it looks like they are congregated and off center. Do I need to worry about it? Is this issue caused by the head _size?

hoechenberger, 2020-11-03:
> so we were using pycorder with python 2.7 to record the data.

Good to know, thanks!

> For the montage plot, we have our own sfp file. In 3d, it looks fine. But in the 2D plot, it looks like they are congregated and off center. Do I need to worry about it? Is this issue caused by the head _size?

The off-center positioning shouldn't be related to the head size. It's really hard to tell what's going on without looking at the montage file. Could you share it here?

arthurlgk, 2020-11-03:
@hoechenberger It doesn't support sfp file here. So I shared the file with you via email. Thank you for your help.

PR #8434 (2020-10-27): [MRG] BUG: Fix BrainVision reading for certain vhdr files
Fixes #8432

I need advice on how to best add a test (specifically, where to add new testing data. Essentially, I can re-use an existing data set and simply drop the "µV" units from the VHDR file)
commit 87553ab4 (2020-10-27): [MRG] BUG: Fix BrainVision reading for certain vhdr files (#8434)

* Fix BrainVision reading for certain vhdr files

Fixes #8432

* Update changelog

* Drop comma

* Make it more robust

* Typo

* Comment style

* Better comments

* Typo

* Add test

* start with lower case to make @cbrnr happy 👍

* Rework test
commit 2690cc0a (2020-10-27): [MRG] BUG: Fix BrainVision reading for certain vhdr files (#8434) (#8437)

* Fix BrainVision reading for certain vhdr files

Fixes #8432

* Update changelog

* Drop comma

* Make it more robust

* Typo

* Comment style

* Better comments

* Typo

* Add test

* start with lower case to make @cbrnr happy 👍

* Rework test