mne-tools/mne-python #8436: "If comment is empty, a 1-based index is assigned by Richard Höchenberger" – What's New entries

hoechenberger, 2020-10-27:
Hello, I just wanted to inquire if we could, maybe, agree to always add a comma before the contributors' names in our changelogs? Otherwise we keep producing sometimes confusing, somethings rather hilarious lines like the one in the title.

The latest changelog(s) have been rather inconsistent in that manner, although my impression is that more and more of those commas have made it through the review process.

So what I'm proposing is that instead of:

> If comment is empty, a 1-based index is assigned by Richard Höchenberger

we'd add an entry

> If comment is empty, a 1-based index is assigned, by Richard Höchenberger

WDYT?

cc @cbrnr @agramfort @sappelhoff @larsoner @drammock

agramfort, 2020-10-27:
+1 for ,

>

sappelhoff, 2020-10-27:
> +1 for ,

agreed

larsoner, 2020-10-27:
I'm happy with that, or switch to `(by whoever, :gh:whatever)`

drammock, 2020-10-27:
I am also pro-comma, and have recently been asked (though I forget by who) to remove them in my PRs for the sake of consistency. So I would like this policy change to be documented somewhere.

hoechenberger, 2020-10-27:
> and have recently been asked (though I forget by who) to remove them in my PRs for the sake of consistency with other people's entries.

Might have been me 😁

cbrnr, 2020-10-27:
I'm 👍 for `(by whoever, :gh:whatever)` - it is easier to see the contributor when there are parens.

hoechenberger, 2020-10-27:
> I'm 👍 for `(by whoever, :gh:whatever)` - it is easier to see the contributor when there are parens.

I like this proposal too, but would prefer it without the `by`. I'm +1 for `,` and +1.75 for `(My Name, #foo)`

larsoner, 2020-10-27:
Works for me

drammock, 2020-10-27:
me too. But again, please document the new standard in our contributing docs.  Since the `latest.inc` gets reset every release, it's too easy for the first couple of PRs to do it their own way and set a precedent for the remainder of the release cycle.

cbrnr, 2020-10-27:
Yes, I like it!

hoechenberger, 2020-10-27:
So... Which one now? Comma? Or the parentheses? I'm confused, and it's my own fault haha

drammock, 2020-10-27:
let's go with
```
(`Contributor Name`_, `Second Contributor`_ and `Third Contributor`_, :gh:`0000`)
```

cbrnr, 2020-10-27:
Yes, but let's use an Oxford comma please.

PR #8442 (2020-11-10): [MRG] DOC: Add new changelog template to contributors' guide
Also touch two changelog entries to see what it looks like when rendered.

Closes #8436
commit 157a0409 (2020-11-10): [MRG] DOC: Add new changelog template to contributors' guide (#8442)

* Add new changelog template to contributors' guide [skip travis]

Fixes #8436

* Typo [skip travis]

* Update following @cbrnr's propoal [skip travis]

* Fix changelog reference [skip travis] [skip github]

* Explain how to add name to names.inc [skip travis] [skip github]

* Add info about picking the correct subsection [skip travis] [skip github]

* Add guidance for first-time contributors [skip travis] [skip github]

* Hopefully fix doc build [skip travis] [skip github]

* Update stylesheet [skip travis] [skip github]

* [... cut]
closed 2020-11-10 by larsoner