mne-tools/mne-python #8441: Raw.plot() issue

BarryLiu-97, 2020-10-28:
I found something maybe a bug.
Now I used the master branch in GitHub.  I use Pycharm and terminal to run codes. One of my examples uses **raw.plot()** to plot raw data.  While I ran codes in Pycharm's console, **raw.plot()** went well, a figure showed up. While I ran .py in terminal like **python test.py**, no figure showed up. And if I type **python** in the terminal and type the codes in this way, also no figures showed up. But if I used the stable mne, everything was OK.

cbrnr, 2020-10-28:
Can you provide a minimal example (ideally using a sample dataset shipped with MNE) that demonstrates the issue?

BarryLiu-97, 2020-10-28:
Now I use https://mne.tools/stable/auto_tutorials/raw/plot_40_visualize_raw.html#sphx-glr-auto-tutorials-raw-plot-40-visualize-raw-py to plot raw data run by the terminal in win10. When I ran raw.plot(), nothing showed up.

BarryLiu-97, 2020-10-28:
my mne version is 0.22.dev0

hoechenberger, 2020-10-28:
> While I ran .py in terminal like **python test.py**, no figure showed up.

Can you try running in the terminal with

```
python -i test.py
```

BarryLiu-97, 2020-10-28:
> > While I ran .py in terminal like **python test.py**, no figure showed up.
> 
> Can you try running in the terminal with
> 
> ```
> python -i test.py
> ```

I did that, it was the same. no figure.

BarryLiu-97, 2020-10-28:
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.crop(tmax=60).load_data()

raw.plot()

BarryLiu-97, 2020-10-28:
In the stable version, it went well.

BarryLiu-97, 2020-10-28:
The interesting thing is, if I run the codes, the figure will show up.

hoechenberger, 2020-10-28:
@drammock You think this could be related somehow to the new figure class?

hoechenberger, 2020-10-28:
Can reproduce on macOS with `master`.

hoechenberger, 2020-10-28:
@BarryLiu-97 
I worked around this problem by adding
```python
import matplotlib.pyplot as plt
plt.show()
```
at the end of your script

BarryLiu-97, 2020-10-28:
> @BarryLiu-97
> I worked around this problem by adding
> 
> ```python
> import matplotlib.pyplot as plt
> plt.show()
> ```
> 
> at the end of your script

Yeah, that's right. Me too.

BarryLiu-97, 2020-10-28:
> > @BarryLiu-97
> > I worked around this problem by adding
> > ```python
> > import matplotlib.pyplot as plt
> > plt.show()
> > ```
> > 
> > 
> > at the end of your script
> 
It reminds me that Pycharm console is like jupyter notebook, don't need plt.show and the figure still shows up.

BarryLiu-97, 2020-10-28:
> > > @BarryLiu-97
> > > I worked around this problem by adding
> > > ```python
> > > import matplotlib.pyplot as plt
> > > plt.show()
> > > ```
> > > 
> > > 
> > > at the end of your script
> 
> It reminds me that Pycharm console is like jupyter notebook, don't need plt.show and the figure still shows up.

Seems that I still have a lot to learn. I didn't connect my experience to this issue.

hoechenberger, 2020-10-28:
> Seems that I still have a lot to learn. I didn't connect my experience to this issue.

That may be true, but you actually discovered a bug in MNE, so it's great you reported it! It should've worked the way you tried to do it.

BarryLiu-97, 2020-10-28:
> > Seems that I still have a lot to learn. I didn't connect my experience to this issue.
> 
> That may be true, but you actually discovered a bug in MNE, so it's great you reported it! It should've worked the way you tried to do it. We'll see how we can fix this.

All right. Looking forward to it.

PR #8450 (2020-10-29): FIX, VIZ: ensure figs appear
closes #8441
closed 2020-10-29 by drammock