mne-tools/mne-python #8443: Some advice on processing sEEG and ECoG data

BarryLiu-97, 2020-10-28:
I noticed that recently you guys are working on the iEEG process. Recently I found a paper called **Multi-Modal Neuroimaging Analysis and Visualization Tool (MMVT)** on arxiv. Here is its official website https://mmvt.mgh.harvard.edu/preprocessing. I think this may provide some inspiration for the process of the iEEG data.

larsoner, 2020-10-28:
Oddly enough [here](https://github.com/pelednoam/mmvt) it says:

> The project’s code is still under active development, but temporarily going off-grid. 

But in any case, feel free to list concrete ideas here. I don't think we plan to support everything that MMVT does (it has quite a feature list!) so we should focus on just what people need most urgently.

BarryLiu-97, 2020-10-29:
> Oddly enough [here](https://github.com/pelednoam/mmvt) it says:
>
>
> But in any case, feel free to list concrete ideas here. I don't think we plan to support everything that MMVT does (it has quite a [...]

Can't argue with that.

closed 2020-11-03 by BarryLiu-97