mne-tools/mne-python #8451: A function get_ch_positions for DigMontage

adam2392, 2020-10-29:
#### Describe the new feature or enhancement
Currently the only way I can find that gets a montage is either you make it from scratch make_dig_montage or you get it from raw using raw.get_montage. But if you say already have an existing montage via raw.get_montage() it seems you would have to hack out the channel positions and then call make_dig_montage again? Since there is no public API for getting the channel positions, I wonder if there's an easier way around this?

#### Describe your proposed implementation
Possibly a `get_ch_positions()` public function for DigMontage, that one can then do:

```
montage  = raw.get_montage()
ch_positions = montage.get_ch_positions()
```

#### Additional comments
TBD

agramfort, 2020-10-29:
sorry I don't get the usecase. Can you clarify what you start from and what
you are aiming for?

>

adam2392, 2020-10-29:
> sorry I don't get the usecase. Can you clarify what you start from and what you are aiming for?
> […](#)

Sorry about that. I updated the description. I just want to add fiducials to an existing montage.

agramfort, 2020-10-29:
ok got it. You want an easy way to get the parameters ch_pos etc... required by make_dig_montage from a DigMontage instance.

montage.get_ch_pos?

but then shall we add get_nasion, get_lpa etc...

@larsoner thoughts?

larsoner, 2020-10-29:
> but then shall we add get_nasion, get_lpa etc...

Maybe a more generic `get_positions(kinds='all' | 'channels' | 'extra' | 'fiducials' | list of str)`?

agramfort, 2020-10-29:
I prefer this yes. You would get a dict as output that you can directly
pass with **kwargs in make_dig_montage

>

larsoner, 2020-10-29:
In that case we should just make it `get_positions()`, and doing `mon_2 = make_dig_montage(**mon.get_positions())` should have all attributes of `mon` and `mon_2` the same. If you want the channel positions you just use `mon.get_positions['ch_pos']`

larsoner, 2020-10-29:
... or maybe `get_params` or `get_parameters()` is better, since `coord_frame` is an argument of `make_dig_montage`. So `mon.get_parameters()['ch_pos']` if you want just channel positions. Or we use `get_positions()` and live it being a misnomer because of `mon.get_positions()['coord_frame']` being there

larsoner, 2020-10-29:
(I guess `coord_frame` is information *about the positions* so I guess this wouldn't be so bad)

agramfort, 2020-10-29:
+1 for get_positions

>

adam2392, 2020-10-30:
So DigMontage.get_positions(), which returns a dictionary of channels positions and all the fiducials (if they're there).

Is there any reason we can't say just take the code inside `get_montage` rn:

```Python
        # obtain coord_frame, and landmark coords
        # (nasion, lpa, rpa, hsp, hpi) from DigPoints
        montage_bunch = _get_data_as_dict_from_dig(self.info['dig'])
        coord_frame = _frame_to_str.get(montage_bunch.coord_frame)

        # get the channel names and chs data structure
        ch_names, chs = self.info['ch_names'], self.info['chs']
        picks = pick_types(self.info, meg=False, eeg=True,
                           seeg=True, ecog=True)

        # channel positions from dig do not match ch_names one to one,
        # so use loc[:3] instead
        ch_pos = {ch_names[ii]: chs[ii]['loc'][:3] for ii in picks}

        # create montage
        montage = make_dig_montage(
            ch_pos=ch_pos,
            coord_frame=coord_frame,
            nasion=montage_bunch.nasion,
            lpa=montage_bunch.lpa,
            rpa=montage_bunch.rpa,
            hsp=montage_bunch.hsp,
            hpi=montage_bunch.hpi,
        )
        return montage
```

and make the part before `make_dig_montage` the code for `get_positions()`? If so it looks pretty straightforward.

larsoner, 2020-10-30:
That code operates on an info object. There is no info as part of DigMontage

PR #8460 (2020-11-03): [MRG] Getting ch positions from DigMontage via get_positions()
#### Reference issue
Fixes: #8451 

#### What does this implement/fix?
Adds `DigMontage.get_positions()` that would allow a full roundtrip with `make_dig_montage`. More importantly it obtains the DigMontage ch positions, coordinate frame and fiducials.

Adds a unit test. 

#### Additional information
~~For some reason the coordinate frame changes when using the template.~~ 

~~Is it safe to say we should have `get_positions` always return things in `coord_frame=head`? If so, what's suggestion to fix the unit test with the template  [... cut]
closed 2020-11-03 by larsoner