mne-tools/mne-python #8455: Unable to set standard montage for file recording using EEGOSPORTS

balandongiv, 2020-10-30:
I have an EEG recording recorded using EEGOSPORTS (ANT Neuro, Hengelo,Netherlands) machine. The recording was conducted using a 64-channels cap. However only 17 channels were active `Fp1, Fp2, Fpz, F3, Fz, F4, C3, Cz, C4, FCz, P3, Pz, P4, T7, T8, O1 and O2`, with reference at `CPz`. 

In mne, I would like to set the montage, as below

raw = mne.io.read_raw_edf ( path_name, preload=True ) # link to file : https://drive.google.com/file/d/1YPwd0uDt2EesSWz7BB5kQl7NDh22bdPL/view?usp=sharing
montage = mne.channels.make_standard_montage ( 'standard_1020' )
raw.set_montage ( montage, match_case=False )

However, the compiler return an error

> `ValueError: DigMontage is only a subset of info. There are 64 channel positions not present in the DigMontage. The required channels are: ['EEG Fp1-CPz', 'EEG Fpz-CPz', 'EEG Fp2-CPz', 'EEG F7-CPz', 'EEG F3-CPz', 'EEG Fz-CPz', 'EEG F4-CPz', 'EEG F8-CPz', 'EEG FC5-CPz', 'EEG FC1-CPz', 'EEG FC2-CPz', 'EEG FC6-CPz', 'EEG M1-CPz', 'EEG T7-CPz', 'EEG C3-CPz', 'EEG Cz-CPz', 'EEG C4-CPz', 'EEG T8-CPz', 'EEG M2-CPz', 'EEG CP5-CPz', 'EEG CP1-CPz', 'EEG CP2-CPz', 'EEG CP6-CPz', 'EEG P7-CPz', 'EEG P3-CPz', 'EEG Pz-CPz', 'EEG P4-CPz', 'EEG P8-CPz', 'EEG POz-CPz', 'EEG O1-CPz', 'EEG O2-CPz', 'EEG EOG-CPz', 'EEG AF7-CPz', 'EEG AF3-CPz', 'EEG AF4-CPz', 'EEG AF8-CPz', 'EEG F5-CPz', 'EEG F1-CPz', 'EEG F2-CPz', 'EEG F6-CPz', 'EEG FC3-CPz', 'EEG FCz-CPz', 'EEG FC4-CPz', 'EEG C5-CPz', 'EEG C1-CPz', 'EEG C2-CPz', 'EEG C6-CPz', 'EEG CP3-CPz', 'EEG CP4-CPz', 'EEG P5-CPz', 'EEG P1-CPz', 'EEG P2-CPz', 'EEG P6-CPz', 'EEG PO5-CPz', 'EEG PO3-CPz', 'EEG PO4-CPz', 'EEG PO6-CPz', 'EEG FT7-CPz', 'EEG FT8-CPz', 'EEG TP7-CPz', 'EEG TP8-CPz', 'EEG PO7-CPz', 'EEG PO8-CPz', 'EEG Oz-CPz']`

I have also tried different kind including `standard_alphabetic` and `biosemi64`

May I know how to properly set the montage.

Remark:
The `edf `file is accessible via the link: https://drive.google.com/file/d/1YPwd0uDt2EesSWz7BB5kQl7NDh22bdPL/view?usp=sharing

agramfort, 2020-10-30:
message is clear:

'EEG Fp1-CPz' ...

these are your channel names. You need to rename your channels to be
able to set_montage.

>

balandongiv, 2020-10-30:
Thanks for the hint @agramfort. Based on your suggestion, I rename the channels name as below 

```
raw = mne.io.read_raw_edf ( path_name, preload=True )
raw.rename_channels ( {s: s.replace('-CPz', '').replace('EEG ', '') for s in raw.ch_names} )

montage = mne.channels.make_standard_montage ( 'standard_1020' )
raw.set_montage ( montage, match_case=False )
```

While there is a channel with the name `EOG`, the compiler still return the following error

> ValueError: DigMontage is only a subset of info. There are 1 channel position not present in the DigMontage. The required channels are: ['EOG']

larsoner, 2020-10-30:
You can use `on_missing` to ignore this message or (better) `set_channel_types` to set the channel type to EOG

balandongiv, 2020-10-31:
Thanks @larsoner.

For other, the working code is as below:

```
raw = mne.io.read_raw_edf ( path_name, preload=True )
raw.rename_channels ( {s: s.replace('-CPz', '').replace('EEG ', '') for s in raw.ch_names} )
raw.set_channel_types ({'EOG': 'eog'})
montage = mne.channels.make_standard_montage ( 'standard_1020' )
raw.set_montage ( montage, match_case=False ) 
```

closed 2020-10-31 by balandongiv