mne-tools/mne-python #8457: Improving documentation for events <-> Annotations

adam2392, 2020-10-30:
#### Proposed documentation ehancement
In gitter, I raised a question:

"""
I'm wondering if the API for events_from_annotations and annotations_from_events are intentionally have flip-flopped event_mapping arguments?

E.g. In events_from_annotations, you get event_id which has key/values as description: ID

In annotations_from_events, you must pass in key/values as ID: description.

I had a use-case where I ended up needing to call them in sequence to create a new Annotations object and realized I can't just pass them into each other. I have to invert the event_id I get out of events_from_annotations. If this is undesirable, I can file an issue.

If it is intended, I might be just be overthinking this then.
"""

@larsoner suggested:

"""
@adam2392 I'm not sure if it's intended or not -- it might have been -- but either way it's probably not worth an API deprecation cycle to change either function. Instead I'd recommend a .. note:: in one of them (probably annotations_from_events since it's used less?) noting the difference and suggesting that people can just use {v: k for k, v in event_id.items()} if they need to reverse it
"""

## Suggested Soln
I will add a short documentation note to the `annotations_from_events` to do the swap key/vals.

drammock, 2020-10-30:
FWIW, I would guess that this *was* an intentional design choice, given that the argument in one case is called `event_id` (and expects what we call `event_id` elsewhere, i.e., a "description → integer" mapping), and in the other case the argument is called `event_desc`.

agramfort, 2020-10-31:
yes it's intentional.

>

hoechenberger, 2020-11-01:
And it's easy to convert between the two in Python 3:
```python
xx = dict(zip(x.values(), x.keys())
```

PR #8458 (2020-11-02): [MRG] Add doc note to annotations_from_events function
#### Reference issue
Closes: #8457 

#### What does this implement/fix?
Adds a section on notes for `annotations_from_events` to remind users to invert dictionary.
closed 2020-11-02 by larsoner