mne-tools/mne-python #8465: attach mne.viz.plot_raw() to a pyqt5 gui interface

BarryLiu-97, 2020-11-02:
Hi, If I want to insert the figure mne.viz.plot_raw() produced onto the GUI interface using PyQt5, how do I do it? I did fig = mne.viz.plot(data), while it showed up two figures. One is on my PyQt5 interface which cannot interact with me using the keyboard or the mouse. The other is almost like the figure we sued to plot while when I interact with it, the figure on the gui interface changed.
Here is my code: 
import matplotlib
matplotlib.use('Qt5Agg')
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
from PyQt5 import QtWidgets
import matplotlib.pyplot as plt
import sys
import mne

class My_Main_window(QtWidgets.QMainWindow):
    def __init__(self,parent=None):
        super(My_Main_window,self).__init__(parent)

        self.resize(800, 659)
        self.menu = QtWidgets.QMenu("绘图")
        self.menu_action = QtWidgets.QAction("绘制",self.menu)
        self.menu.addAction(self.menu_action)
        self.menuBar().addMenu(self.menu)
        self.menu_action.triggered.connect(self.plot_)
        self.setCentralWidget(QtWidgets.QWidget())

    def plot_(self):

        fpath = 'D:\data\plot_test.fif'
        data = mne.io.read_raw_fif(fpath, preload=True)
        fig = mne.viz.plot_raw(data)
        cavans = FigureCanvas(fig)
        self.setCentralWidget(cavans)

if __name__ == '__main__':
    app = QtWidgets.QApplication(sys.argv)
    main_window = My_Main_window()
    main_window.show()
    app.exec()

BarryLiu-97, 2020-11-02:
when I use mne.viz.plot_raw(data, show=False), the other figure disappeared, but I have no way to interact with the figure. I found that type(mne.viz.plot_raw) is the son of class Figure, so I think it is possible to use my code implementate it.

cbrnr, 2020-11-02:
I think you need to use `show=False` in your statement `fig = mne.viz.plot_raw(data)`, then the figure is drawn in the main window (see also https://www.learnpyqt.com/courses/graphics-plotting/plotting-matplotlib/ for a nice tutorial).

BarryLiu-97, 2020-11-02:
> I think you need to use `show=False` in your statement `fig = mne.viz.plot_raw(data)`, then the figure is drawn in the main window (see also https://www.learnpyqt.com/courses/graphics-plotting/plotting-matplotlib/ for a nice tutorial).

Thank you so much and I did setshoe=False. I will dig into this tutorial.

BarryLiu-97, 2020-11-02:
and I also need  to write callbacks so the figures can be interactice like you guys implementated.

drammock, 2020-11-02:
@BarryLiu-97 can you explain the big picture of what you to do?  Are you trying to build a GUI like [mnelab](https://pypi.org/project/mnelab/)?

BarryLiu-97, 2020-11-02:
> @BarryLiu-97 can you explain the big picture of what you to do? Are you trying to build a GUI like [mnelab](https://pypi.org/project/mnelab/)?

Yeah, something like that. But more like an interest.

BarryLiu-97, 2020-11-02:
A good chance to practice python and mne.

drammock, 2020-11-02:
@BarryLiu-97 I think your request is pretty similar to #8142.  Can you look at that issue and see if these are similar enough that one of them could be closed (and the other one made more general, if needed)?

BarryLiu-97, 2020-11-02:
> @BarryLiu-97 I think your request is pretty similar to #8142. Can you look at that issue and see if these are similar enough that one of them could be closed (and the other one made more general, if needed)?

Yes it is similar. Mine can be closed. I think it may be a general issue if people want to embed the interactive figure inside a gui.

BarryLiu-97, 2020-11-02:
I will close mine and if there are any issues, I will ask there.

closed 2020-11-02 by BarryLiu-97

drammock, 2020-11-02:
perfect