mne-tools/mne-python #8470: mne.io.read_raw_edf generates an error for an edf startdate : 00.00.00

klacourse, 2020-11-02:
The start date of my edf file is 00.00.00 for dd.mm.yy. I'm assuming the file has been anonymized (not properly). The file does not meet the EDF specification, but the data saved is still valid. Would it be possible to change the "ValueError: month must be in 1..12" error from mne.io.read_raw_edf to a warning? Otherwise, Is there any way I could read this edf file with the mne library?

cbrnr, 2020-11-03:
This should have been fixed by #8283. Which MNE version did you use to load the data?

larsoner, 2020-11-03:
We have not released a patch 0.21.1 release yet. So it's only fixed in `master` at the moment. It's probably time to push out 0.21.1...

sappelhoff, 2020-11-13:
@klacourse, mne 0.21.1 has been released meanwhile. Can you upgrade your installation and tell us whether your problem is solved, please? If yes, feel free to close this issue. Thanks!

klacourse, 2020-11-13:
@sappelhoff, perfect it is solved, I got this warning : <ipython-input-3-e17fde07d2c7>:1: RuntimeWarning: Invalid date encountered (2000-00-00 00:00:00). Thanks!

closed 2020-11-13 by klacourse