mne-tools/mne-python #8472: mne.viz.plot_raw issue

BarryLiu-97, 2020-11-03:
Here is my  first code:
```
import matplotlib
matplotlib.use('Qt5Agg')
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
import matplotlib.pyplot as plt
from mne.datasets import sample
import mne

data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
raw.plot()
plt.show()
fig = mne.viz.plot_raw(raw, show=False)
canvas = FigureCanvas(fig)
```
And the result is this:
[image]

BarryLiu-97, 2020-11-03:
Here is my second code:
```
import matplotlib
matplotlib.use('Qt5Agg')
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
import matplotlib.pyplot as plt
from mne.datasets import sample
import mne

data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)

fig = mne.viz.plot_raw(raw, show=False)
canvas = FigureCanvas(fig)
raw.plot()
plt.show()
```
And the traceback is:
```
Traceback (most recent call last):
  File "plot_test_2.py", line 14, in <module>
    raw.plot()
  File "D:\test\test\lib\site-packages\mne\io\base.py", line 1433, in plot
    show_scalebars=show_scalebars, verbose=verbose)
  File "<decorator-gen-148>", line 24, in plot_raw
  File "D:\test\test\lib\site-packages\mne\viz\raw.py", line 584, in plot_raw
    plt_show(show, block=block)
  File "D:\test\test\lib\site-packages\mne\viz\utils.py", line 111, in plt_show
    (fig or plt).show(**kwargs)
  File "D:\test\test\lib\site-packages\matplotlib\pyplot.py", line 272, in show
    return _show(*args, **kw)
  File "D:\test\test\lib\site-packages\matplotlib\cbook\deprecation.py", line 400, in wrapper
    return func(*args, **kwargs)
  File "D:\test\test\lib\site-packages\matplotlib\backend_bases.py", line 3389, in show
    manager.canvas.figure.show()
  File "D:\test\test\lib\site-packages\matplotlib\figure.py", line 439, in show
    "created by pyplot.figure()." % err)
AttributeError: 'FigureCanvasQTAgg' object has no attribute 'manager'
Figure.show works only for figures managed by pyplot, normally created by pyplot.figure().
```

BarryLiu-97, 2020-11-03:
As you can see, I just changed the order of these two-part of codes:
```
fig = mne.viz.plot_raw(raw, show=False)
canvas = FigureCanvas(fig)
```

```
raw.plot()
plt.show()
```

BarryLiu-97, 2020-11-03:
So I didn't know why this happened, maybe because some scripts have conflict?

BarryLiu-97, 2020-11-03:
I did this because I wanted to embed the figure from plot_raw inside a PyQt5 GUI, so I needed to set up a canvas to the main window of my GUI.

BarryLiu-97, 2020-11-03:
I ran the codes in the terminal.

BarryLiu-97, 2020-11-03:
But if I ran codes using Pycharm's Python Console or IPyhton, everything is OK.

cbrnr, 2020-11-03:
@BarryLiu-97 you can edit existing comments - it will be easier for other people to read if you try to create as few comments with as much information as possible.

BarryLiu-97, 2020-11-03:
> @BarryLiu-97 you can edit existing comments - it will be easier for other people to read if you try to create as few comments with as much information as possible.

Of course, thanks for the heads-up!

BarryLiu-97, 2020-11-03:
Hello, I think I solve this question although I just have a little guess for the reason. As we can see, the error **said AttributeError: 'FigureCanvasQTAgg' object has no attribute 'manager'**, which means the attribute 'manager' is the key. Then I found that if we want to use plt.show(), we must need 'manager', if we use plt.figure(), then the manager is generated by default, but plt.figure.Figure() is not. While what I used here:
```
fig = mne.viz.plot_raw(raw, show=False)
```
is the latter one, which is plt.figure.Figure(). So what I do is add
```
plt.close()
```
to stop the figure show for all I need is to add it onto my GUI. Then everything works well. I think my explanation must have some bugs for I am not really as good as you guys in python, but I think I ought to go in the right direction.

drammock, 2020-11-03:
I'm not 100% clear what you're trying to do here, but I'm guessing you're still trying to get a raw plot into a FigureCanvas object that you created (like in #8465), instead letting our plotting function create its own figure.  This works for me in a plain python (i.e., not ipython) interpreter:

```python
import matplotlib
matplotlib.use('Qt5Agg')
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
import matplotlib.pyplot as plt
from mne.datasets import sample
import mne
data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
fig = raw.plot(show=False)  # same as fig = mne.viz.plot_raw(raw, show=False)
canvas = FigureCanvas(fig)
plt.show()
```

Notice that I removed the initial `raw.plot()` and moved `plt.show()` to the end.  Note also that when you do this you lose figure interactivity (which is expected).

With that said, I want to draw your attention to a few points:

1. as mentioned in a code comment in my snippet above, `raw.plot(...)` and `mne.viz.plot_raw(raw, ...)` are completely equivalent. I don't know why you're calling them each once in the code samples you posted, but you don't need to.

2. What I think you're trying to do, while not unreasonable, is not documented anywhere as a supported use case of MNE-Python.  That doesn't mean we can't or won't support it, but it does mean that your difficulties in making it work should probably not be filed as bugs.  If you want it to work that way, there is already an enhancement request in #8142, which I directed you to yesterday.  Please use that issue for any further discussion of embedding raw.plot windows into other figure canvases / GUIs.

3. The errors that you're getting are *matplotlib* errors. The last four lines of the traceback you posted are lines within matplotlib, for example:

```
                                      ↓↓↓↓↓↓↓↓↓↓↓
  File "D:\test\test\lib\site-packages\matplotlib\figure.py", line 439, in show
```

When you see that, it's possible that the underlying problem *might* still be with MNE-Python, but it might not be. This is another reason to be cautious about treating it as a bug report (instead of framing it as a question like "is it possible to do X?", or as a feature request like "I want to be able to do X with MNE-Python").

closed 2020-11-03 by drammock

BarryLiu-97, 2020-11-04:
Hi, the reason that I was calling them each once in the code samples I posted is that in my design, my GUI has a button to plot figures and also a figure in my main window to show the data, which means in my codes, the button is connected to a function using raw.plot and a function which is always running using
```
fig = raw.plot(show=False) 
canvas = FigureCanvas(fig)
```
but I can't plot data using raw.plot() to get the figures when clicking the plot button. So I used simplified codes and try to find out why, and I have fixed it. 
And yes, I know ```raw.plot()```and ```mne.viz.plot_raw()``` are equivalent, I was just too lazy to change that.

BarryLiu-97, 2020-12-31:
Hi, recently I was trying to embed an interactive matplotlib figure into a PyQt5 GUI. I think I have some progress. While here is one sample I think is relevent.
```
import matplotlib
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
from matplotlib.backends.backend_qt5agg import NavigationToolbar2QT as NavigationToolbar
# from matplotlib.backends.qt_compat import QtCore
from PyQt5 import QtCore
from matplotlib.figure import Figure
from PyQt5 import QtWidgets
import sys
import numpy as np

class My_Main_window(QtWidgets.QMainWindow):
    def __init__(self, parent=None):
        super(My_Main_window, self).__init__(parent)

        self.canvas = FigureCanvas(Figure(constrained_layout=True))
        self.canvas.draw()
        self.ax = self.canvas.figure.add_subplot()

        self.con_1 = np.random.random((30, 40, 10))
        self.num = 0
        con_plot = self.con_1[:, :, self.num]
        self.im = self.ax.matshow(con_plot)
        self.ax.set_title(self.num)
        self.canvas.figure.colorbar(self.im)
        self.canvas.mpl_connect('key_press_event', self.on_move)
        self.canvas.mpl_connect(
            "button_press_event", lambda *args, **kwargs: print(args, kwargs)
        )
        self.canvas.mpl_connect(
            "key_press_event", lambda *args, **kwargs: print(args, kwargs)
        )

        self.setCentralWidget(self.canvas)

        # Give the keyboard focus to the figure instead of the manager:
        # StrongFocus accepts both tab and click to focus and will enable the
        # canvas to process event without clicking.
        # https://doc.qt.io/qt-5/qt.html#FocusPolicy-enum
        self.canvas.setFocusPolicy(QtCore.Qt.StrongFocus)
        self.canvas.setFocus()
        print(type(self.canvas))

    def on_move(self, event):
        print(f"activate this {event.key}, {self.num}")
        if event.key == "left":
            if self.num == 0:
                self.num = 0
                con_plot = self.con_1[:, :, self.num]
            else:
                self.num -= 1
                con_plot = self.con_1[:, :, self.num]
        elif event.key == "right":
            if self.num == self.con_1.shape[2] - 1:
                self.num = 0
                con_plot = self.con_1[:, :, self.num]
            else:
                self.num += 1
                con_plot = self.con_1[:, :, self.num]
        else:
            con_plot = self.con_1[:, :, self.num]
        # update the image in place
        self.im.set_data(con_plot)
        self.ax.set_title(self.num)
        # tell Qt that the widget needs to be re-repainted
        self.canvas.draw_idle()

if __name__ == '__main__':
    app = QtWidgets.QApplication(sys.argv)
    app.setApplicationName("Plot")
    main_win = My_Main_window()
    main_win.show()
    app.exec()
```
The key part is 
```
self.canvas.setFocusPolicy(QtCore.Qt.StrongFocus)
        self.canvas.setFocus()
```
so the gui will focus on the events aiming at the canvas. While it doesn't work when I  used it i mne.viz.plot()
Here is the code:
```
import matplotlib
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
from matplotlib.backends.qt_compat import QtCore
from PyQt5 import QtWidgets
import sys
from mne.datasets import sample
import mne

class My_Main_window(QtWidgets.QMainWindow):
    def __init__(self, parent=None):
        super(My_Main_window, self).__init__(parent)
        
        data_path = sample.data_path()
        fpath= data_path + '/MEG/sample/sample_audvis_raw.fif'
        self.data = mne.read_epochs(fpath, preload=True)
        self.fig = mne.viz.plot_epochs(self.data, title='1', show=False)
        self.canvas = FigureCanvas(self.fig)
        self.canvas.draw()
        self.stack = QtWidgets.QStackedWidget()
        self.stack.addWidget(self.canvas)
        self.setCentralWidget(self.stack)

        # Give the keyboard focus to the figure instead of the manager:
        # StrongFocus accepts both tab and click to focus and will enable the
        # canvas to process event without clicking.
        # https://doc.qt.io/qt-5/qt.html#FocusPolicy-enum
        self.canvas.setFocusPolicy(QtCore.Qt.StrongFocus | QtCore.Qt.WheelFocus)
        self.canvas.setFocus()

if __name__ == '__main__':
    app = QtWidgets.QApplication(sys.argv)
    app.setApplicationName("Plot")
    main_win = My_Main_window()
    main_win.show()
    app.exec()
```
@drammock