mne-tools/mne-python #8473: Add "unit" parameter to get_data method for Raw and Epochs

sappelhoff, 2020-11-03:
I often find myself doing things like this:

```Python
# get EEG data and scale to micro volts
# ntrials x nchannels x ntimes
eeg_data_uV = epochs.get_data() * 1e6
```

... just to do different kinds of analyses with custom functions.

I think we should add a `"unit"` parameter to `.get_data()`:

```python
Parameters
----------
unit : str, one of ["nV, uV, mV, V"]
    The unit in which the data will be returned. Defaults to "V".
```

This would make it more transparent for users what units the data are in they are receiving, and also offer a chance for customization.

I have a complete déjà vu writing up this issue, but I couldn't find existing threads on this.

agramfort, 2020-11-03:
get data will give you any of any channel type not just EEG. What if you
have misc, eeg, eog, grad and mag? you pass a dict of units?

>

larsoner, 2020-11-03:
> you pass a dict of units?

That was my thinking. It keeps it consistent with the `units` param we have in viz. Though in viz you're *saying what the unit of the data is*, meaning no scale factors get applied unless you also supply `scalings` as well -- and in this function the data *will be converted to those units* so we'll have to be careful about the docstring.

sappelhoff, 2020-11-03:
> you pass a dict of units?

and if in the data there are channel types for which no keys are passed, then the default unit is used :thinking:

larsoner, 2020-11-03:
> and if in the data there are channel types for which no keys are passed, then the default unit is used ￼

This is also how it works in `units` for our viz functions, yes.

How do we handle `units='uV'`? Do we treat this effectively as:

1. `dict(eeg='uV', eog='uV', ecg='uV', misc='uV')`, "anything that typically has V units should be returned in uV"
2. raise an error, "more than one channel type could be in uV units"
3. `dict(eeg='uV')`, "whatever the first data channel type with this unit will get uV" (probably not safe for example because of reference MEG channels)

agramfort, 2020-11-03:
> and if in the data there are channel types for which no keys are passed, then the default unit is used

yes that seems natural

sappelhoff, 2020-11-03:
> How do we handle units='uV'?

perhaps we could make our lives easy and only accept a dict?

```python
"""
Parameters
----------
unit : dict | None
    Key-value pairs in this dictionary determine the unit in which the data
    of a particular channel type will be returned. For example
    ``{"eeg": "uV", "eog": "nV"}`` will return data from all channels of type "eeg"
    in micro Volts and those of type "eog" in nano Volts. All other potentially
    existing channel types in the data will retain the unit they are currently in.
    Defaults to None, returning MNE standard unit for each channel type.    
"""
```

larsoner, 2020-11-03:
I think the `str` option is nice for EEG users, or really any users with a single channel type (other than STIM/trigger) in their data. I see your edit and agree (1) is probably the way to go.

HanBnrd, 2021-03-15:
Hi, would it work with fNIRS as well? The fNIRS channels when the types are `hbo` and `hbr` are in M (mol/L) but it may be interesting to have the data in mM or uM.

sappelhoff, 2021-03-15:
Hey @HanBnrd - that'd probably be possible. Currently, nothing has been implemented: We only have a good idea of what we want :-)

The start was made here: https://github.com/mne-tools/mne-python/pull/8476

If you want to take this on, I'd be happy to help along the way

HanBnrd, 2021-03-15:
Okay, will take it then :)

HanBnrd, 2021-03-16:
I now had a look at it more in details, are you sure this is worth implement this? I feel that this would be quite a consequent code addition for little reward. For example it would enable to write `epochs.get_data(unit=dict(eeg='uV'))` instead of `epochs.get_data() * 1e6` which is even longer. While the code change needs to take into account different units or different structures (raw, epochs...).

sappelhoff, 2021-03-16:
The benefits are (at least) twofold:

1. when somebody (a **non-user**) reads your code, something like `epochs.get_data() * 1e6` is much more confusing than `epochs.get_data(unit=dict(eeg='uV'))` --> where does the magic number `1e6` come from? They'd need to learn that in MNE all EEG data is represented in Volts, and that 1e6 scales to microVolts
1. as a fresh MNE **user**, I was confused about these units and a parameter `unit` would certainly have cleared up the situation for me

But I agree that this is an issue that requires quite some work, so no worries if you'd rather work on something else.

HanBnrd, 2021-03-16:
Okay makes sense, I just wanted to be sure before starting working on it :)

HanBnrd, 2021-03-16:
I was wondering why the EOG channels are not considered as data channels in [`get_channel_types()`](https://mne.tools/dev/generated/mne.io.BaseRaw.html#mne.io.BaseRaw.get_channel_types) and whether this `units` parameter should work on them?

larsoner, 2021-03-16:
> I was wondering why the EOG channels are not considered as data channels in get_channel_types() 

In some sense *all* channels are data channels. I think the idea is that the "data channels" in MNE correspond mostly to the channels that record brain data.

> whether this units parameter should work on them?

I wouldn't restrict it just to data channels. Any channels with units (which is most of them I think?) should be able to be modified by the `unit` parameter.

sappelhoff, 2021-04-22:
If I remember correctly there was also some talk somewhere (can't find it right now) to add a `.get_data` method to `Evoked` objects. Once that is added, the `units` parameter should also work for that.

PR #9553 (2021-07-15): [MRG] Add `units` parameter to get_data for Epochs
closes #8473 

Follow up to #9136

Adds the `units` parameter to the `get_data` method of `Epochs`.

~~**Work in Progress**~~

To do

- [x] refactor units in Raw
- [x] add units to Epochs
- [x] add tests for Epochs
- [x] pull units docs into docdict
- [x] document all params
- [x] add whatsnew entry
closed 2021-07-15 by larsoner