mne-tools/mne-python #8481: Re-reference methods about sEEG data

BarryLiu-97, 2020-11-04:
**Describe the new feature or enhancement**
In sEEG data analysis, reference is not alike in EEG. Many methods are used to reference sEEG data, linear and non-linear, such as 
Common Average Reference(CAR), Gray-white Matter Reference(GWR), Electrode Shaft Reference(ESR), Bipolar Reference, Laplacian Reference and Monopolar Reference.
**Describe your proposed implementation**
I read the paper to know the reference in sEEG analysis. Here is the link:https://www.sciencedirect.com/science/article/abs/pii/S1053811918307183

larsoner, 2020-11-04:
Which do you need for your analysis? ESR? Is this just CAR for subsets of electrodes?

It would help if you could guide us specifically about what's needed here. Papers are helpful but it takes time to read them -- if others already have expertise and knowledge things will proceed faster. Maybe @adam2392 could comment as well.

larsoner, 2020-11-04:
And just a general plan of action, we should not implement every method available. We should implement the ones will benefit a large proportion of users. So if 90% of sEEG folks use CAR and monopolar (which we already support) and ESR (which we do not), we should just implement ESR and not worry about GWR and Laplacian.

BarryLiu-97, 2020-11-04:
Actually, I don't have somespecific methods. I am asked to implement them all by my mentor. But CAR is not recommended in most cases in sEEG analysis. While monopolar is widely used, especially in the hospital, well, at least in China. And ESR and Laplacian is quite the same for they just delete different channels' data in the same shaft. And as to GWR, it is a little complex for you need to know what channels are in the white or gray zones for sEEG is very Individually.

adam2392, 2020-11-04:
My impression is that for sEEG:

- monopolar (just don't do anything)
- CAR (set_eeg_reference('average'))
- and bipolar 

are supported in mne. These other references are possible but need to happen outside mne. E.g. for gray/white, one must have a list of the wm channels and then gm channels to then use `set_eeg_reference(wm_chs)`. 

ESR would be desirable altho I've never used it (becuz mne didn't have it before :p)... This can be done in sEEG by taking the unique letters for each electrode and then referencing the average signal of those with the channels that are part of that electrode. Although a user can do that themselves, I see this as being implementable in the mne-python codebase! 

WDYT @larsoner

BarryLiu-97, 2020-11-04:
And some little tips on ESR, some channels' name rea very tricky for some shaft named ```A```, and some named ```A'```

BarryLiu-97, 2020-11-04:
To compute the Laplacian, each channel was re-referenced to the mean value of its two adjacent contacts along the electrode shaft. This seems similar to Bipolar. I said it wrong before.

adam2392, 2020-11-04:
Laplacian can generalize based on a distance metric defined, but... idk if that's desirable. 

@BarryLiu-97 do you want to implement ESR in a PR?

larsoner, 2020-11-04:
> This can be done in sEEG by taking the unique letters for each electrode and then referencing the average signal of those with the channels that are part of that electrode.

> And a little tip on ESR, some channels' name is very tricky for some shaft named A, and some named A'

Agreed, I would not write channel-letter-based referencing stuff. I would make people give the channel names they want to use, it allows them to deal with their own naming schemes.

Really the question is what API do we need to support ESR then I think. Currently the relevant API is:
```
mne.set_eeg_reference(inst, ref_channels='average', ch_type='auto')
```
If we were to add the argument `picks`, which would supersede / ignore `ch_type` (it actually could replace `ch_type` entirely but probably isn't worth the deprecation to do so?), then you could for example do:

```
refs = [ch_name for ch_name in raw.ch_names if ch_name.endswith("A'")]
picks = refs + [ch_name for ch_name in raw.ch_names if ch_name.endswith("A")
raw.set_eeg_reference(refs, picks=picks)
```
This would reference all `picks` channels (i.e., those ending in A or A') based on the average of all channels in `refs` (i.e., those ending in A').

> Laplacian can generalize based on a distance metric defined, but... idk if that's desirable.

Let's wait to do this until we see if `set_bipolar_reference` is insufficient for most users

BarryLiu-97, 2020-11-04:
I am implementing ESR on my own, but I don't know how to do that in mne. @ @adam2392