mne-tools/mne-python #8488: Migrate one of Travis job to GitHub Actions

GuillaumeFavelier, 2020-11-05:
#### Describe the new feature or enhancement
It's been some months now that we use [GitHub Actions for MacOS testing](https://github.com/mne-tools/mne-python/pull/8161) and AFAIK the experience has been okay (I am not aware of any issue). So I would like to proceed with the [incremental approach](https://github.com/mne-tools/mne-python/issues/8159#issuecomment-680020111) and create a GHA job that would replace one of our current Travis job. I believe this change will also decrease the load on Travis in general. If possible, which job should be moved?

#### Describe your proposed implementation
I would update the current workflow file to add support for Linux.

#### Additional comments
Related to https://github.com/mne-tools/mne-python/issues/8159

larsoner, 2020-11-05:
I would go with whatever the next easiest to migrate would be. Probably the topmost build, 3.8 + environment.yml? This uses almost the same miniconda + minimal_cmds (with different URL) + test calls, right?

GuillaumeFavelier, 2020-11-05:
> I would go with whatever the next easiest to migrate would be. Probably the topmost build, 3.8 + environment.yml

Good idea, they're also the slowest tests.

cbrnr, 2020-11-05:
:+1: &ndash; more and more people seem to be complaining about Travis performance/support getting worse, so the long-term goal should be to completely replace Travis.

closed 2020-11-09 by larsoner