mne-tools/mne-python #8499: Issue with Epochs .__getitem__

sappelhoff, 2020-11-09:
#### Describe the bug

Either the documentation for the [`Epochs.__getitem__`](https://mne.tools/stable/generated/mne.Epochs.html?highlight=selecting%20epochs#mne.Epochs.__getitem__) method is wrong, or it doesn't work as advertised because of a bug.

Specifically, point 3 from the docs:

> epochs[['name_1', 'name_2', ... ]]: Return Epochs object with a copy of the subset of epochs corresponding to multiple experimental conditions as specified by 'name_1', 'name_2', ... .

> If conditions are separated by ‘/’, selects every item containing every list tag (e.g. [‘audio’, ‘left’] selects ‘audio/left’ and ‘audio/center/left’, but not ‘audio/right’).

#### Steps to reproduce

```Python
sfreq = 100
data = np.random.randn(10, 5, sfreq*2)
info = mne.create_info(
    ch_names=['MEG1', 'MEG2', 'EEG1', 'EEG2', 'EOG'],
    ch_types=['grad', 'grad', 'eeg', 'eeg', 'eog'],
    sfreq=sfreq
)

events = np.array([
    [0, 0, 1],
    [1, 0, 2],
    [2, 0, 3],
    [3, 0, 4],
    [4, 0, 5],
    [5, 0, 6],
    [6, 0, 7],
    [7, 0, 8],
])

event_id = {
    "active/left/out1": 1,
    "active/right/out1": 2,
    "active/left/out2": 3,
    "active/right/out2": 4,
    "passive/left/out1": 5,
    "passive/right/out1": 6,
    "passive/left/out2": 7,
    "passive/right/out2": 8,
}

custom_epochs = mne.EpochsArray(data, info, events, tmin=-0.1, event_id)

# Now check
# Expected:
# active/left/out1
# active/left/out2
custom_epochs[["active", "left"]]

# but returns
 #'active/left/out1': 2
 #'active/left/out2': 1
 #'active/right/out1': 2
 #'active/right/out2': 1
 #'passive/left/out1': 1
 #'passive/left/out2': 1>

# works as expected:
custom_epochs["active/left"]
```

#### Expected results

Based on the docs I would have expected 

```
# active/left/out1
# active/left/out2
```

to be the result, see the docs:

> e.g. [‘audio’, ‘left’] selects ‘audio/left’ and ‘audio/center/left’, but not ‘audio/right’)

#### Actual results
*Replace this text with the actual output, traceback, screenshot, or other
description of the results.*

```
 #'active/left/out1': 2
 #'active/left/out2': 1
 #'active/right/out1': 2
 #'active/right/out2': 1
 #'passive/left/out1': 1
 #'passive/left/out2': 1>
```

#### Additional information

Using MNE 0.22.1

Using `custom_epochs["active/left"]` works as expected, but is not documented.

larsoner, 2020-11-09:
This is just a documentation problem. `['a', 'b']` is like saying A or B. `['a/b']` is like saying A and B.

larsoner, 2020-11-09:
@drammock can you take a stab at improving the docs for this one?

commit 6734953d (2020-11-10): MRG, DOC: epochs getitem clarification (#8500)

* closes #8499

* clarify some more

* formatting; clearer

* missed outdent
closed 2020-11-10 by drammock