mne-tools/mne-python #8501: building docs on Windows: a few remaining problems

kalenkovich, 2020-11-10:
#### Describe the bug
Docs build with errors on Windows.

#### Steps to reproduce
Try building docs on Windows.

#### Expected results
Docs build without errors.

#### Actual results
Docs build with errors:

- `limo.py` concatenates URL parts using `os.path.join` which, on Windows, uses `\` as separators resulting in invalid URLs. I suggest hardcoding the concatenation.
- `plot_eeg_erp.py` writes a file to the `/tmp` folder which does not exist on Windows. I suggest writing to `tempfile.gettempdir()`.
- 
	```
	WARNING: dot command 'dot' cannot be run (needed for graphviz output), check the graphviz_dot setting
	```

	There is a problem with conda-installed graphviz on Windows: `dot.bat` is added to path, not `dot.exe` and some shells (git-bash in my case) don't know that running `dot` means running `dot.bat`. I suggest adding to the docs a note that Windows users which get a dot-related error might have to install graphviz manually (and check "add to path" options while at it) even if they have already installed it using conda.
- 
	```
	C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_examples\inverse\plot_vector_mne_solution.rst:: WARNING: image file not readable: C:UsersevgeniiOneDriveprojectsssvep_tutorialrepodocauto_examplesinverseimagessphx_glr_plot_vector_mne_solution_001.gif
	C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_tutorials\misc\plot_ecog.rst:: WARNING: image file not readable: C:UsersevgeniiOneDriveprojectsssvep_tutorialrepodocauto_tutorialsmiscimagessphx_glr_plot_ecog_004.gif
	C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_tutorials\misc\plot_seeg.rst:: WARNING: image file not readable: C:UsersevgeniiOneDriveprojectsssvep_tutorialrepodocauto_tutorialsmiscimagessphx_glr_plot_seeg_003.gif
	C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_tutorials\source-modeling\plot_mne_dspm_source_localization.rst:: WARNING: image file not readable: C:UsersevgeniiOneDriveprojectsssvep_tutorialrepodocauto_tutorialssource-modelingimagessphx_glr_plot_mne_dspm_source_localization_008.gif
	C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_tutorials\source-modeling\plot_visualize_stc.rst:: WARNING: image file not readable: C:UsersevgeniiOneDriveprojectsssvep_tutorialrepodocauto_tutorialssource-modelingimagessphx_glr_plot_visualize_stc_002.gif
	```
	
	Here is a string  from `plot_vector_mne_solution.rst` that leads to one of these problems:
	
	```
	.. image:: /C:\Users\evgenii\OneDrive\projects\ssvep_tutorial\repo\doc\auto_examples\inverse\images\sphx_glr_plot_vector_mne_solution_001.gif
	```
	
	This image (and all the other error-causing ones) is specified using an absolute path that starts with `/` which does not work on Windows. I don't know where these paths come from so I could not figure out how to solve this.

#### Additional information
```
Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\evgenii\miniconda3\envs\mnedev\python.exe
CPU:           AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD: 8 cores
Memory:        15.4 GB

mne:           0.22.dev0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.3
dipy:          1.2.0
mayavi:        4.7.2
pyvista:       0.26.1 {pyvistaqt=0.2.0, OpenGL 4.5.13587 Core Profile Context 19.40.44.03 26.20.14044.3001 via AMD Radeon(TM) Graphics}
vtk:           9.0.1
PyQt5:         5.15.1
```

larsoner, 2020-11-10:
> This image (and all the other error-causing ones) is specified using an absolute path that starts with / which does not work on Windows. I don't know where these paths come from so I could not figure out how to solve this.

This comes from sphinx-gallery. Must be a bug there. Feel free to open an issue, or even fix it!

The first thing to try would be, on a Windows system, clone the `sphinx-gallery` repo, then go into `sphinx_gallery/tests/tinybuild` and run `make` in git-bash. It should build a little mini site, and hopefully you can see an errant path in one of the RST files it generates, which will lead to an errant path in the images.

kalenkovich, 2020-11-10:
> The first thing to try would be, on a Windows system, clone the `sphinx-gallery` repo, then go into `sphinx_gallery/tests/tinybuild` and run `make` in git-bash. It should build a little mini site, [...]

I tried that. I haven't got any path errors. There were some other errors though. Here is the output:

<details>

```
$ make html
sphinx-build -b html -d _build/doctrees  . _build/html
Running Sphinx v3.3.0
[autosummary] generating autosummary for: examples\future\README.rst, gen_modules\sphinx_gallery.backreferences.rst, gen_modules\sphinx_gallery.docs_resolv.rst, gen_modules\sphinx_gallery.downloads.rst, gen_modules\sphinx_gallery.gen_gallery.rst, gen_modules\sphinx_gallery.gen_rst.rst, gen_modules\sphinx_gallery.notebook.rst, gen_modules\sphinx_gallery.py_source_parser.rst, gen_modules\sphinx_gallery.sorting.rst, index.rst, minigallery.rst
loading intersphinx inventory from https://docs.python.org/3/objects.inv...
loading intersphinx inventory from https://numpy.org/doc/stable/objects.inv...
loading intersphinx inventory from https://matplotlib.org/objects.inv...
loading intersphinx inventory from https://joblib.readthedocs.io/en/latest/objects.inv...
generating gallery...
WARNING: optipng binaries not found, PNG images and thumbnails will not be optimized
WARNING: 'pypandoc' not available. Using Sphinx-Gallery to convert rst text blocks to markdown for .ipynb files.
C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\examples\plot_scraper_broken.py failed to execute correctly: Traceback (most recent call last):
  File "c:\users\evgenii\miniconda3\envs\mnedev\lib\site-packages\sphinx_gallery\scrapers.py", line 333, in save_figures
    rst = scraper(block, block_vars, gallery_conf)
  File "C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\conf.py", line 18, in __call__
    return matplotlib_scraper(block, block_vars, gallery_conf, **kwargs)
  File "c:\users\evgenii\miniconda3\envs\mnedev\lib\site-packages\sphinx_gallery\scrapers.py", line 149, in matplotlib_scraper
    fig.savefig(image_path, **these_kwargs)
  File "C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\conf.py", line 44, in <lambda>
    Figure.savefig = lambda *args, **kwargs: np.min([])
  File "<__array_function__ internals>", line 5, in amin
  File "c:\users\evgenii\miniconda3\envs\mnedev\lib\site-packages\numpy\core\fromnumeric.py", line 2830, in amin
    return _wrapreduction(a, np.minimum, 'min', axis, None, out,
  File "c:\users\evgenii\miniconda3\envs\mnedev\lib\site-packages\numpy\core\fromnumeric.py", line 87, in _wrapreduction
    return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
ValueError: zero-size array to reduction operation minimum which has no identity

generating gallery for auto_examples... [100%] plot_numpy_matplotlib.py
generating gallery for auto_examples\future... [ 50%] plot_future_imports_brokenC:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\examples\future\plot_future_imports_broken.py failed to execute correctly: Traceback (most recent call last):
  File "C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\examples\future\plot_future_imports_broken.py", line 18, in <module>
    raise RuntimeError('Forcing this example to fail on Python 3')
RuntimeError: Forcing this example to fail on Python 3

generating gallery for auto_examples\future... [100%] plot_future_imports.py
computation time summary:
    - examples\plot_animation.py:                      5.06 sec   0.0 MB
    - examples\plot_pickle.py:                         0.54 sec   0.0 MB
    - examples\plot_matplotlib_alt.py:                 0.48 sec   0.0 MB
    - examples\plot_numpy_matplotlib.py:               0.13 sec   0.0 MB
    - examples\plot_svg.py:                            0.12 sec   0.0 MB
    - examples\plot_command_line_args.py:              0.07 sec   0.0 MB
    - examples\plot_scraper_broken.py:                 0.01 sec   0.0 MB
    - examples\future\plot_future_imports_broken.py:   0.00 sec   0.0 MB
    - examples\plot_log.py:                            0.00 sec   0.0 MB
    - examples\plot_repr.py:                           0.00 sec   0.0 MB
    - examples\future\plot_future_imports.py:          0.00 sec   0.0 MB
    - examples\plot_second_future_imports.py:          0.00 sec   0.0 MB
    - examples\local_module.py:                        0.00 sec   0.0 MB
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 11 source files that are out of date
updating environment: [new config] 27 added, 0 changed, 0 removed
reading sources... [100%] minigallery
looking for now-outdated files... none found
pickling environment... done
checking consistency... C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\minigallery.rst: WARNING: document isn't included in any toctree
done
preparing documents... done
writing output... [100%] minigallery
C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\auto_examples\index.rst:30: WARNING: toctree contains reference to document 'auto_examples/local_module' that doesn't have a title: no link will be generated
C:\Users\evgenii\OneDrive\projects\sphinx-gallery\repo\sphinx_gallery\tests\tinybuild\auto_examples\index.rst:30: WARNING: toctree contains reference to document 'auto_examples/local_module' that doesn't have a title: no lin [... cut ...] bedding documentation hyperlinks for auto_examples... [ 62%] plot_scraper_brokembedding documentation hyperlinks for auto_examples... [ 68%] plot_second_futurembedding documentation hyperlinks for auto_examples... [ 81%] sg_execution_timeembedding documentation hyperlinks for auto_examples... [ 87%] plot_future_imporembedding documentation hyperlinks for auto_examples... [ 93%] plot_future_imporembedding documentation hyperlinks for auto_examples... [100%] sg_execution_times.html

```
</details>

larsoner, 2020-11-10:
> I tried that. I haven't got any path errors. There were some other errors though. Here is the output:

I did, too :) It's what brought me around to [making the fourth-point suggestion](https://github.com/mne-tools/mne-python/pull/8502#issuecomment-724792976) in #8502

PR #8502 (2020-11-12): DOC: get docs to build on Windows
#### Reference issue
Fixes #8501

#### What does this implement/fix?
There are a few problems preventing docs from building on Windows. This PR is intended to fix them. As of now, it solves three out of four problems described in #8501.
closed 2020-11-12 by larsoner