mne-tools/mne-python #8505: matplotlib 3.3.3

GuillaumeFavelier, 2020-11-12:
`matplotlib 3.3.3` has been released recently and now [I notice](https://github.com/mne-tools/mne-python/pull/8495#issuecomment-725995183):

```
================================== FAILURES ===================================
________________________________ test_stc_mpl _________________________________
mne\viz\tests\test_3d.py:499: in test_stc_mpl
    stc.plot(subjects_dir=subjects_dir, time_unit='s', views='ven',
mne\source_estimate.py:653: in plot
    brain = plot_source_estimates(
<decorator-gen-139>:24: in plot_source_estimates
    ???
mne\viz\_3d.py:1803: in plot_source_estimates
    return _plot_mpl_stc(stc, spacing=spacing, **kwargs)
mne\viz\_3d.py:1529: in _plot_mpl_stc
    _smooth_plot(initial_time, params)
mne\viz\_3d.py:1437: in _smooth_plot
    facecolors = polyc._facecolors3d
E   AttributeError: 'Poly3DCollection' object has no attribute '_facecolors3d'
```

Should we pin it or does someone knows a fix?

I try e0cbc74 on https://github.com/mne-tools/mne-python/pull/8495

cbrnr, 2020-11-12:
Did you find if they changed something from 3.3.2 to 3.3.3? The [release notes](https://github.com/matplotlib/matplotlib/releases/tag/v3.3.3) mention "Fix inconsistent color mapping in scatter for 3D plots" - is that it? I wish they'd link to the GH PR in their changelog...

larsoner, 2020-11-12:
We were using private attributes, #8495 stops doing so

cbrnr, 2020-11-12:
I see. So this will soon work I assume? Pinning to < 3.3.3 would be a pretty bad solution, because 3.3.3 is the first version with a binary wheel for Python 3.9 and we should start adding support for that version.

larsoner, 2020-11-12:
Yes #8495 should work, just waiting for CI confirmation

closed 2020-11-12 by larsoner