mne-tools/mne-python #8510: fif file extension not required when writing FIFF files

bloyl, 2020-11-12:
I tried a `raw.save('test.h5')` sort of by accident and while it complained that I wasn't using standard `_raw.fif` conventions it happily saved a fif file with an h5 extension. It took me awhile to realize that this wasn't actually an h5 file (I had hoped the h5 support had been extended).

I'm not sure this is a _bug_, as there isn't really the expectation that it should have written an h5 file but it was a bit confusing.

Is there a reason we don't require a `.fif` ending for files that are always written as fiff?

seems like it would be simple to use the `endings_err` parameter to `check_fname` 
https://github.com/mne-tools/mne-python/blob/1fa17ddf895b42c08aea3120d127572834ac5c96/mne/utils/check.py#L35

larsoner, 2020-11-12:
I think it should probably emit a warning

bloyl, 2020-11-12:
ok it currently does issue a warning, so I'll close.

agramfort, 2020-11-12:
but the warning is maybe misleading. I feel a clearer warning should appear
if you don't use a .fif or .fif.gz extension as it's very like a bug

>

larsoner, 2020-11-12:
Or we could add it to `endings_err` as @bloyl suggests, people can get old behavior with a `shutil.movefile` if they need it. I guess this is better than a warning given the likelihood of error versus legitimate use.

@bloyl do you want to make a PR for this? It should "just" be a one-liner plus two lines of `with pytest.raises` context and `raw.save(...)`...

bloyl, 2020-11-12:
I'll put together a PR over the weekend hopefully.
FYI `git grep` suggests there are 22 `save` methods that could potentially use something like this.
there is also a `mne.utils.check_fname` and an `mne.utils._check_fname` which do different things :)

vagechirkov, 2021-03-15:
I'll work on this issue (sprint 2021) :)

PR #9062 (2021-03-16): FIX saving fif file with the wrong extension
Fixes #8510.

Restrict saving options of the fif file to .fif and .fif.gz and add the test for this error raising.
closed 2021-03-16 by larsoner