mne-tools/mne-python #8512: Source depth for vertices

olafhauk, 2020-11-13:
#### Describe the new feature or enhancement
I would like to compute a metric of "source depth" for vertices in source space. This could be the Euclidean distance of a vertex to the nearest sensor/electrode or the nearest BEM surface.

#### Describe your proposed implementation
I've got something like this in mind (for source_space.py):

def vertex_depth(src, reference, mode='dist'):
    """Compute depth of vertices in source space.

    Parameters
    ----------
    src : instance of Source Space
        The vertices for which depth to be computed.
    reference : measurement info or BEM surfaces
        The locations with respect to which depth shall be computed.
    mode : 'dist'
        The distance metric to be computed.

    Returns
    -------
    vertex_depth: array, shape (n_vertices)
        The distance of each vertex with respect to the reference locations.
    """

#### Describe possible alternatives

#### Additional comments
I would need some advice getting relevant sensor and BEM locations with the appropriate transformations.

larsoner, 2020-11-13:
+1 for this. Thoughts:

1. Do we even need `mode`? Do people want something other than the minimum Euclidean distance to a sensor / any surface vertex? Maybe we can omit it to start and then see if people want other options, unless you already see a use case.
2. The API will probably need a `trans` argument to deal with head vs MRI coordinate frames.
3. I might just shorten `"reference"` to `"to"`, or `"other"` or something. "reference" is used a lot in MNE to talk about EEG reference options so it's better to avoid it here if there is a reasonable alternative.

@ckiefer0 @wmvanvliet @britta-wstnr recently needed something like this, so they might also have ideas.

olafhauk, 2020-11-13:
Great. ```mode``` could also include options like average distance or max distance.

wmvanvliet, 2020-11-13:
I've got some code for this: https://github.com/AaltoImagingLanguage/conpy/blob/master/conpy/forward.py#L27

It selects all vertices within a certain distance from any sensor. In order to make this selection, it has to compute the distance between each vertex and the closest sensor. Deals with transforms along the way.

olafhauk, 2020-11-13:
Thanks @wmvanvliet. I'll use your example and add you as co-author if that's ok. I'll start with distances to sensors, BEM surfaces can be included later.

larsoner, 2021-01-07:
Fixed by #8534

closed 2021-01-07 by larsoner