mne-tools/mne-python #8513: Option to disable tqdm (crashes on Matlab/Windows10)

ftadel, 2020-11-13:
The Matlab-Python integration works beautifully for most applications now, however there are some glitches left, especially in the way the standard streams are handled on Windows. As known unsupported features, Mathworks lists `py.input` and `py.raw_input`: https://fr.mathworks.com/help/matlab/matlab_external/limitations-to-python-support.html
But I just found out that there are also other things that do not work correctly, like the tqdm progress bars.

Minimal example to reproduce the bug, suggested by @larsoner : 
```
clear all; close all; home
tqdm = py.importlib.import_module('tqdm');
cell(py.list(tqdm.tqdm(py.str('10 11 12').split())))
```

It works on WSL(Win10)/Ubuntu18 + Matlab 2018b + Python 3.6 + tqdm 4.51.0:
```
ans =
  1×3 cell array
    {1×2 py.str}    {1×2 py.str}    {1×2 py.str}
```

It fails on Windows10 + Matlab 2019a/2020a/2020b + Python 3.8 + tqdm 4.51.0:
```
cell(py.list(tqdm.tqdm(py.str('10 11 12').split())))
Error using utils>__getattr__ (line 169)
Python Error: AttributeError: 'NoneType' object has no attribute 'write'
Error in std>fp_write (line 329)
Error in std>print_status (line 336)
Error in std>display (line 1520)
Error in std>refresh (line 1386)
Error in std>__init__ (line 1092)
```

This is most likely due to a bug in MATLAB on Windows, therefore you could decide that it is 2x not your problem :)
But since there are tons of forum posts about broken tqdm progress bars in various non-tty contexts, I think it might be a relevant question even for non-matlab/non-windows users.
How complicated for you would it be to add a global flag to disable tqdm in MNE-Python? 
@larsoner suggested something like `MNE_TQDM='off'` on gitter. 

Note that I'm using CPython instead of Anaconda (as recommended in your install doc), because the integration with Matlab is much smoother. Matlab+CPython really works out-of-the-box now. FYI: the notes I'm preparing as I move forward:
https://neuroimage.usc.edu/brainstorm/MnePython

As a side request: it would be great to add an option to skip the validation of the licenses when downloading, and not via `argv`. I see that there is an option to accept the license with a flag `--accept-brainstorm-license` on the command line, but this means calling python via a system call, which is not exactly what we want here (the python environment configured in Matlab is not necessarily the one available from the system path of the process)
And same goes for anything that uses `input()`, actually.

Thanks!
Francois

larsoner, 2020-11-13:
> MNE_TQDM='off'

This would be easy enough to add and support. Looking at the traceback another option would be to quickly test it with something like `fp.write('')` in a try/except and just not print anything if it fails. I can make a quick PR for this and you can test it out.

>  this means calling python via a system call, which is not exactly what we want here (the python environment configured in Matlab is not necessarily the one available from the system path of the process)

Technically you can set `sys.argv` to whatever you want -- it's just a list so you can put the `accept` string in there programatically. But I agree it's not the cleanest interface and I wouldn't want to force actual end users to do this :)

We can add an `accept=False` default kwarg that means "prompt for answer", and people have to do `accept=True` to accept it. Would that work for you? And I mean this not just as a user of the API, but also as a representative of Brainstorm who can speak to how we are allowed to distribute the Brainstorm datasets -- IIRC we put in this accept-input protocol to force people to read and accept the Brainstorm license.

I was wondering if there was some way to detect if Python is being run within MATLAB, but it looks like another try/except pattern will probably work:
```
>> py.input()
Python Error: EOFError: EOF when reading a line
```
In other words, anywhere in MNE-Python we use `input` we should wrap it in a try/except giving people an informative error. In the case of the datasets, we can tell them to use the `accept` keyword argument. In cases for overwriting files (I think this is the only other place?), we can tell them to pass `overwrite=True`.

I'll probably just wrap all of these changes in a PR, I think it should be easy enough to test. Then if there are more snags in your testing @ftadel we can iterate until it's smooth!

ftadel, 2020-11-13:
> This would be easy enough to add and support. Looking at the traceback another option would be to quickly test it with something like fp.write('') in a try/except and just not print anything if it fails.

Given everything I've read on tqdm yesterday, I'm sure there will be other use cases where it will be interesting to completely turn it off :) 

> We can add an accept=False default kwarg that means "prompt for answer", and people have to do accept=True to accept it. 
> In other words, anywhere in MNE-Python we use input we should wrap it in a try/except giving people an informative error.

For solving both problem with a very simple fix, maybe you could simply add a try/except around the `input()` call, and if it fails, consider it's worth `answer = 'y'`? 
This is a bug, and hopefully it will be fixed in Matlab at some point. Maybe this doesn't need to be handled too carefully, these are very rare use cases anyway, maybe I'm the only person who will ever download the MNE datasets from Matlab using a Python call. And if this fails, it means that the user is working in Matlab, and have probably already accepted the Brainstorm license in other ways...

> I was wondering if there was some way to detect if Python is being run within MATLAB, but it looks like another try/except pattern will probably work: `py.input()`

Considering that this problem will (hopefully) disappear soon, it might be better to find another way. 
Is there a way from Python to access its parent process?

larsoner, 2020-11-13:
> Given everything I've read on tqdm yesterday, I'm sure there will be other use cases where it will be interesting to completely turn it off :)

Fair enough, I'll just implement `'off'` env var support then.

> For solving both problem with a very simple fix, maybe you could simply add a try/except around the input() call, and if it fails, consider it's worth answer = 'y'?

If you (as a Brainstorm person) are okay with us bypassing the license message when `input` is unavailable then I'm fine with implementing this.

> Is there a way from Python to access its parent process?

This works but requires `psutil`:
```
>> py.psutil.Process(py.os.getppid()).name

ans = 

  Python str with no properties.

    matlab
```
Can you see if it works on your machine?

I actually think the `try/except` is more general, though, since it should work on any platform where `input` is unavailable for any reason. And we can catch just `EOFError` so we can be fairly precise with our check. So I'd prefer to go that route rather than rely on `psutil` (and add it as a an optional requirement).

ftadel, 2020-11-13:
> Fair enough, I'll just implement 'off' env var support then.

Great, thanks!

> If you (as a Brainstorm person) are okay with us bypassing the license message when input is unavailable then I'm fine with implementing this.

Sure, go for it.

> ```
> >> py.psutil.Process(py.os.getppid()).name
> ```

Works from a Python console, but not from Matlab:

```
>> py.psutil.Process(py.os.getppid())
Error using __init__>_init (line 365)
Python Error: NoSuchProcess: psutil.NoSuchProcess no process found with pid 4740
Error in __init__ (line 324)
```

Ok, let's drop this idea.

> I actually think the try/except is more general, though, since it should work on any platform where input is unavailable for any reason. And we can catch just EOFError so we can be fairly precise [...]

I thought you were talking about a more formal way of identifying if the Python is called from a Matlab context, for more general purposes than we discussed here. But that's not needed anywhere so far. The try/except around `input()` sounds good.

I'll let you know if I find other functions that require some extra care for Matlab.

agramfort, 2020-11-13:
nice !

that's great to see how mne-python core algorithms can be made available
to the matlab world !

>

PR #8515 (2020-11-14): MRG, ENH: Improve MATLAB compat
Closes #8513

@ftadel can you give this branch a shot? Marking as `backport-candidate` so that we can backport then release 0.21.2 to make things easier once we get everything working (hopefully quickly!).
closed 2020-11-14 by larsoner