mne-tools/mne-python #8517: Spherical searchlights as labels for all vertices in source space

olafhauk, 2020-11-13:
#### Describe the new feature or enhancement
I would like to use spherical searchlights as labels for all vertices in source space to compute PSFs/CTFs for sources with spatial extent (rather than point sources). I tried to use mne.grow_labels(), but it produces the error "NotImplementedError: Can't produce more than 1024 unique colors". 

#### Describe your proposed implementation
Could one implement an option like '''colors=False''' to prevent the error message?

#### Describe possible alternatives
Possible workaround: One can use list comprehension to call grow_labels per vertex/seed, but I don't see a reason to limit it to 1024 colors.

#### Additional comments

larsoner, 2020-11-13:
Yes having `colors` as a limitation seems silly. I would add a `colors=None` argument that by default does what is done now, and allow `ndarray, shape ([n_seeds, ][3])` that allows setting the color of the labels manually. The `n_seeds` and `3` are both optional because it should be any array that broadcasts to shape `(n_seeds, 3)` so for example you can use `colors=0.` to get all black labels.

olafhauk, 2020-11-13:
@larsoner I've just pushed #8519. Please let me know if that's what you had in mind.

PR #8519 (2020-11-17): WIP: added colors option to grow_labels
Closes #8517

#### What does this implement/fix?
Added ```colors``` option to mne.grow_label() to avoid error message for large number of seeds.

#### Additional information
closed 2020-11-17 by agramfort