mne-tools/mne-python #8518: MAINT: CI

larsoner, 2020-11-13:
A few things we should do for CIs:

1. Migrate away from Travis (#8514)
2. Add `[skip azp]` [support](https://github.com/mne-tools/mne-python/pull/8514#issuecomment-726797122)
3. Add [inline annotations](https://github.com/mne-tools/mne-python/pull/8514#issuecomment-726797122) for codespell and flake
4. Add 3.9 CI. For this I have no idea when VTK will ship a wheel, so I might just build a Linux one locally and upload it somewhere for us to use, assuming tests actually pass when using it.

larsoner, 2020-11-13:
cc @GuillaumeFavelier feel free to take care of 2/3/4 as part of #8514 if you want or we can do it in separate PRs. (1) is really the critical thing to alleviate CI pressure anyway, so maybe best to do that first.

GuillaumeFavelier, 2020-11-18:
I think we're done here too, feel free to re-open if I missed something.

Closing through https://github.com/mne-tools/mne-python/pull/8533, https://github.com/mne-tools/mne-python/pull/8514/

closed 2020-11-18 by GuillaumeFavelier

larsoner, 2020-11-18:
I think just https://github.com/marketplace/actions/ci-skip-action or something similar. Do you want to try?

GuillaumeFavelier, 2020-11-18:
I'm on it :)

We might be more interested in the `fail-safe` version (which is very verbose):

https://github.com/marketplace/actions/ci-skip-action#fail-safe

And AFAIK, only one `commit-filter` can be used

larsoner, 2020-11-18:
> And AFAIK, only one commit-filter can be used

It seems like it should be easy enough to modify their action to support `str` or `list of str`, want to look into that? From my experience it's actually kind of fun to make the actions...

GuillaumeFavelier, 2020-11-18:
Okay, I'll look into it :+1: