mne-tools/mne-python #8523: Debug console in PyCharm broken with pytest

cbrnr, 2020-11-14:
OK, so I decided to report this issue I'm having here, because it might be something we're doing to the test setup.

I want to debug a pytest run and enter the debug console. When I do that and type any command/name, I get the following error (here I want to debug the function `test_one_channel_gdf` in line 111 of `mne/io/edf/tests/test_gdf.py`:

```
/usr/local/bin/python3.9 /Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/pydevd.py --multiproc --qt-support=auto --client 127.0.0.1 --port 59837 --file /Applications/PyCharm.app/Contents/plugins/python/helpers/pycharm/_jb_pytest_runner.py --target test_gdf.py::test_one_channel_gdf
pydev debugger: process 23317 is connecting
Connected to pydev debugger (build 202.7660.27)
Launching pytest with arguments test_gdf.py::test_one_channel_gdf in /Users/clemens/Repositories/mne-python/mne/io/edf/tests
============================= test session starts ==============================
platform darwin -- Python 3.9.0, pytest-6.1.2, py-1.9.0, pluggy-0.13.1 -- /usr/local/opt/python@3.9/bin/python3.9
cachedir: .pytest_cache
rootdir: /Users/clemens/Repositories/mne-python, configfile: setup.cfg
plugins: cov-2.10.1
collecting ... collected 1 item
test_gdf.py::test_one_channel_gdf >>> ecg
Traceback (most recent call last):
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_console_integration.py", line 84, in get_code_executor
    code_executor = getattr(__builtin__, 'interpreter')
AttributeError: module 'builtins' has no attribute 'interpreter'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 1721, in do_it
    result = pydevd_console_integration.console_exec(self.thread_id, self.frame_id, self.expression, dbg)
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_console_integration.py", line 195, in console_exec
    need_more = exec_code(CodeFragment(expression), updated_globals, updated_globals, dbg)
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_console_integration.py", line 98, in exec_code
    code_executor = get_code_executor()
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_console_integration.py", line 86, in get_code_executor
    code_executor = CodeExecutor()
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_code_executor.py", line 24, in __init__
    self.interpreter = get_pydev_frontend(rpc_client)
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_console_011.py", line 504, in get_pydev_frontend
    _PyDevFrontEndContainer._instance = _PyDevFrontEnd()
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_console_011.py", line 349, in __init__
    self.ipython = PyDevTerminalInteractiveShell.instance(config=load_default_config())
  File "/usr/local/lib/python3.9/site-packages/traitlets/config/configurable.py", line 537, in instance
    inst = cls(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/IPython/terminal/interactiveshell.py", line 525, in __init__
    super(TerminalInteractiveShell, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/IPython/core/interactiveshell.py", line 676, in __init__
    self.init_completer()
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_console_011.py", line 271, in init_completer
    self.Completer = self._new_completer_600()
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_console_011.py", line 229, in _new_completer_600
    completer = PyDevIPCompleter6(shell=self,
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_ipython_console_011.py", line 96, in __init__
    IPCompleter.__init__(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/IPython/core/completer.py", line 1079, in __init__
    warnings.warn('The `use_readline` parameter is deprecated and ignored since IPython 6.0.',
DeprecationWarning: The `use_readline` parameter is deprecated and ignored since IPython 6.0.
```

Here's how it looks like in PyCharm (click on the "Play" icon and select "Debug 'pytest for test_gdf.py...'", you might have to select `pytest` as the default test runner in Preferences - Tools - Python Integrated Tools - Testing):

[image]

When I create a new file (e.g. `test.py` in the project root) with the following test, everything works just fine (meaning that I can start the debugging console and use it without any errors) - that's why I'm assuming that something with the way MNE creates tests or some relevant objects might be the problem.

[image]

Any ideas? I'd be super happy to solve this issue because it keeps haunting me (see e.g. https://youtrack.jetbrains.com/issue/PY-40703, which was never resolved because a MWE doesn't trigger the issue).

larsoner, 2020-11-14:
> Traceback (most recent call last):
>  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydevd_bundle/pydevd_console_integration.py", line 84, in get_code_executor [...]

This part looks like Pycharm error, but maybe they are trying to catch it...

> DeprecationWarning: The `use_readline` parameter is deprecated and ignored since IPython 6.0.

This part looks like it might be due to us treating warnings as errors. You can try temporarily removing the `error::` from here:

https://github.com/mne-tools/mne-python/blob/master/mne/conftest.py#L64

And see if it works. If it does, then a more targeted line (or set of lines if there is more than one of these being caught) like `ignore:.*deprecated and ignored since IPython.*:DeprecationWarning` could be added instead of removing `error::`.

cbrnr, 2020-11-14:
Thanks @larsoner, that was it! See #8524.

PR #8524 (2020-11-14): Don't treat IPython warning as error (fixes debugging in PyCharm)
Fixes #8523.
closed 2020-11-14 by cbrnr