mne-tools/mne-python #8525: BUG: Big Sur compat

larsoner, 2020-11-14:
More of an FYI than anything else -- PyQt5 appears not to work on Big Sur. I've tried 5.12 and 5.15.1, and both basically hang as soon as you make a mayavi, PyVista, or matplotlib Qt5Agg window.

So those thinking about upgrading, don't do so if you want to be able to use Python. For matplotlib at least we can use the MacOSX backend in the meantime, but 3D plotting is now (hopefully temporarily) totally broken for people who have upgraded. :(

Or maybe I'm the only one with this issue. If someone else upgrades, feel free to try it out and let me know!

larsoner, 2020-11-14:
Closing because there isn't really anything for us to do. But we can keep any discussion about this here as things progress

cbrnr, 2020-11-14:
Thanks for the heads up. I think this time I will upgrade as late as possible (this is the first time I haven't upgraded on day 1, and just like the new CPUs the macOS update doesn't really hype me).

christianbrodbeck, 2020-11-22:
`os.environ['QT_MAC_WANTS_LAYER'] = '1'` ([stackoverflow](https://stackoverflow.com/a/64856281/166700)) at least gets the GUIs to open, will do some more testing.

cbrnr, 2020-11-28:
I've done the upgrade and of course things don't work.
https://github.com/numpy/numpy/issues/17784

hoechenberger, 2020-11-29:
I have been running Big Sur for over a week or so and no problems besides the Qt this be. I'm using conda-forge.

cbrnr, 2020-11-29:
Yes, it's a pip issue. I know we recommend/support conda, but pip is still the default package manager and currently that doesn't work.

hoechenberger, 2020-11-29:
Yep. Didn't mean to suggest otherwise, just wanted to share that it works well for me with `conda-forge` :)

cbrnr, 2020-12-04:
The latest releases of pip and Homebrew fixed all issue, so everything is working on macOS/Homebrew.

closed 2020-12-04 by cbrnr

hoechenberger, 2021-01-25:
I'm constantly running into this issue if I forget to update my VS Code Run/Debug configuration accordingly. I **thought** we had added a workaound (setting the variable on macOS automatically) a while back, but cannot find any reference anymore?

hoechenberger, 2021-01-25:
Never mind, it's mentioned right above:
https://github.com/mne-tools/mne-python/pull/8554

But like I said, this apparently doesn't fix the problem for me.

larsoner, 2021-01-25:
> But like I said, this apparently doesn't fix the problem for me. Only setting the variable manually has helped so far.

That PR only fixed it for `mne coreg` and related GUIs that use traits+mayavi, not for code that uses matplotlib. For that, people should look to a matplotlib-generic solution such as setting this env var in their dev environment.

cbrnr, 2021-01-25:
@hoechenberger are you using an old PyQt5 version? This has been fixed in 5.15.2.

PR #8554 (2020-11-23): FIX:  qt5 macOS 11 compatibility
#### Reference issue
Fixes #8525 by setting the environment variable before launching GUIs.

#### TODO
 - [x] test on macOS 10