mne-tools/mne-python #8530: 3D example does not work on Windows 7

jasminetan6032, 2020-11-16:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should ask for help on one of our other channels:

- [email list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
- [Gitter (chat)](https://gitter.im/mne-tools/mne-python)

Our GitHub issue tracker is only used to report bugs and suggest improvements
to MNE-Python. For any other questions, please use the email list or Gitter.
Usage questions that are posted as GitHub issues are usually closed without
being answered. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
please **delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block (see
[the GitHub guide to markdown](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown)
for help with issue formatting). Alternatively, you can put your MWE in a
[public gist](https://gist.github.com) and link to it in this issue.

#### Describe the bug
```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-6-5903cabfd600> in <module>
----> 1 mne.viz.set_3d_backend('pyvista')
      2 for method, stc in zip(['DICS', 'LCMV', 'dSPM'],
      3                        [stc_dics, stc_lcmv, stc_dspm]):
      4     title = '%s source power in the 12-30 Hz frequency band' % method
      5     brain = stc.plot(hemi='rh', subjects_dir=subjects_dir,

<decorator-gen-152> in set_3d_backend(backend_name, verbose)

~\Anaconda3\envs\mne-new\lib\site-packages\mne\viz\backends\renderer.py in set_3d_backend(backend_name, verbose)
    114     _check_option('backend_name', backend_name, VALID_3D_BACKENDS)
    115     if MNE_3D_BACKEND != backend_name:
--> 116         _reload_backend(backend_name)
    117         MNE_3D_BACKEND = backend_name
    118 

~\Anaconda3\envs\mne-new\lib\site-packages\mne\viz\backends\renderer.py in _reload_backend(backend_name)
     29 def _reload_backend(backend_name):
     30     global backend
---> 31     backend = importlib.import_module(name=_backend_name_map[backend_name],
     32                                       package='mne.viz.backends')
     33     logger.info('Using %s 3d backend.\n' % backend_name)

~\Anaconda3\envs\mne-new\lib\importlib\__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap.py in _load_unlocked(spec)

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap_external.py in exec_module(self, module)

~\Anaconda3\envs\mne-new\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\Anaconda3\envs\mne-new\lib\site-packages\mne\viz\backends\_pyvista.py in <module>
     33     from pyvista import Plotter, PolyData, Line, close_all, UnstructuredGrid
     34     try:
---> 35         from pyvistaqt import BackgroundPlotter  # noqa
     36     except ImportError:
     37         from pyvista import BackgroundPlotter

~\Anaconda3\envs\mne-new\lib\site-packages\pyvistaqt\__init__.py in <module>
      1 """PyVista package for 3D plotting and mesh analysis."""
      2 from pyvistaqt._version import __version__
----> 3 from pyvistaqt.plotting import BackgroundPlotter, MainWindow, QtInteractor
      4 
      5 __all__ = ["__version__", "BackgroundPlotter", "MainWindow", "QtInteractor"]

~\Anaconda3\envs\mne-new\lib\site-packages\pyvistaqt\plotting.py in <module>
     49 import scooby
     50 import vtk
---> 51 from PyQt5 import QtCore, QtGui
     52 from PyQt5.QtCore import QTimer, pyqtSignal
     53 from PyQt5.QtWidgets import QAction, QFrame, QMenuBar, QVBoxLayout

~\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__.py in <module>
     47 
     48 
---> 49 find_qt()
     50 del find_qt

~\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__.py in find_qt()
     42 
     43     try:
---> 44         os.add_dll_directory(dll_dir)
     45     except AttributeError:
     46         pass

~\Anaconda3\envs\mne-new\lib\os.py in add_dll_directory(path)
   1107         """
   1108         import nt
-> 1109         cookie = nt._add_dll_directory(path)
   1110         return _AddedDllDirectory(
   1111             path,

OSError: [WinError 127] The specified procedure could not be found: 'C:\\Users\\jasmine\\Anaconda3\\envs\\mne-new\\lib\\site-packages\\PyQt5\\Qt\\bin'
```
​

#### Steps to reproduce

Followed the installation instructions for Windows and ran the plot_evoked_ers_source_power.ipynb. 

Thank you!

hoechenberger, 2020-11-16:
@jasminetan6032 is attending the Pybrain 2020 workshop on MNE and was having this issue today. I couldn't resolve it via a screenshare. I am wondering if it's related to Python 3.8 on Windows?

GuillaumeFavelier, 2020-11-16:
> I couldn't resolve it via a screenshare.

What did you try @hoechenberger ?

@jasminetan6032 could you share the output of:

```py
python -c "import mne; mne.sys_info()"
```

hoechenberger, 2020-11-16:
> What did you try @hoechenberger ?

- verified that PyQt5 was actually installed
- ensured MNE and pyvista is the latest version
- added `mne.viz.set_3d_backend('pyvista')`

hoechenberger, 2020-11-16:
MNE was installed by using the `environment.yml` from the MNE installation instructions

jasminetan6032, 2020-11-16:
(base) C:\Users\jasmine>mne sys_info
Platform:      Windows-7-6.1.7601-SP1
Python:        3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)
]
Executable:    c:\users\jasmine\anaconda3\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:        32.0 GB

mne:           0.21.1
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
numba:         0.43.1
nibabel:       3.2.0
cupy:          Not found
pandas:        0.24.2
dipy:          1.3.0
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

(base) C:\Users\jasmine> python -c "import mne; mne.sys_info()"
Platform:      Windows-7-6.1.7601-SP1
Python:        3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)
]
Executable:    C:\Users\jasmine\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:        32.0 GB

mne:           0.21.1
numpy:         1.16.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=Qt5Agg}

sklearn:       0.20.3
numba:         0.43.1
nibabel:       3.2.0
cupy:          Not found
pandas:        0.24.2
dipy:          1.3.0
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

(base) C:\Users\jasmine>

jasminetan6032, 2020-11-16:
Apologies, I ran it in the new environment Richard helped me set up and this happened. 

(base) C:\Users\jasmine>conda activate mne-new

(mne-new) C:\Users\jasmine>python -c "import mne; mne.sys_info()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\mne\utils\conf
ig.py", line 544, in sys_info
    extra += ' {backend=%s}%s' % (mod.get_backend(), extra)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\__i
nit__.py", line 1188, in get_backend
    return rcParams['backend']
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\__i
nit__.py", line 702, in __getitem__
    from matplotlib import pyplot as plt
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\pyp
lot.py", line 2336, in <module>
    switch_backend(rcParams["backend"])
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\__i
nit__.py", line 703, in __getitem__
    plt.switch_backend(rcsetup._auto_backend_sentinel)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\pyp
lot.py", line 256, in switch_backend
    switch_backend(candidate)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\pyp
lot.py", line 276, in switch_backend
    class backend_mod(matplotlib.backend_bases._Backend):
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\pyp
lot.py", line 277, in backend_mod
    locals().update(vars(importlib.import_module(backend_name)))
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\importlib\__init__.py", line
 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\bac
kends\backend_qt5agg.py", line 11, in <module>
    from .backend_qt5 import (
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\bac
kends\backend_qt5.py", line 16, in <module>
    import matplotlib.backends.qt_editor.figureoptions as figureoptions
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\bac
kends\qt_editor\figureoptions.py", line 11, in <module>
    from matplotlib.backends.qt_compat import QtGui
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\bac
kends\qt_compat.py", line 170, in <module>
    _setup()
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\matplotlib\bac
kends\qt_compat.py", line 75, in _setup_pyqt5
    from PyQt5 import QtCore, QtGui, QtWidgets
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__
.py", line 49, in <module>
    find_qt()
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__
.py", line 44, in find_qt
    os.add_dll_directory(dll_dir)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\os.py", line 1109, in add_dl
l_directory
    cookie = nt._add_dll_directory(path)
OSError: [WinError 127] The specified procedure could not be found: 'C:\\Users\\
jasmine\\Anaconda3\\envs\\mne-new\\lib\\site-packages\\PyQt5\\Qt\\bin'

(mne-new) C:\Users\jasmine>

GuillaumeFavelier, 2020-11-16:
I think there is something going on with this installation of `PyQt5` 🤔 

Still on `mne-new`, my first idea would be to reinstall `PyQt5`, did you try `pip install --force --no-cache PyQt5` 😅 

This is what I use to test my installation:

```py
python -c "from PyQt5.QtWidgets import QApplication, QWidget; app=QApplication([]); w=QWidget(); w.show(); app.exec()"
```

---

I did a quick check on my windows 10 (I did my install with `pip -r requirements.txt` though). Except that I ran into an issue with the latest numpy (I had to downgrade), I did not reproduce this behaviour. I'll try again later with a proper conda install 👍

jasminetan6032, 2020-11-16:
Hi Guillaume,

I tried your recommendations and PyQt5 was successfully installed. But when I tested it, this happened. 

(mne-new) C:\Users\jasmine>pip install --force --no-cache PyQt5
Collecting PyQt5
  Downloading PyQt5-5.15.1-5.15.1-cp35.cp36.cp37.cp38.cp39-none-win_amd64.whl (5
9.4 MB)
     |████████████████████████████████| 59.4 MB 3.3 MB/s
Collecting PyQt5-sip<13,>=12.8
  Downloading PyQt5_sip-12.8.1-cp38-cp38-win_amd64.whl (63 kB)
     |████████████████████████████████| 63 kB 4.5 MB/s
Installing collected packages: PyQt5-sip, PyQt5
  Attempting uninstall: PyQt5-sip
    Found existing installation: PyQt5-sip 12.8.1
    Uninstalling PyQt5-sip-12.8.1:
      Successfully uninstalled PyQt5-sip-12.8.1
  Attempting uninstall: PyQt5
    Found existing installation: PyQt5 5.15.1
    Uninstalling PyQt5-5.15.1:
      Successfully uninstalled PyQt5-5.15.1
Successfully installed PyQt5-5.15.1 PyQt5-sip-12.8.1

(mne-new) C:\Users\jasmine>python -c "from PyQt5.QtWidgets import QApplication,
QWidget; app=QApplication([]); w=QWidget(); w.show(); app.exec()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__
.py", line 49, in <module>
    find_qt()
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\site-packages\PyQt5\__init__
.py", line 44, in find_qt
    os.add_dll_directory(dll_dir)
  File "C:\Users\jasmine\Anaconda3\envs\mne-new\lib\os.py", line 1109, in add_dl
l_directory
    cookie = nt._add_dll_directory(path)
OSError: [WinError 127] The specified procedure could not be found: 'C:\\Users\\
jasmine\\Anaconda3\\envs\\mne-new\\lib\\site-packages\\PyQt5\\Qt\\bin'

Thank you so much!

hoechenberger, 2020-11-16:
Would it be possible for you to entirely uninstall Anaconda, grad a fresh installer and start from scratch? Maybe there is something wrong with Anaconda on your system

larsoner, 2020-11-16:
Yeah this looks a bit like https://github.com/conda/conda/issues/9003

jasminetan6032, 2020-11-16:
The latest installer doesn't work for me (Possibly a Windows 7 problem). I've tried the following installers and I'm still having the same problem. 

Anaconda3-2019.03-Windows-x86.exe
Anaconda3-2019.10-Windows-x86_64.exe
Anaconda3-2020.02-Windows-x86_64.exe

mne sys_info still results in the following error.

```
(mne) C:\Users\jasmine\mne_environment>mne sys_info
Traceback (most recent call last):
  File "c:\users\jasmine\anaconda3\envs\mne\lib\runpy.py", line 194, in _run_mod
ule_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\jasmine\anaconda3\envs\mne\lib\runpy.py", line 87, in _run_code

    exec(code, run_globals)
  File "C:\Users\jasmine\anaconda3\envs\mne\Scripts\mne.exe\__main__.py", line 7
, in <module>
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\mne\commands\utils
.py", line 107, in main
    cmd.run()
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\mne\commands\mne_s
ys_info.py", line 26, in run
    mne.sys_info()
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\mne\utils\config.p
y", line 544, in sys_info
    extra += ' {backend=%s}%s' % (mod.get_backend(), extra)
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\__init_
_.py", line 1188, in get_backend
    return rcParams['backend']
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\__init_
_.py", line 702, in __getitem__
    from matplotlib import pyplot as plt
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\pyplot.
py", line 2336, in <module>
    switch_backend(rcParams["backend"])
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\__init_
_.py", line 703, in __getitem__
    plt.switch_backend(rcsetup._auto_backend_sentinel)
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\pyplot.
py", line 256, in switch_backend
    switch_backend(candidate)
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\pyplot.
py", line 276, in switch_backend
    class backend_mod(matplotlib.backend_bases._Backend):
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\pyplot.
py", line 277, in backend_mod
    locals().update(vars(importlib.import_module(backend_name)))
  File "c:\users\jasmine\anaconda3\envs\mne\lib\importlib\__init__.py", line 127
, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\backend
s\backend_qt5agg.py", line 11, in <module>
    from .backend_qt5 import (
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\backend
s\backend_qt5.py", line 16, in <module>
    import matplotlib.backends.qt_editor.figureoptions as figureoptions
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\backend
s\qt_editor\figureoptions.py", line 11, in <module>
    from matplotlib.backends.qt_compat import QtGui
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\backend
s\qt_compat.py", line 170, in <module>
    _setup()
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\matplotlib\backend
s\qt_compat.py", line 75, in _setup_pyqt5
    from PyQt5 import QtCore, QtGui, QtWidgets
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\PyQt5\__init__.py"
, line 49, in <module>
    find_qt()
  File "c:\users\jasmine\anaconda3\envs\mne\lib\site-packages\PyQt5\__init__.py"
, line 44, in find_qt
    os.add_dll_directory(dll_dir)
  File "c:\users\jasmine\anaconda3\envs\mne\lib\os.py", line 1109, in add_dll_di
rectory
    cookie = nt._add_dll_directory(path)
OSError: [WinError 127] The specified procedure could not be found: 'c:\\users\\
jasmine\\anaconda3\\envs\\mne\\lib\\site-packages\\PyQt5\\Qt\\bin'
```

And the jupyter notebook "Compute evoked ERS source power using DICS, LCMV beamformer, and dSPM" still produces the same error at the last cell.

hoechenberger, 2020-11-19:
> The latest installer doesn't work for me (Possibly a Windows 7 problem).

Mmmh. I suggest you update your operating system ;)

drammock, 2020-11-19:
> I've tried the following installers and I'm still having the same problem.
> 
> Anaconda3-2019.03-Windows-x86.exe
> Anaconda3-2019.10-Windows-x86_64.exe
> Anaconda3-2020.02-Windows-x86_64.exe

[The Anaconda docs say](https://docs.anaconda.com/anaconda/install/#old-os) that only 2019.10 and earlier can be expected to work with Windows 7. But that is not a guarantee; Anaconda's official minimum supported version is Windows 8 ([same is true of Microsoft](https://www.microsoft.com/en-us/windows/windows-7-end-of-life-support-information)).

I have an old virtual machine that runs windows 7 (enterprise edition).  I just fired it up, downloaded the Anaconda3-2019.10-Windows-x86_64.exe installer from their archive page, and ran it. It worked. Then I downloaded the environment file and followed the instructions [here](), which also worked. But then:

<details>
```
(base) C:\Users\dan\Desktop>conda activate mne
(mne) C:\Users\dan\Desktop>mne sys_info
2020-11-19 10:14:00.450 (   1.335s) [                ]vtkWin32OpenGLRenderWin:63
5    ERR| vtkWin32OpenGLRenderWindow (000000000E39AB40): failed to get wglChoose
PixelFormatARB
ERROR:root:failed to get wglChoosePixelFormatARB
2020-11-19 10:14:00.481 (   1.359s) [                ]vtkWin32OpenGLRenderWin:71
3    ERR| vtkWin32OpenGLRenderWindow (000000000E39AB40): failed to get valid pix
el format.
ERROR:root:failed to get valid pixel format.
2020-11-19 10:14:00.528 (   1.406s) [                ]vtkOpenGLRenderWindow.c:56
9    ERR| vtkWin32OpenGLRenderWindow (000000000E39AB40): GLEW could not be initi
alized: Missing GL version
ERROR:root:GLEW could not be initialized: Missing GL version
Windows fatal exception: access violation

Current thread 0x00000c40 (most recent call first):
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\pyvista\plotting\plott
ing.py", line 766 in render
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\pyvista\plotting\plott
ing.py", line 4102 in show
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\pyvista\utilities\erro
rs.py", line 117 in get_gpu_info
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\pyvista\utilities\erro
rs.py", line 130 in __init__
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\mne\utils\config.py",
line 558 in sys_info
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\mne\commands\mne_sys_i
nfo.py", line 26 in run
  File "C:\Users\dan\Anaconda3\envs\mne\lib\site-packages\mne\commands\utils.py"
, line 107 in main
  File "C:\Users\dan\Anaconda3\envs\mne\Scripts\mne-script.py", line 10 in <modu
le>
```
</details>

Similar results if I start python and run `import mne; mne.sys_info()`. I'm afraid @hoechenberger may be right that the best path forward is a Windows OS upgrade (or switch to Linux, it's free!).

closed 2020-11-19 by drammock