mne-tools/mne-python #8544: EDF/EDF+ headers parsed incompletely/incorrectly

a-hurst, 2020-11-19:
When reading in EDF/EDF+/BDF files with MNE, there are a few issues with the current parser:

**Patient/Participant Info**

https://github.com/mne-tools/mne-python/blob/8b0b8a967b1a8e2fd88c47221f535d73ca7d6fb2/mne/io/edf/edf.py#L524-L530

According to the [EDF+ spec](https://www.edfplus.info/specs/edfplus.html#additionalspecs), participant info is stored in the format of "{id} {sex} {birthdate} {name}" (minus the braces), with those four fields being separated by single spaces (and not containing any spaces themselves). At present, MNE only parses the first two fields, and incorrectly identifies the second field as containing the participant name.

**Recording Info**

https://github.com/mne-tools/mne-python/blob/8b0b8a967b1a8e2fd88c47221f535d73ca7d6fb2/mne/io/edf/edf.py#L532-L534

Currently, EDF/BDF recording info isn't parsed at all by MNE, and the string read in by the above lines isn't part of the exported header info so the info also can't be parsed later. As per the EDF+ spec, it takes the format "Startdate {startdate} {admin_code} {technician} {equipment}", also separated by spaces but not containing any.

The 'startdate' field is particularly important to parse if available, because it contains the meas_date for the file with a 4-digit year (e.g. '02-AUG-2014'), whereas the normal 'meas_date' region of the header only retains the last two digits (e.g. '02.08.14'). For regular use this is fine, but it creates issues with the BIDS-EEG spec which recommends the dates of recordings should be shifted to a year before 1925 during anonymization. A solution would be to attempt to parse the start date in the recording info and use the actual 'meas_date' value as a fallback.

**Date**

https://github.com/mne-tools/mne-python/blob/8b0b8a967b1a8e2fd88c47221f535d73ca7d6fb2/mne/io/edf/edf.py#L536-L540

As per the EDF+ spec, the earliest supported year for EDF files is 1985. If a 4-digit year isn't specified in the recording info, the parser should assume that a year of < 85 is a 20xx year and not an 19xx one. This will probably never affect anyone, but it might as well match the official docs.

larsoner, 2020-11-19:
@a-hurst this sounds reasonable to me, would you be up for making a pull request to fix these things?

sappelhoff, 2020-11-20:
crossref to where this came up: https://github.com/mne-tools/mne-bids/pull/636#issuecomment-730556344

sappelhoff, 2020-11-20:
Thanks a lot for opening this detailed issue @a-hurst! I want to add a point for "Date"

### date

> As per the EDF+ spec, the earliest supported year for EDF files is 1985

... and the latest supported year for the EDF "Startdate" field is 2084, because only two digits are allowed for year.

The [spec](https://www.edfplus.info/specs/edfplus.html) says the following on how to deal with years that do not fall into the 1985-2084 range:

> After 2084, yy must be 'yy' and only item 4 of this paragraph defines the date.

Where item 4 of that paragraph says: Look at the "recording info" section for the real date.

This is also great to have for anonymizing dates to a time earlier than 1985. --> We put "yy" as the year into the EDF "Startdate" field, which tells EDF-Readers to please look at the "Recording info" section to get the proper year ... and in that section, the year can be specified with 4 digits.

IMHO this is something we should support.

a-hurst, 2020-11-20:
@larsoner @sappelhoff I started work on a PR, but I don't think I'll have time to write tests or finish it for a while. Here's the modified header reading code in case anyone's up to the task themselves:

```python
    with open(fname, 'rb') as fid:

        fid.read(8)  # version (unused here)
        
        # Patient ID
        patient = {}
        id_info = fid.read(80).decode('ascii').rstrip()
        id_info = id_info.split(' ')
        if len(id_info)[0]:
            patient['id'] = id_info[0]
            if len(id_info) == 4:
                try:
                    birthdate = datetime.strptime(id_info[2], "%d-%b-%Y")
                except ValueError:
                    birthdate = "X"
                patient['sex'] = id_info[1]
                patient['birthday'] = birthdate
                patient['name'] = id_info[3]

        # Recording ID
        meas_id = {}
        rec_info = fid.read(80).decode('ascii').rstrip()
        rec_info = rec_info.split(' ')
        valid_startdate = False
        if len(rec_info) == 5:
            try:
                startdate = datetime.strptime(rec_info[1], "%d-%b-%Y")
                valid_startdate = True
            except ValueError:
                startdate = "X"
            meas_id['startdate'] = startdate
            meas_id['study_id'] = rec_info[2]
            meas_id['technician'] = rec_info[3]
            meas_id['equipment'] = rec_info[4]

        # If startdate available in recording info, use it instead of the
        # file's meas_date since it contains all 4 digits of the year
        if valid_startdate:
            day = meas_id['startdate'].day
            month = meas_id['startdate'].month
            year = meas_id['startdate'].year
            fid.read(8) # skip file's meas_date
        else:
            meas_date = fid.read(8).decode('ascii')
            day, month, year = [int(x) for x in meas_date.split('.')]
            year = year + 2000 if year < 85 else year + 1900

        meas_time = fid.read(8).decode('ascii')
        hour, minute, sec = [int(x) for x in meas_time.split('.')]
        try:
            meas_date = datetime(year, month, day, hour, minute, sec,
                                 tzinfo=timezone.utc)
        except ValueError:
            warn(f'Invalid date encountered ({year:04d}-{month:02d}-'
                 f'{day:02d} {hour:02d}:{minute:02d}:{sec:02d}).')
            meas_date = None
```

Since this code only tries to parse the 8-byte startdate if a more precise startdate isn't found in the recording info section, I think it satisfies @sappelhoff's request of supporting the 'yy' part of the spec.

My only other change was adding `recording_info=meas_id,` to the `edf_info.update()` call further down so the recording info gets carried over to the final Raw object.

larsoner, 2020-11-23:
@a-hurst can you open a PR with this code, plus your name in `names.inc` and a change line in `latest.inc`:

https://mne.tools/dev/install/contributing.html#first-time-contributors

That would be enough for a maintainer to finish your work but still get you credit for it -- as long as you have the "allow edits from maintainers" checked (which is the default) in the PR, we can push commits to help

larsoner, 2020-11-23:
... or if you aren't sure where to put it or have time for the GitHub gymnastics, your name + a URL to link to would also work

closed 2020-11-24 by larsoner

Teuniz, 2020-12-04:
In EDF+, startdates before 1985 are NOT allowed.  
Startdates from 2085 and later, are allowed in EDF+ only but it brakes backward compatibility with EDF.  
In EDF and EDF+, the only correct "solution" for anonymization purposes is  
to use a startime of 00:00:00 and startdate 1985-01-01.  
  
FYI next versions of EDFbrowser and EDFlib will refuse to open a recording when  
the startdate of the recording is not in the range 1985 - 2084 inclusive.  
  
I wish there was a better solution but there isn't.  
  
Kind Regards,  
  
Teunis van Beelen

sappelhoff, 2020-12-04:
Thanks Teunis,

maybe for context for others, I asked @Teuniz (as an EDF "authority") about anonymization in https://gitlab.com/Teuniz/EDFlib/-/issues/5

agramfort, 2020-12-04:
I foresee some issues with BIDS standard no?

>

sappelhoff, 2020-12-04:
> I foresee some issues with BIDS standard no?

yes ... we have now agreed in MNE-BIDS to write the dates to `1985` if we anonymize them. It's the best we can do given the EDF standard. :-( discussion is here: https://github.com/mne-tools/mne-bids/pull/636#issuecomment-731101078

PR #8558 (2020-11-24): MRG, BUG: Fix EDF header parsing
1. Implements a variant of @a-hurst's code from #8544 (most of the PR)
2. Fixes dipy decorator (1 line)
3. Fixes str-as-int parsing from `gitter` (1 line)
4. Adds stub test with `XXX` for carrying over `edf_info['subject_info']` to `raw.info['subject_info']`, which we should probably do to the extent possible (but in another PR)

@a-hurst the one thing I didn't follow was (what is keeping this WIP from my end):

> My only other change was adding recording_info=meas_id, to the edf_info.update() call further down so the recording info gets carried over to the fin [... cut]