mne-tools/mne-python #8550: Unnecessary FileNotFoundError in Persyst Reader

pmyers16, 2020-11-20:
#### Describe the bug
The Persyst reader unnecessarily throws a FileNotFound error when both the .lay and .dat
files have been moved to a new directory. The reader attempts to account for this by replacing
the dirname of the .dat file's path with the dirname of the .lay file, but still throws the error if 
the path to the .dat file as listed in the .lay file's "file" field does not exist. 

This issue does not occur if the File field is a relative path. I am not sure under what circumstances
the File field is an absolute path.

#### Steps to reproduce

1. Download the Persyst files from [mne testing data](https://github.com/mne-tools/mne-testing-data) and save both files to the same folder, say "C:/Users/<username>/persyst_data"

2. Change line 2 of the .lay file to an absolute path that does not actually exist, say "C:/Users/<username>/persyst_data/other_folder/sub-pt1_ses-02_task-monitor_acq-ecog_run-01_clip2.dat"

3. Run the following code:
`lay_fname = "C:/Users/<username>/persyst_data/sub-pt1_ses-02_task-monitor_acq-ecog_run-01_clip2.lay"`
`rawPersyst = mne.io.read_raw_persyst(lay_fname)`

#### Expected results
The code snippet should not fail since `read_raw_persyst` correctly identifies that the .dat file has moved directories along with the .lay file. The reader should succeed for **either** of the two following scenarios:

1. The path listed in the 'File' field in the .lay file exists
OR
2.  Both the .lay and .dat files have been moved to the same directory (i.e. the dirname in the 'File' field is ignored)

If neither of these two scenarios occurs, the error message should say this, listing both paths checked, not the path to the .lay file that does exist.
 
#### Actual results
The following FileNotFoundError: 

>   File "D:\ProgramFiles\Anaconda\envs\ezscalp\lib\site-packages\mne\io\persyst\persyst.py", line 39, in read_raw_persyst
>     return RawPersyst(fname, preload, verbose)
>   File "<decorator-gen-215>", line 24, in __init__
>   File "D:\ProgramFiles\Anaconda\envs\ezscalp\lib\site-packages\mne\io\persyst\persyst.py", line 101, in __init__
>     raise FileNotFoundError(error_msg)
> FileNotFoundError: The data path you specified does not exist for the lay path,

#### Additional information
Output of mne.sys_info(): 

> D:\ProgramFiles\Anaconda\envs\ezscalp\python.exe D:/Desktop/convert_persyst_bids/ezpreprocess/convert_testing_data.py
> Platform:      Windows-10-10.0.19041-SP0
> Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
> Executable:    D:\ProgramFiles\Anaconda\envs\ezscalp\python.exe
> CPU:           AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD: 24 cores
> Memory:        31.9 GB
> 
> mne:           0.21.2
> numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
> scipy:         1.5.0
> matplotlib:    3.3.2 {backend=Qt5Agg}
> 
> sklearn:       0.23.2
> numba:         0.51.2
> nibabel:       3.2.0
> cupy:          Not found
> pandas:        1.1.3
> dipy:          1.2.0
> mayavi:        4.7.2
> pyvista:       0.26.1 {OpenGL 4.5.0 NVIDIA 432.00 via GeForce RTX 2070 SUPER/PCIe/SSE2}
> vtk:           9.0.1
> PyQt5:         5.9.2
> None

hoechenberger, 2020-11-22:
Is your proposed solution following the Persyst file format specification? I'm always a little concerned if we're trying to be too clever... I've never used the Persyst format, but e.g. with BrainVison you will have to change the metadata as well if you rename the files. But things might be different here, I don't know?

pmyers16, 2020-11-23:
Yes, the Persyst file format requires the "File" field to be either a full path or just the file name. In the latter case, the corresponding .dat file must be in the same folder as the .lay file. The current reader attempt to handle both cases, but fails in the scenario described above.

If the requirement is that the metadata needs to be manually changed when the file is moved, the reader can be simplified and the error message can still be updated.

adam2392, 2020-11-25:
Hmm yes it seems undesirable to "require" that the metadata be adapted when say a file is moved to a different directroy. Instead what would be ideal is if the `File` field just encodes the "filename", and then it should be in the same directory as the corresponding `.lay` file. 

This isn't as "expansive" as the Persyst spec format, but it seems like an okay assumption. 

Perhaps you can make a PR to:

1. read in the .lay file and check the `File` field for the corresponding `.dat` filename (even if it is a path).
2. if `.dat` file is in the same directory as the `.lay` file, read in successfully.
3. if `.dat` file is not, error out. If it is a fully specified path, specify an error saying that we require the files to be in the same directory as each other, else just say it doesn't exist.

^ @hoechenberger wdyt?

PR #8806 (2021-02-01): [MRG] Fix persyst reader - adds test for multiple repeated Comments, allow commas and require lay and dat file to be in same directory
#### Reference issue
Closes: #8311
Closes: #8550 

Initial strategy was taken from #8311, and then I simplified the approach.

#### What does this implement/fix?
Adds unit test for repeated comments, and comments with `,` characters.

In addition, cleans up the logic and explicitly requires `.lay` and `.dat` files to be in the same directory.

#### Additional information
Requires `mne-testing-data` to be updated to 0.114
closed 2021-02-01 by larsoner