mne-tools/mne-python #8555: Why mne.decode.CSP error after artifact removal with ICA? & How to normalize EEG data by using MNE?

palm-audtachai, 2020-11-23:
I have 2 question.
1. Why mne.decode.CSP error after artifact removal with ICA? 
I know process of EEG processing that we need to remove artifact before feature extraction so I do follow with BCI competition V dataset. after I remove artifact with ICA and use mne.decode.CSP. it has an error like this

[image]
[image]

but if I delete or comment codes of artifact removal with ICA and run again. it hasn't an error anymore. so I want to know why.

2. How to normalise EEG data by using MNE?
Is it has codes for normalise data? If it hasn't, Does it has other ways? 

Thank you for all answers.

agramfort, 2020-11-23:
if you ICA clean your data you reduce their rank making covariance matrices
not full rank (they have some eigenvalues which are 0).
This is the error you see. What you could do is apply a PCA step before CSP
to reduce the dimension of the data so the covariances
become full rank.

>

palm-audtachai, 2020-11-23:
> if you ICA clean your data you reduce their rank making covariance matrices not full rank (they have some eigenvalues which are 0). This is the error you see. What you could do is apply a PCA step before CSP to reduce the dimension of the data so the covariances become full rank.
> […](#)

How can I apply a PCA? & Which line that I need to apply PCA? Do they have example codes?
 this is my codes. 
[image]
[image]

agramfort, 2020-11-23:
see if you can insert in your Pipeline a PCA estimator

>

palm-audtachai, 2020-11-24:
> see if you can insert in your Pipeline a PCA estimator
> […](#)

I try to apply PCA and it doesn't have error anymore.
[image]

but output still reduce data rank from 22 to 19. Is this correct or not?
[image]

agramfort, 2020-11-24:
yes ICA cleaning and PCA does reduce the rank of the data. It's expected.

closed 2020-11-24 by agramfort