mne-tools/mne-python #8559: Upper triangular region of the matrix not filled up by the spectral_connectivity function

raphaelchristin, 2020-11-23:
I am experiencing a problem with the spectral_connectivity function from the connectivity module. The outputted matrix's upper triangular region is not filled up. The code I am running is very similar to this one, but with my own data : 

[image]

It appears as though the code enters the part where it should fill up the upper region, since it displays the text that happens right before it : 

[image]

Maybe it doesn't enter the loop where the matrix is filled up, but I see no reason for that.

Another weird that happened, when I downloaded the sample data to execute the the above code, the first time I ran it the matrix appeared to be filled up. When I switched the parameter faverage to False to test all the functionalities that I need, the matrices were not filled up. After that, I switched back to faverage = True, to see if that was the problem, but the resulting matrix was not filled up either. Any help would be appreciated!

agramfort, 2020-11-24:
please share a script not a screen capture that allows to replicate the problem

raphaelchristin, 2020-11-24:
Sorry about that, here is the code : 
`import mne
from mne import io
from mne.connectivity import spectral_connectivity
from mne.datasets import sample
from mne.viz import plot_sensors_connectivity

#print(__doc__)

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

# Setup for reading the raw data
raw = io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)

# Add a bad channel
raw.info['bads'] += ['MEG 2443']

# Pick MEG gradiometers
picks = mne.pick_types(raw.info, meg='grad', eeg=False, stim=False, eog=True,
                       exclude='bads')

# Create epochs for the visual condition
event_id, tmin, tmax = 3, -0.2, 1.5  # need a long enough epoch for 5 cycles
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(grad=4000e-13, eog=150e-6))

# Compute connectivity for band containing the evoked response.
# We exclude the baseline period
fmin, fmax = 3., 9.
sfreq = raw.info['sfreq']  # the sampling frequency
tmin = 0.0  # exclude the baseline period
epochs.load_data().pick_types(meg='grad')  # just keep MEG and no EOG now
con, freqs, times, n_epochs, n_tapers = spectral_connectivity(
    epochs, method='pli', mode='multitaper', sfreq=sfreq, fmin=fmin, fmax=fmax,
    faverage=True, tmin=tmin, mt_adaptive=False, n_jobs=1)

print(con)`

raphaelchristin, 2020-11-24:
Sorry about that, here is the code : 
```
import mne
from mne import io
from mne.connectivity import spectral_connectivity
from mne.datasets import sample
from mne.viz import plot_sensors_connectivity

#print(__doc__)

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

# Setup for reading the raw data
raw = io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)

# Add a bad channel
raw.info['bads'] += ['MEG 2443']

# Pick MEG gradiometers
picks = mne.pick_types(raw.info, meg='grad', eeg=False, stim=False, eog=True,
                       exclude='bads')

# Create epochs for the visual condition
event_id, tmin, tmax = 3, -0.2, 1.5  # need a long enough epoch for 5 cycles
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(grad=4000e-13, eog=150e-6))

# Compute connectivity for band containing the evoked response.
# We exclude the baseline period
fmin, fmax = 3., 9.
sfreq = raw.info['sfreq']  # the sampling frequency
tmin = 0.0  # exclude the baseline period
epochs.load_data().pick_types(meg='grad')  # just keep MEG and no EOG now
con, freqs, times, n_epochs, n_tapers = spectral_connectivity(
    epochs, method='pli', mode='multitaper', sfreq=sfreq, fmin=fmin, fmax=fmax,
    faverage=True, tmin=tmin, mt_adaptive=False, n_jobs=1)

print(con)
```

drammock, 2020-11-24:
I can replicate; plotting `con` yields this:

[image]

generated by:

```python
import matplotlib.pyplot as plt
import mne
from mne.connectivity import spectral_connectivity
data_path = mne.datasets.sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
# read raw data
raw = mne.io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)
# Add a bad channel
raw.info['bads'] += ['MEG 2443']
# Pick MEG gradiometers
picks = mne.pick_types(raw.info, meg='grad', eeg=False, stim=False, eog=True,
                       exclude='bads')
# Create epochs for the visual condition
event_id, tmin, tmax = 3, -0.2, 1.5  # need a long enough epoch for 5 cycles
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(grad=4000e-13, eog=150e-6))
# Compute connectivity for band containing the evoked response.
# We exclude the baseline period
epochs.load_data().pick_types(meg='grad')  # just keep MEG and no EOG now
con, freqs, times, n_epochs, n_tapers = spectral_connectivity(
    epochs, method='pli', mode='multitaper', sfreq=raw.info['sfreq'],
    fmin=3., fmax=9., faverage=True, tmin=0., mt_adaptive=False,
    n_jobs=1)
plt.imshow(con)
plt.show()
```

agramfort, 2020-11-25:
maybe it's on purpose as this matrix should be symmetric anyway? just asking

dengemann, 2020-11-25:
Yes it is a feature, not a bug. It has been like this for very long time now. You can always do C + C.T when you want to display it.

dengemann, 2020-11-25:
It is even documented in the API, see section Notes: https://mne.tools/stable/generated/mne.connectivity.spectral_connectivity.html#mne.connectivity.spectral_connectivity

raphaelchristin, 2020-11-25:
Thank you, sorry to bother with something that was in the API

drammock, 2020-11-25:
Perhaps the log message should be changed or removed then. It seems odd to say "filling the upper triangle" if we're filling it with zeros.

raphaelchristin, 2020-11-25:
I agree, that definitely is confusing

anaradanovic, 2021-03-18:
I can take care of this!:)

sappelhoff, 2021-03-19:
closed in #9147

closed 2021-03-19 by sappelhoff