mne-tools/mne-python #8562: set_3d_view doesn't set view in docs

rob-luke, 2020-11-24:
#### Describe the bug
I think that `set_3d_view` does not work any more, all the examples in https://mne.tools/stable/generated/mne.viz.set_3d_view.html#mne.viz.set_3d_view look incorrect to me. I don't know what some looked like before but the fNIRS one is definitely now not correct. 

#### Steps to reproduce
https://mne.tools/stable/auto_examples/visualization/plot_3d_to_2d.html#sphx-glr-auto-examples-visualization-plot-3d-to-2d-py

https://mne.tools/stable/auto_tutorials/preprocessing/plot_70_fnirs_processing.html#id1

#### Expected results
I expect the function to set the 3d view, but it docent seem to

#### Actual results
See doc links

#### Additional information

agramfort, 2020-11-24:
@GuillaumeFavelier can you have a look?

GuillaumeFavelier, 2020-11-24:
Thanks for reporting this, I'm on it!

closed 2020-11-24 by agramfort