mne-tools/mne-python #8563: TOPOs appear circular for MEG

agramfort, 2020-11-24:
now in a number of plots we have MEG topos that appear circular instead of following the helmet shape.

See for example:

https://mne.tools/dev/_images/sphx_glr_plot_20_visualize_epochs_008.png

in

https://mne.tools/dev/auto_tutorials/epochs/plot_20_visualize_epochs.html#sphx-glr-auto-tutorials-epochs-plot-20-visualize-epochs-py

I don't know if it's a problem of defaults but I also see around me a lot of people showing me MEG topos with circular shapes and I feel it should not happen.

larsoner, 2020-11-24:
@drammock do you want to look / add it to your queue somewhere?

drammock, 2020-11-24:
sure; topos are coming soon on my list

larsoner, 2020-12-01:
@drammock this one would be nice to fix for 0.22

commit 2b0f0c5a (2020-12-01): MRG, FIX, VIZ: correctly triage extrapolation mask for topos based on channel type (#8589)

* closes #8563

* touch tutorial

* update what's new
closed 2020-12-01 by larsoner