mne-tools/mne-python #8571: fNIR signal priors to EEG source reconstruction

LF9076, 2020-11-25:
#### Describe the new feature or enhancement
*Please provide a clear and concise description of what you want to add or
change.*
what can i do if i want to use the fNIR signals to be a spatial priors in the source estimates functions? Actually i think it need to change the Source covariance matrix.The matrix is a Is an identity matrix in the L2 norm formula.

#### Describe your proposed implementation
*Describe how you think the feature or improvement should be implemented (e.g., 
as a new method on an existing class? as new capability added to an existing
method?) If you're not sure, please delete this section and the next section.*

#### Describe possible alternatives
*If you've suggested an implementation above, list here any alternative
implementations you can think of, and brief comments explaining why the chosen
implementation is better.*

#### Additional comments
*Add any other context or screenshots about the feature request here.*

agramfort, 2020-11-25:
Please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) for usage questions.

agramfort, 2020-11-25:
or at least provide a detailed description of the ENH you want

closed 2020-11-25 by agramfort

LF9076, 2020-11-27:
The question is that i want to use the fNIR signal information to be a Spatial prior in EEG source reconstruction proceduce. some source reconstruction software can use fMRI information.Does MNE can do this?or in the future MNE can achieve it?Thanks a lot.