mne-tools/mne-python #8573: Saving annotations as CSV results in wrong orig_time

schneijan, 2020-11-25:
#### Describe the bug
When creating annotations with `orig_time=None` and saving them to a `*.csv` file using the `save` function, `orig_time=None` is ignored and replaced by `1970-01-01 00:00:00`. Thus, when importing the annotations again, the `orig_time` of the imported annotations differs from the original ones. This seems odd to me and not intended, because the issue not occurs when saving the annotations to `*.txt` file

#### Steps to reproduce
Here is a minimal working example:
```
# Create annotation and save it to CSV file
annotations = mne.Annotations(onset=[2.5], duration=[5], description=["Test"], orig_time=None)
annotations.save("annotation.csv")

# Read annotation from file again
read_annotations = mne.read_annotations("annotation.csv")

# Output orig_time
print(read_annotations[0]['orig_time']) #Outputs 1970-01-01 00:00:02.500000+00:00 instead of None
```

#### Expected results
I expetected the output to be `None`. This is even the result when using a `*.txt` file instead of a `*.csv` file.

#### Actual results
1970-01-01 00:00:02.500000+00:00
In the annotations dict, the orig_time is stored as `('orig_time', datetime.datetime(1970, 1, 1, 0, 0, 2, 500000, tzinfo=datetime.timezone.utc))`

agramfort, 2020-11-25:
there is a use case for files with absolute time stamps and a relative.

what was discussed in the past is that .csv will be index by time to
orig_time None is
hard to make it work ok. What I would suggest is to present saving
annotations to csv
without a valid orig_time

>

mmagnuski, 2020-12-06:
(@agramfort - I think you meant "prevent" so I took the liberty to correct the typo :) )

closed 2026-11-10 by larsoner