mne-tools/mne-python #8578: Toggle Annotations visibility in plot_raw()

hoechenberger, 2020-11-29:
If working with heavily-annotated data, it can become very difficult to tell overlapping Annotations apart, and their descriptions may overlap too – see e.g. this example based on the `sample` data:
[image]

I'm now intending to add EOG, ECG, and artifact Annotations to this dataset, but I'm **already** struggling to keep track of what's going on 🧐 Of course I could shrink the displayed time span, but this may not always be a feasible approach.

I wondered if we could offer an option to toggle visibility for each annotation individually? When in Annotations mode, we already display a window with radio buttons to select which Annotation(s) to add / modify. Could we maybe simply add checkboxes – probably left of the radio buttons – to toggle visibility?

cc @drammock

cbrnr, 2020-11-29:
I suggested this at some point, but don't know if it was in a separate issue or just buried in some longer discussion. In any case, +💯 for this feature!

agramfort, 2020-11-29:
one idea is to have rather than the test a filled circle and a tooltip that
appears when you pass your mouse
over it. Not clear how easy this would be with matplotlib

>

cbrnr, 2020-11-29:
I'd prefer an option to toggle individual annotations, but maybe another option could be to hide/minimize only the text.

hoechenberger, 2020-11-29:
A tooltip won't work well for overlapping annotations though...

hoechenberger, 2020-11-29:
Oh you meant, an area above the plot where we now have the text? Hm yes this could work

cbrnr, 2020-11-29:
SigViewer shows a tooltip when hovering over annotations. The tooltip contains all present annotations with their corresponding colors, so this also works.

hoechenberger, 2020-11-29:
@cbrnr Wow, is that an annotation for an individual channel? 😍

cbrnr, 2020-11-29:
Yes, although not all file formats support this.

larsoner, 2020-11-30:
A tooltip doesn't seem too bad from a `picker` standpoint I guess. But the overlap is still going to be a problem in that case.

A checkbox in the `annotations` window to disable at least the labeling, maybe also the annotation itself (maybe two check boxes then?) would be a good first step. @drammock up to you

PR #8624 (2020-12-12): VIZ, ENH: allow show/hide annotations by label
closes #8578
closed 2020-12-12 by hoechenberger