mne-tools/mne-python #8579: CSP.plot_filters() is actually plotting the transpose of filters

rubyyhj, 2020-11-29:
#### Describe the bug
The `plot_filters()` method of `mne.decoding.CSP` is not plotting filters, but actually plotting the transpose of filters.
Using version `0.21.2`.

#### Steps to reproduce
Use the code from [Motor imagery decoding from EEG data using the Common Spatial Pattern (CSP)](https://mne.tools/stable/auto_examples/decoding/plot_decoding_csp_eeg.html?highlight=csp) and change the last line of the second code block `csp.plot_patterns` to `csp.plot_filters`, and the result is:

[image]

And in `mne.decoding.CSP`, the each row of `filters_` attribute is one CSP filter, see: https://github.com/mne-tools/mne-python/blob/maint/0.21/mne/decoding/csp.py#L184. But if we set the second row of `filters_` to be all zeros, the figure almost has no change:
```python
csp.filters_[1, :] = 0  # the coeffs in the second filter(row) are all 0 
csp.plot_filters(epochs.info, ch_type='eeg', units='Patterns (AU)', size=1.5)
```
[image]

Instead, if we set the second column of `filters_` to be all zeros, then the "second filter" CSP1 becomes empty:
```python
csp.filters_[:, 1] = 0  # the second column of filters_
csp.plot_filters(epochs.info, ch_type='eeg', units='Patterns (AU)', size=1.5)
```
[image]

So `plot_filters` is actually plotting the transpose of filters (the columns of `filters_`), not the csp filters (the rows of `filters_`).

#### To fix it
I think just change this line of code https://github.com/mne-tools/mne-python/blob/maint/0.21/mne/decoding/csp.py#L479 to 
`filters = EvokedArray(self.filters_.T, info, tmin=0)` could fix this issue. I tried the following code and the figure looks more reasonable to be CSP filters:
```
csp.filters_ = csp.filters_.T
csp.plot_filters(epochs.info, ch_type='eeg', units='Patterns (AU)', size=1.5)
```
[image]

BTW, the figure from `plot_patterns` method is correct.

cbrnr, 2020-11-29:
LGTM. Can you submit a PR?

rubyyhj, 2020-11-29:
Submitted: #8580. @cbrnr

PR #8580 (2020-11-29): FIX: Transpose issue of CSP.plot_filters
#### Reference issue
Fixes #8579.

#### What does this implement/fix?
Just add a transpose to correct the CSP filters plot.
closed 2020-11-29 by cbrnr