mne-tools/mne-python #8581: read_ica_eeglab & ica.plot_components: ica.info needs to be filled

behinger, 2020-11-30:
(was discussed on gitter as well)
tldr; read_raw_eeglab returns no ica.info, ica.info=raw.info is a bad idea 

``` python
import mne
raw = mne.io.read_raw_eeglab('sub-009_ses-P3_task-P3_eeg.set')
raw.set_channel_types(mapping={'HEOG_left':'eog','HEOG_right':'eog','VEOG_lower':'eog'})

raw.set_montage('standard_1020',match_case=False)
badChannels = [11 ,27]
raw.info['bads'].extend([raw.ch_names[i] for i in badChannels])  # add a list of channels

ica = mne.preprocessing.read_ica_eeglab('sub-009_ses-P3_task-P3_ica.set')
```
```
print(raw)
print(ica)
```
> <RawEEGLAB | sub-009_ses-P3_task-P3_eeg.fdt, 33 x 361472 (353.0 s), ~48 kB, data not loaded>
> <ICA | epochs decomposition, fit (imported_eeglab):  samples, 28 components>

```python
#ica.set_montage('standard_1020',match_case=False) <-- doesnt work because function does not exist

ica.info = raw.info # <-- alex mentioned we shouldnt do that either.
ica.plot_components(1) # <-- we get an error here because ICA is 28x28 matrix, but raw has shape 30 because the two channels were not yet removed (thus their indices go from 0-29, with 11 and 27 removed, but we cant indice a 28 matrix on position 29))
```
Results in following error:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-4-57fc80c15773> in <module>
      2 
      3 ica.info = raw.info # <-- alex mentioned we shouldnt do that either.
----> 4 ica.plot_components(1) # <-- we get an error here because ICA is 28x28 matrix, but raw has shape 30 because the two channels were not yet removed (thus their indices go from 0-29, with 11 and 27 removed, but we cant indice a 28 matrix on position 29))

/netpool/work/ccs-srv-001/users/ehinger/projects/course_eeg_WS2020/local/venv/lib/python3.8/site-packages/mne/preprocessing/ica.py in plot_components(self, picks, ch_type, res, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, inst, plot_std, topomap_args, image_args, psd_args, reject, sphere)
   1802                         image_args=None, psd_args=None, reject='auto',
   1803                         sphere=None):
-> 1804         return plot_ica_components(self, picks=picks, ch_type=ch_type,
   1805                                    res=res, vmin=vmin,
   1806                                    vmax=vmax, cmap=cmap, sensors=sensors,

/netpool/work/ccs-srv-001/users/ehinger/projects/course_eeg_WS2020/local/venv/lib/python3.8/site-packages/mne/viz/topomap.py in plot_ica_components(ica, picks, ch_type, res, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, inst, plot_std, topomap_args, image_args, psd_args, reject, sphere)
   1170 
   1171     data = np.atleast_2d(data)
-> 1172     data = data[:, data_picks]
   1173 
   1174     # prepare data for iteration

IndexError: index 28 is out of bounds for axis 1 with size 28
```

**Important note** This error can only occur if you copy over raw.info to ica.info (which is empty by default), I have seen this at multiple places.

This is because in order to match raw to ICA-matrices, the "bad" channels are removed, but the original indices are still being used. Therefore we try to indice e.g. data[:,29] which doesnt work because data.shape = 28x28. So I think the underlying reason is, that there is no match of channel names used in the imported ICA against the channel names in header. 

### Proposed solution:
read_eeglab_ica L2743: add     `ica.info = info` (which was already calculated but never saved to the ICA object.)
and add `ica._update_ica_names()` somewhere at the end

If I do these two steps (and read in the montage afterwards) my problem is resolved. I can organize a pull-request and unit-test if this is a viable solution.

### Intermediate solution:
My problem gets fixed if I manually match the channels (this is effectively the same thing as the proposed solution but outside of ica.py

```
# assuming ica is subset of raw 
ch_raw = raw.info['ch_names']
ch_ica = ica.ch_names

ix = [k for k,c in enumerate(ch_raw) if c in ch_ica and not c in raw.info['bads']]

info = raw.info.copy()
mne.io.pick.pick_info(info, ix, copy=False)
ica.info = info
ica.info.set_montage('standard_1020',match_case=False) # not strictly necessary I think if raw.info has a montage already
ica._update_ica_names()

ica.plot_components(0)
```

### Minimal reproducible example
[bug ipynb html output](https://benediktehinger.de/upload/mne_bug.html)
[data+code, 30mb, ERP core dataset](https://benediktehinger.de/upload/ica_bug.zip)

agramfort, 2020-11-30:
I confirm that the read_ica_eeglab is not doing what it should do. 

You have all you need in the ica file to get channel names and locations.
Now I am not sure these are standard but if they are read_ica_eeglab
should populate the info attribute of ica.

```
>> EEG.chaninfo.filecontent

ans =

1	 -17.926	 0.51499	     FP1
2	 -39.947	 0.34459	      F3
3	 -53.913	 0.52808	      F7
4	 -62.425	 0.28822	     FC3
5	     -90	 0.26669	      C3
6	     -90	  0.3999	      C5
7	 -140.05	 0.34459	      P3
8	 -126.09	 0.52808	      P7
9	 -126.69	 0.66022	      P9
10	 -144.11	 0.52233	     PO7
11	 -157.54	 0.42113	     PO3
12	 -162.07	 0.51499	      O1
13	     180	 0.50669	      Oz
14	     180	 0.25338	      Pz
15	     180	 0.12662	     CPz
16	  17.926	 0.51499	     FP2
17	       0	 0.25338	      Fz
18	  39.897	  0.3445	      F4
19	  53.867	 0.52807	      F8
20	  62.425	 0.28822	     FC4
21	       0	 0.12662	     FCz
22	       0	       0	      Cz
23	      90	 0.26667	      C4
24	      90	  0.3999	      C6
25	   140.1	  0.3445	      P4
26	  126.13	 0.52807	      P8
27	  126.69	 0.66022	     P10
28	  144.14	 0.52231	     PO8
29	  157.54	 0.42113	     PO4
30	  162.07	 0.51499	      O2
31	       0	     0.5	HEOG_left
32	       0	     0.5	HEOG_right
33	       0	     0.5	VEOG_lower
34	       0	     0.5	REF
```

it also contains EEG.chanlocs

and icachansind should be used to pick in the info.

so the summary is @behinger it's not your fault. This code has not been
used enough yet.

do you have time to look into? shall we?

behinger, 2020-12-02:
I have a solution & will write a short unit-test and generate a pull-request. But it might take a week or so

agramfort, 2020-12-02:
great thanks!

>