mne-tools/mne-python #8582: Bug in mne.io.read_raw_gdf()

apospraf, 2020-11-30:
#### Describe the bug
Lowpass and highpass cut off frequencies are swapped when using `mne.io.read_raw_gdf()` to load a gdf file from the BCI IV 2b dataset.

#### Steps to reproduce

1. Download BCI IV 2b dataset from [here](http://www.bbci.de/competition/iv/) (it requires a name and an email)
2. Load one of the gdf files using `mne.io.read_raw_gdf()`

#### Expected results
`raw_gdf.info` should return:
```
<Info | 7 non-empty values
 bads: []
 ch_names: EEG:C3, EEG:Cz, EEG:C4, EOG:ch01, EOG:ch02, EOG:ch03
 chs: 6 EEG
 custom_ref_applied: False
 highpass: 0.5 Hz
 lowpass: 100 Hz
 meas_date: 2006-03-07 09:23:36 UTC
 nchan: 6
 projs: []
 sfreq: 250.0 Hz
>
```

Also `raw_gdf.plot()` should look like the figure below
[image]

#### Actual results
> `raw_gdf.info`
```
<Info | 7 non-empty values
 bads: []
 ch_names: EEG:C3, EEG:Cz, EEG:C4, EOG:ch01, EOG:ch02, EOG:ch03
 chs: 6 EEG
 custom_ref_applied: False
 highpass: 100.0 Hz
 lowpass: 0.5 Hz
 meas_date: 2006-03-07 09:23:36 UTC
 nchan: 6
 projs: []
 sfreq: 250.0 Hz
>
```
> `raw_gdf.plot()`
[image]

#### Additional information
I'm using the 0.22.dev0 version of mne on Ubuntu 18.04

cbrnr, 2020-11-30:
I can confirm the issue (my response to the mailing list assumed that you were referring to data set 2A, which is also available as GDF and does work fine).

agramfort, 2020-11-30:
so it works fine with on gdf file and not with another one?

is this a problem in the files or in the reader?

>

cbrnr, 2020-11-30:
It's strange that only these files do not work whereas others work just fine. Our code seems to read the header fields correctly, according to the [GDF specifications (page 13)](https://arxiv.org/pdf/cs/0608052.pdf) the fields are read in correct order (lowpass, highpass, notch).

I did notice that filter information in the fixed header does contain the values `LP:     1 Hz; HP: 100.00 Hz; Notch: NaN` in the GDF files of data set 2B (but not 2A, where these values are correct in the file and therefore also correctly read by MNE).

Now GDF says that we shouldn't read filter settings from the fixed header, because these settings are stored in the variable header - which is exactly what we do. It seems that these (wrong) values are actually stored in the variable header as well, because we read 1 Hz for the lowpass settings and 100 Hz for the highpass settings. Therefore, I'm inclined to think that these files contain invalid values, but we read them correctly.

Maybe we should sanity-check these settings and ignore them if e.g. LP < HP? @schloegl could you please chime in and let me know what's going on here?

schloegl, 2020-11-30:
> It's strange that only these files do not work whereas others work
> just fine. Our code seems to read the header fields correctly,
> according to the GDF specifications (page 13) [...]

I can confirm, that the information in these GDF files is incorrect, 
Lowpass and HighPass information should be switched in data set 2B.

Of course, it would be possible to re-encode the data in the correct 
way, but after 10+ years there are too many copies around. Therefore, 
I'd recommend to document this flawed "property" of that data set.

Kind regards,
    Alois

cbrnr, 2020-11-30:
Thank you Alois! So I guess we can close this issue. @apospraf you already figured out a workaround how to work with the data. Let me know if you have any further questions.

PR #8584 (2020-12-02): Detect/correct invalid HP/LP settings in EDF/BDF/GDF
Fixes #8582. EDF/BDF/GDF files can contain invalid LP/HP filter settings (cutoff frequencies for HP can be greater than the one for LP). This PR (1) issues a warning and (2) sets HP/LP values for affected channels to `None`.
closed 2020-11-30 by cbrnr